riscv-collab/riscv-gnu-toolchain #1518: Executing my targeted workload on spike RISC-V ISA after adding custom trigonometric Instructions

baul-iisc, 2024-07-23:
Hello Team,

I hope this message finds you well.

I am seeking assistance with adding new instructions to the RISC-V ISA in order to characterize and test my targeted workloads. After consulting various sources, and with support from the RISC-V community, I have successfully added these instructions to the Spike binutils.

However, I am unsure where to implement these trigonometric micro-operations within the Spike simulator to execute my targeted workload. I would greatly appreciate it if you could provide guidance on the end-to-end process of adding new instructions to the Spike RISC-V ISA to ensure proper execution of the workload.

Your assistance in this matter would be immensely valuable.

Thank you for your time and support.

Best regards,
Mr. BOUL CHANDRA GARAI

TommyMurphyTM1234, 2024-07-23:
Also asked here: 

- https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/oPFIZKykdfE/m/RTmAJSCcAQAJ

This is a question about Spike and not `riscv-isa-sim` so should probably be asked over there and not here: 

- https://github.com/riscv-software-src/riscv-isa-sim/issues

In fact, I'm pretty sure that existing open or closed issues there probably already cover this topic. 

- https://github.com/riscv-software-src/riscv-isa-sim/issues?q=is%3Aissue+custom+instruction

TommyMurphyTM1234, 2024-07-23:
For completeness and to avoid possible duplicated effort, please note that this question was also asked here on the riscv-gnu-toolchain GitHub repository:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1518

TommyMurphyTM1234, 2024-07-25:
Any update @baul-iisc?

baul-iisc, 2024-07-25:
Thank you, I have gone through all the references you had provided. But I am still unable to figure it out the full end-to-end procedures, which provides what are all the modifications are required to the spike riscv-toolchain, so that my targeted workload can execute on the riscv spike simulator with these newly added trigonometric instructions.
It will be really helpful if someone can help me on this.
Thank you inanticipations.

External Email

TommyMurphyTM1234, 2024-07-25:
> Thank you, I have gone through all the references you had provided. But I am still unable to figure it out the full end-to-end procedures, which provides what are all the modifications are required to the spike riscv-toolchain

There is no such thing as "the spike riscv-toolchain".
Your issue seems to be with modifying Spike.

> However, I am unsure where to implement these trigonometric micro-operations within the Spike simulator to execute my targeted workload.

In which case you need to ask about that in the Spike community/GitHub repo and not here on the `riscv-gnu-toolchain` repo.

TommyMurphyTM1234, 2024-07-25:
I don't know how to to date or not this is but it may provide some guidance: 

- https://pcotret.gitlab.io/riscv-custom/sw_toolchain.html#adding-a-custom-instruction-in-spike

baul-iisc, 2024-07-25:
Thank you for your response. I have reviewed the details provided here regarding the addition of new instructions in Spike and Gem5, I had seen also, but this is giving the details about adding the new instructions on spike and gem5 .
However, I am specifically looking for the steps required to compile my workload in Spike and Gem5 with these newly added instructions and then characterize the workload to evaluate the performance impact of these instructions.
I would greatly appreciate it if someone could provide guidance on how to compile code that utilizes these newly added instructions and execute it on Spike pk to characterize the workload.
Thank you in advance for your assistance.

External Email

  *   https://pcotret.gitlab.io/riscv-custom/sw_toolchain.html#adding-a-custom-instruction-in-spike

TommyMurphyTM1234, 2024-07-25:
> However, I am specifically looking for the steps required to compile my workload in Spike and Gem5 with these newly added instructions and then characterize the workload to evaluate the performance impact of these instructions.

As I've already said, these are issues/questions that you really need to raise in the Spike (and Gem5) forums and not here on the `riscv-gnu-toolchain` repo.

baul-iisc, 2024-07-25:
Thank you Tommy,
Would you please share the spike community email, where I can raise my request?

External Email

TommyMurphyTM1234, 2024-07-25:
> Thank you Tommy, Would you please share the spike community email, where I can raise my request?

-

TommyMurphyTM1234, 2024-08-01:
Closing this as it's essentially a question about Spike and not `riscv-gnu-toolchain`.

closed 2024-08-01 by TommyMurphyTM1234