riscv-collab/riscv-gnu-toolchain #1521: PLT section in RISCV ELF

tianboh, 2024-07-25:
Hi there,

I am working on a disassembling project for RISCV ELF. I need to consider function call towards shared library, something like `malloc@plt` in `x86`. I noticed that [ELF toolchain does not support shared library](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/803). Does this mean that RISCV ELF do not have a PLT section for relocation? If so, does this mean that each executable RISCV file does not rely on any shared library(it is statically linked)?

tianboh, 2024-07-25:
Well, after some search, I found that RISCV ELF is supported [[ELF][RISCV] Support PLT, GOT, copy and relative relocations](https://reviews.llvm.org/D63076) and [RISCV ELF SPEC](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-elf.adoc). However, I am still wondering how can I generate a shared library on RISCV.

TommyMurphyTM1234, 2024-07-25:
Bare metal toolchains target bare metal systems with no OS or a minimal/lightweight [RT]OS layer that has no dynamic link loader support which means that shared libraries are generally not supported. This is basically what the first link that you posted says. I don't know about the LLVM link that you posted but I would not expect a bare metal toolchains and bare metal platform to support dynamic linking and relocating. For that you almost certainly really need a Linux toolchain.

tianboh, 2024-07-25:
@TommyMurphyTM1234 Thank you for your reply and it really helps to clarify my misconceptions! 

To provide a general picture, I am working on RISCV target of a [disassembling project](https://github.com/microsoft/llvm-mctoll). The goal is to raise a RISCV ELF file to LLVM IR. However, when this ELF file uses some shared libraries(seems impossible on bare metal now, but what about Linux?), I am not sure how its PLT section looks like.

In other words, I am not running this ELF file on bare metal or Linux, I am just disassembling it. I want to check its PLT section to better disassemble it. So I guess my question now is, how can I generated a RISCV shared library on Linux?

tianboh, 2024-07-25:
> The goal is to raise a RISCV ELF file to LLVM IR.

It is like reverse engineering, given a bytecode file of RISCV(ELF), this project can generate its corresponding LLVM IR file.

Thank you for the link! It's a good tutorial for shared library on x86 linux.

> you'll need to adjust things to use the RISC-V toolchain commands and any relevant RISC-V specific compile time options

I am confused here. Do you mean tools like `riscv64-unknown-elf-gcc` and `riscv64-unknown-elf-ld`? I run the following
```
$riscv64-unknown-elf-gcc -shared -fpic hello.c
riscv64-unknown-elf/bin/ld: -shared not supported
```

TommyMurphyTM1234, 2024-07-25:
> I am confused here. Do you mean tools like `riscv64-unknown-elf-gcc` and `riscv64-unknown-elf-ld`? I run the following
> 
> ```
> $riscv64-unknown-elf-gcc -shared -fpic hello.c
> riscv64-unknown-elf/bin/ld: -shared not supported
> collect2: error: ld returned 1 exit status
> ```
As already stated, bare-metal targets and toolchains don't support shared libraries. If you want to compile/deal with shared libraries then you need to use the Linux toolchain.

tianboh, 2024-07-25:
Okay, I see. Thank you for your answer! Really appreciate :)

closed 2024-07-25 by tianboh