riscv-collab/riscv-gnu-toolchain #1522: When rm is added to the assembly, GCC fails to compile for fcvt.d.w/fcvt.d.wu/fcvt.s.h/fcvt.d.h/fcvt.d.s instructions

Shanchuang666, 2024-08-01:
[image]

**fcvt.d.w / fcvt.d.wu / fcvt.s.h / fcvt.d.h / fcvt.d.s**

Although these instructions are always accurate, and the implementation is not affected by the rounding mode, the specification appears to be able to configure rm, but after adding rm to the assembly, gcc cannot compile successfully, indicating "illegal operands", then does the display at the specification screenshot need to be modified?

TommyMurphyTM1234, 2024-08-01:
Please provide clear instructions on how to reproduce your issue. Ideally including a small self-contained reproducible test case.

aswaterman, 2024-08-01:
This isn’t a bug. At the ISA level, the instructions do have an `rm` field, but there is no requirement that the assembler support all instruction variants that the ISA supports.

TommyMurphyTM1234, 2024-08-01:
Maybe also relevant? 

- https://github.com/riscvarchive/riscv-binutils-gdb/issues/139

In particular this: 

> These instructions do not round. Their result is always exact

cmuellner, 2024-08-01:
The spec states "FCVT.D.W[U] always produces an exact result and is unaffected by rounding mode.".
Therefore, it was decided not to accept rounding modes for these instructions.

Note, that this is not implemented in GCC, but in the assembler (Binutils).

E.g.:
```
fcvt.d.lu f0, x0
fcvt.d.lu f0, x0, rne
fcvt.d.lu f0, x0, rtz
fcvt.d.w f0, x0
fcvt.d.w f0, x0, rne
fcvt.d.w f0, x0, rtz
fcvt.d.wu f0, x0
fcvt.d.wu f0, x0, rne
fcvt.d.wu f0, x0, rtz
```
```
$ riscv64-unknown-linux-gnu-as -march=rv64gc fcvt.s 
fcvt.s: Assembler messages:
fcvt.s:5: Error: illegal operands `fcvt.d.w f0,x0,rne'
fcvt.s:6: Error: illegal operands `fcvt.d.w f0,x0,rtz'
fcvt.s:8: Error: illegal operands `fcvt.d.wu f0,x0,rne'
fcvt.s:9: Error: illegal operands `fcvt.d.wu f0,x0,rtz'
```

TommyMurphyTM1234, 2024-08-01:
Closing this since it's not a valid issue.

closed 2024-08-01 by TommyMurphyTM1234

Shanchuang666, 2024-08-02:
Thank you for all your responses.