riscv-collab/riscv-gnu-toolchain #1524: Missing asset in latest release

gilbsgilbs, 2024-08-03:
Hi,

The new [2024.08.03](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.08.03) release is missing a riscv64 glibc LLVM toolchain. Apparently, it is just due to a transient issue when uploading the artifact from github action: https://github.com/riscv-collab/riscv-gnu-toolchain/actions/runs/10223699266/job/28293017317 . Someone probably needs to upload the asset manually from [the artifacts](https://github.com/riscv-collab/riscv-gnu-toolchain/actions/runs/10223699266#artifacts), or just retry the failed job.

Thanks.

kito-cheng, 2024-08-03:
Thanks for reporting that, let me restart that to see if it can fix :p

gilbsgilbs, 2024-08-03:
Looks like it worked :tada: . Thanks!

closed 2024-08-03 by gilbsgilbs