riscv-collab/riscv-gnu-toolchain #1526: Custom Float Instructions

amithmath, 2024-08-07:
Hello, 

I want to add custom  single precision floating point sine, cosine, exp instructions,  and I have designed hardware for this.  I was going through tutorials on how to add custom instructions at: https://pcotret.gitlab.io/riscv-custom/sw_toolchain.html after adding custom instructions,  I think one has to use  asm volatile macro to use custom instructions in  C. Is there anyway where one do not need  to use this macro? I would like the compiler automatically map to custom instructions in the risc-v processor, like how it will map to fadd.s, fmul.s instructions, where one do not need to use asm volatile macro.

Amith

TommyMurphyTM1234, 2024-08-07:
You'd need to change the compiler so that it can identify intermediate code patterns for which your custom instruction can be generated automatically. How to do this is outside the scope of the `riscv-gnu-toolchain` repository and, in any case, is not really a RISC-V specific issue/question. The existing open and closed issues here and maybe in the archived `riscv-gcc` repository:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue
- https://github.com/riscvarchive/riscv-gcc/issues?q=is%3Aissue

may contain some relevant info - this one may be relevant for example:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837

But, ultimately, this is something that you need to research and ask about in a GCC forum such as the GCC internals documentation and mailing lists for example:

- https://gcc.gnu.org/onlinedocs/gccint/
- https://gcc.gnu.org/lists.html

TommyMurphyTM1234, 2024-08-14:
Closing issue as it's not really question about `riscv-gnu-toolchain` specifically.

closed 2024-08-14 by TommyMurphyTM1234