riscv-collab/riscv-gnu-toolchain #1529: unrecognized option '--64'

EventScheduler, 2024-08-08:
Hi all,
  when I try to build riscv-gnu-toolchain, I got the following errors, who know why? commands are:
      **export RISCV=/home/leiyu/source_code/tools/riscv/rocket-tools
      rm -rf build;mkdir build;cd build; ../configure --prefix=$RISCV --disable-gdb;make -j**
  error output:
g++ -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../../../gcc/gcc -I../../../gcc/gcc/build -I../../../gcc/gcc/../include  -I../../../gcc/gcc/../libcpp/include  \
        -o build/sort.o ../../../gcc/gcc/sort.cc
echo timestamp > doc/gfdl.7
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=7 gfdl.pod > doc/gfdl.7.T$$ && \
        mv -f doc/gfdl.7.T$$ doc/gfdl.7) || \
        (rm -f doc/gfdl.7.T$$ && exit 1)
echo timestamp > doc/gpl.7
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=7 gpl.pod > doc/gpl.7.T$$ && \
        mv -f doc/gpl.7.T$$ doc/gpl.7) || \
        (rm -f doc/gpl.7.T$$ && exit 1)
echo timestamp > doc/fsf-funding.7
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=7 fsf-funding.pod > doc/fsf-funding.7.T$$ && \
        mv -f doc/fsf-funding.7.T$$ doc/fsf-funding.7) || \
        (rm -f doc/fsf-funding.7.T$$ && exit 1)
echo timestamp > doc/gcov-tool.1
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=1 gcov-tool.pod > doc/gcov-tool.1.T$$ && \
        mv -f doc/gcov-tool.1.T$$ doc/gcov-tool.1) || \
        (rm -f doc/gcov-tool.1.T$$ && exit 1)
echo timestamp > doc/gcov-dump.1
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=1 gcov-dump.pod > doc/gcov-dump.1.T$$ && \
        mv -f doc/gcov-dump.1.T$$ doc/gcov-dump.1) || \
        (rm -f doc/gcov-dump.1.T$$ && exit 1)
echo timestamp > doc/lto-dump.1
(pod2man --center="GNU" --release="gcc-12.2.0" --date=2022-08-19 --section=1 lto-dump.pod > doc/lto-dump.1.T$$ && \
        mv -f doc/lto-dump.1.T$$ doc/lto-dump.1) || \
        (rm -f doc/lto-dump.1.T$$ && exit 1)
../../../gcc/gcc/gengtype-lex.l: In function ‘int yylex(const char**)’:
gengtype-lex.cc:365:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
 #define YY_DO_BEFORE_ACTION \
  ~~~~~~~~~~~~~^~~~~~~
../../../gcc/gcc/gengtype-lex.l:118:1: note: in expansion of macro ‘YY_DO_BEFORE_ACTION’
 }
 ^
../../../gcc/gcc/gengtype-lex.l:113:1: note: here
 "~"     |
 ^~~~
gengtype-lex.cc:365:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
 #define YY_DO_BEFORE_ACTION \
  ~~~~~~~~~~~~~^~~~~~~
../../../gcc/gcc/gengtype-lex.l:138:1: note: in expansion of macro ‘YY_DO_BEFORE_ACTION’
     ;
 ^   ~
../../../gcc/gcc/gengtype-lex.l:134:1: note: here
 "ENUM_BITFIELD"{WS}?"("{WS}?{ID}{WS}?")" {
 ^~~~
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/genversion.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/genchecksum.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/errors.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/gengenrtl.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/genhooks.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/sort.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/gengtype-parse.o] Error 1
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/gengtype-lex.o] Error 1
/bin/sh ../../../gcc/gcc/../move-if-change tmp-optionlist optionlist
echo timestamp > s-options
/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2823: build/genmodes.o] Error 1
rm gcov.pod fsf-funding.pod gpl.pod cpp.pod gfdl.pod lto-dump.pod gcov-dump.pod gcov-tool.pod
make[2]: Leaving directory '/home/radl/hdd/leiyu/source_code/Inchi/rocket-tools/riscv-gnu-toolchain/build/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4619: all-gcc] Error 2
make[1]: Leaving directory '/home/radl/hdd/leiyu/source_code/Inchi/rocket-tools/riscv-gnu-toolchain/build/build-gcc-newlib-stage1'
make: *** [Makefile:589: stamps/build-gcc-newlib-stage1] Error 2

TommyMurphyTM1234, 2024-08-08:
Please detail all of the commands used to reproduce this issue. 
All commands from the initial `git clone`, to the `configure` command to the `make` command. 
Also capture and attach (don't copy and paste) the build log. 
I built the toolchain yesterday as follows and didn't have any problems. 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
```
If you issue is with something in the Rocket Chip repo then you need to report it there and not here. 
As far as I know the `rockethtools` stuff over there is deprecated and should not be used:

- https://github.com/chipsalliance/rocket-chip/issues/3584#issuecomment-2246603585

TommyMurphyTM1234, 2024-08-08:
> I built the toolchain yesterday as follows and didn't have any problems.
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> ./configure --prefix=`pwd`/installed-tools
> make 2>&1 | tee build.log
> ```

I did this again just now and had no problems:

- `build.log`: [build.log](https://github.com/user-attachments/files/16551368/build.log)

TommyMurphyTM1234, 2024-08-08:
> I did this again just now and had no problems:
>
> * `build.log`: [build.log](https://github.com/user-attachments/files/16551368/build.log)

BTW - there's also no occurrence of `--64` anywhere in that `build.log` so if that's the root of the original issue then it's not at all clear where it's coming from.

EventScheduler, 2024-08-10:
I reproduced my issue with the following commands:

> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log

[build.log](https://github.com/user-attachments/files/16568373/build.log)

Thank you!

TommyMurphyTM1234, 2024-08-10:
> I reproduced my issue with the following commands:
> 
> > git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> > cd riscv-gnu-toolchain
> > ./configure --prefix=`pwd`/installed-tools
> > make 2>&1 | tee build.log
> 
> [build.log](https://github.com/user-attachments/files/16568373/build.log)
> 
> Thank you!

What is your `PATH` environment variable set to?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1529#issuecomment-2276076072

EventScheduler, 2024-08-10:
This is my $PATH:
/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/bin:/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/examples/tutorials/apps/coherency/bin:.:/home/radl/hdd/leiyu/source_code/tools/firtool-1.56.1/bin:/home/radl/hdd/leiyu/.local:/home/radl/hdd/leiyu/.local/bin:/home/leiyu/source_code/tools/riscv64-lp64d--glibc--stable-2024.02-1/bin:/home/radl/hdd/leiyu/.local/bin:/home/radl/hdd/leiyu/bin:/usr/share/Modules/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/leiyu/source_code/tools/synopsys/install_dir/vcs/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/verdi/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/scl/2021.12/linux64/bin

I've removed the "/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/" setting from $PATH before building.

EventScheduler, 2024-08-10:
> This is my $PATH: [...]

> > I reproduced my issue with the following commands:
> > > git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> > > cd riscv-gnu-toolchain
> > > ./configure --prefix=`pwd`/installed-tools
> > > make 2>&1 | tee build.log
> > 
> > 
> > [build.log](https://github.com/user-attachments/files/16568373/build.log)
> > Thank you!
> 
> What is your `PATH` environment variable set to?
> 
> * [unrecognized option '--64' #1529 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1529#issuecomment-2276076072)

This is my $PATH:
/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/bin:/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/examples/tutorials/apps/coherency/bin:.:/home/radl/hdd/leiyu/source_code/tools/firtool-1.56.1/bin:/home/radl/hdd/leiyu/.local:/home/radl/hdd/leiyu/.local/bin:/home/leiyu/source_code/tools/riscv64-lp64d--glibc--stable-2024.02-1/bin:/home/radl/hdd/leiyu/.local/bin:/home/radl/hdd/leiyu/bin:/usr/share/Modules/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/leiyu/source_code/tools/synopsys/install_dir/vcs/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/verdi/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/scl/2021.12/linux64/bin

I've removed the "**/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/**" setting from $PATH before building.

EventScheduler, 2024-08-10:
This is my $PATH:
/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/bin:/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/examples/tutorials/apps/coherency/bin:.:/home/radl/hdd/leiyu/source_code/tools/firtool-1.56.1/bin:/home/radl/hdd/leiyu/.local:/home/radl/hdd/leiyu/.local/bin:/home/leiyu/source_code/tools/riscv64-lp64d--glibc--stable-2024.02-1/bin:/home/radl/hdd/leiyu/.local/bin:/home/radl/hdd/leiyu/bin:/usr/share/Modules/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/leiyu/source_code/tools/synopsys/install_dir/vcs/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/verdi/T-2022.06/bin:/home/leiyu/source_code/tools/synopsys/install_dir/scl/2021.12/linux64/bin

I've removed the "**/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/**" setting from $PATH before building.

TommyMurphyTM1234, 2024-08-10:
This part of your log makes no sense to me:
```
/bin/sh ../.././gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././gcc/gcc/genmodes.cc
/tmp/www/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2858: build/genmodes.o] Error 1
make[2]: Leaving directory '/tmp/www/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4627: all-gcc] Error 2
make[1]: Leaving directory '/tmp/www/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:636: stamps/build-gcc-newlib-stage1] Error 2
```
1. I can't see where `--64` is coming from and
2. I have no idea why the attempt to compile `genmodes.cc` using the host's `x86_64-pc-linux-gnu` `g++` seems to end up trying to call the RISC-V bare-metal assembler `/tmp/www/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/as`.

This is the equivalent part of my log:
```
/bin/bash ../.././gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././gcc/gcc/genmodes.cc
g++ -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
	-o build/errors.o ../.././gcc/gcc/errors.cc
```
You seem to be using Red Hat Enterprise Linux 8 - is that correct?
I'm not sure if that is available for download for me to try.
And I'm not sure if alternatives such as CentOS or Rocky Linux would be directly comparable?

I thought that enabling verbose logging for C++ might help here and give more details about what's happening with your failing build but that made my own previously successful build fail for some reason,,,
```
make distclean && rm -rf installed-tools
make CXXFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build.log

...

 /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/./gcc/cc1plus -quiet -nostdinc++ -v -I /home/user/Downloads/issue-1529/gcc-14/libstdc++-v3/../libgcc -I /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf -I /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include -I /home/user/Downloads/issue-1529/gcc-14/libstdc++-v3/libsupc++ -imultilib . -iprefix /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/14.2.1/ -isysroot /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../riscv64-unknown-elf -isystem /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/./gcc/include -isystem /home/user/Downloads/issue-1529/build-gcc-newlib-stage2/./gcc/include-fixed -isystem /home/user/Downloads/issue-1529/installed-tools/riscv64-unknown-elf/include -isystem /home/user/Downloads/issue-1529/installed-tools/riscv64-unknown-elf/sys-include /home/user/Downloads/issue-1529/gcc-14/libstdc++-v3/src/c++11/cxx11-ios_failure.cc -quiet -dumpbase cxx11-ios_failure.cc -dumpbase-ext .cc -mcmodel=medlow -mtune=rocket -march=rv64imafdc_zicsr -mabi=lp64d -misa-spec=20191213 -march=rv64imafdc_zicsr -gno-as-loc-support -Os -Wall -Wextra -Wwrite-strings -Wcast-qual -Wabi=2 -std=gnu++11 -version -fno-implicit-templates -fdiagnostics-show-location=once -ffunction-sections -fdata-sections -frandom-seed=cxx11-ios_failure-lt.s -o cxx11-ios_failure.s
GNU C++11 (g897cd794d) version 14.2.1 20240809 (riscv64-unknown-elf)
        compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/home/user/Downloads/issue-1529/installed-tools/riscv64-unknown-elf/sys-include"
ignoring nonexistent directory "/home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/14.2.1/include"
ignoring nonexistent directory "/home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/14.2.1/include-fixed"
ignoring nonexistent directory "/home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/14.2.1/../../../../riscv64-unknown-elf/include"
ignoring nonexistent directory "/home/user/Downloads/issue-1529/build-gcc-newlib-stage2/gcc/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/14.2.1/include"
[... cut ...]
make[3]: Leaving directory '/home/user/Downloads/issue-1529/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[2]: *** [Makefile:12244: all-target-libstdc++-v3] Error 2
make[2]: Leaving directory '/home/user/Downloads/issue-1529/build-gcc-newlib-stage2'
make[1]: *** [Makefile:1055: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1529/build-gcc-newlib-stage2'
make: *** [Makefile:742: stamps/build-gcc-newlib-stage2] Error 2
```
Unfortunately, at the moment I have no idea why your build fails the way that it does...

TommyMurphyTM1234, 2024-08-10:
Are you absolutely sure about this?

> I've removed the "/home/leiyu/source_code/tools/riscv/rocket-tools/riscv64-unknown-elf/bin/" setting from $PATH before building.

This log output suggests otherwise and it looks like the native x86_64 compiler is incorrectly invoking the RISC-V assembler:

> ```
> /bin/sh ../.././gcc/gcc/mkconfig.sh bconfig.h
> g++ -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing [...]

- https://stackoverflow.com/questions/42205916/as-unrecognized-option-64

TommyMurphyTM1234, 2024-08-10:
OK - so the x86 assembler does take a `--64` option:

- https://sourceware.org/binutils/docs/as.html#:~:text=Target%20i386%20options%3A%0A%20%20%20%5B%2D%2D32%7C%2D%2Dx32%7C%2D%2D64%5D%20%5B%2Dn%5D%0A%20%20%20%5B%2Dmarch%3DCPU%5B%2BEXTENSION%E2%80%A6%5D%5D%20%5B%2Dmtune%3DCPU%5D

So it's almost certainly a case of your setup having the RISC-V assembler on the `PATH` and the x86_64 `g++` invoking that instead of the native x86_64 one as originally suggested by @kito-cheng above.

Even if you have removed the RISC-V tools from your `PATH` when you invoke the toolchain build it may be that subsequent shells forked by the `Makefile` are using the original `PATH` pointing to those tools e.g. from some configuration file such as the shell specific `rc` file or `/etc/profile` etc.

You probably need to clean up your setup/environment so that you are certain that you have eliminated the RISC-V tools from the `PATH`. One way to confirm this hypothesis might be to create a new user account, log into that and see if the tools build OK there. Assuming, of course, that any `PATH` pointing to the RISC-V tools isn't a global configuration setting applying to all user accounts.

EventScheduler, 2024-08-14:
Yes, you're right, there are some RISCV tools in "/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/bin:/home/radl/hdd/leiyu/source_code/breker/from_felix/trek5-2.1.4-GCC6_el7/examples/tutorials/apps/coherency/bin" directories, and the "--64" issue disappered when I removed these two directories from PATH.
Thank you for your kindly help!

closed 2024-08-14 by EventScheduler