riscv-collab/riscv-gnu-toolchain #1530: Question: why does a `make CXXFLAGS_FOR_TARGET_EXTRA=-v` build fail?

TommyMurphyTM1234, 2024-08-11:
I've cloned and configured the latest sources:

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --disable-gdb
```

And then I have tried four different builds (doing `make distclean && rm -rf installed-tools` before each for a clean build):

1. `make 2>&1 | tee build.log`: [`build.log`](https://github.com/user-attachments/files/16573391/build.log)
2. `make ASFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build-extra-asflags.log`: [`build-extra-asflags.log`](https://github.com/user-attachments/files/16573394/build-extra-asflags.log)
3. `make CFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build-extra-cflags.log`: [`build-extra-cflags.log`](https://github.com/user-attachments/files/16573395/build-extra-cflags.log)
4. `make CXXFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build-extra-cxxflags.log`: [`build-extra-cxxflags.log`](https://github.com/user-attachments/files/16573401/build-extra-cxxflags.log)

The first three complete successfully (and give verbose output for the relevant tools) but the third fails:
```
 /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/cc1plus -quiet -nostdinc++ -v -I /home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/../libgcc -I /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf -I /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include -I /home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/libsupc++ -imultilib . -iprefix /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/13.2.0/ -isysroot /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../riscv64-unknown-elf -isystem /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/include -isystem /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/include-fixed -isystem /home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include -isystem /home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/sys-include ../../../../.././gcc/libstdc++-v3/src/c++11/cxx11-ios_failure.cc -quiet -dumpbase cxx11-ios_failure.cc -dumpbase-ext .cc -mcmodel=medlow -mtune=rocket -march=rv64imafdc_zicsr -mabi=lp64d -misa-spec=20191213 -march=rv64imafdc_zicsr -gno-as-loc-support -Os -Wall -Wextra -Wwrite-strings -Wcast-qual -Wabi=2 -std=gnu++11 -version -fno-implicit-templates -fdiagnostics-show-location=once -ffunction-sections -fdata-sections -frandom-seed=cxx11-ios_failure-lt.s -o cxx11-ios_failure.s
GNU C++11 (gc891d8dc23e) version 13.2.0 (riscv64-unknown-elf)
        compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/sys-include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/13.2.0/include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/13.2.0/include-fixed"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../riscv64-unknown-elf/usr/local/include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/include-fixed"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include"
ignoring nonexistent directory "/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/../libgcc
 /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf
 /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include
 /home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/libsupc++
 /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/include
 /home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/include-fixed
 /home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include
End of search list.
Compiler executable checksum: d5806b54b201f879501338c7bbc69a9b
COMPILER_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
LIBRARY_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-shared-libgcc' '-B' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc' '-nostdinc++' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/.libs' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/libsupc++/.libs' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/sys-include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/../libgcc' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/libsupc++' '-std=gnu++11' '-fno-implicit-templates' '-Wall' '-Wextra' '-Wwrite-strings' '-Wcast-qual' '-Wabi=2' '-fdiagnostics-show-location=once' '-ffunction-sections' '-fdata-sections' '-frandom-seed=cxx11-ios_failure-lt.s' '-Os' '-v' '-mcmodel=medlow' '-gno-as-loc-support' '-S' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr'
libtool: compile: mv -f "cxx11-ios_failure.o" "tmp-cxx11-ios_failure-lt.o"
mv: cannot stat 'cxx11-ios_failure.o': No such file or directory
make[6]: *** [Makefile:876: cxx11-ios_failure-lt.s] Error 1
make[6]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/c++11'
make[5]: *** [Makefile:792: all-recursive] Error 1
make[5]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src'
make[4]: *** [Makefile:576: all-recursive] Error 1
make[4]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[3]: *** [Makefile:501: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[2]: *** [Makefile:12289: all-target-libstdc++-v3] Error 2
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[1]: *** [Makefile:1035: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make: *** [Makefile:742: stamps/build-gcc-newlib-stage2] Error 2
```
Anybody got any idea why?
[... cut ...] mafdc_zicsr -gno-as-loc-support -Os -Wall -Wextra -Wwrite-strings -Wcast-qual -Wabi=2 -std=gnu++11 -version -fno-implicit-templates -fdiagnostics-show-location=once -ffunction-sections -fdata-sections -frandom-seed=cxx11-ios_failure-lt.s -o cxx11-ios_failure.s

...

libtool: compile: mv -f "cxx11-ios_failure.o" "tmp-cxx11-ios_failure-lt.o"
mv: cannot stat 'cxx11-ios_failure.o': No such file or directory
```
I.e. the output file is `cxx11-ios_failure.s` but it tries to copy `cxx11-ios_failure.o`.

cmuellner, 2024-08-26:
Pointer for further analysis (I have not reproduced the issue yet): the relevant Makefile is `libstdc++-v3/src/c++11/Makefile.am` (GCC).

Things to analyse:
* is the generated `Makefile` equal in both cases?
* is `ENABLE_DUAL_ABI` being set in both cases?

Also, does this happen with `upstream/master`?

TommyMurphyTM1234, 2024-08-26:
Thanks for the feedback @cmuellner.

> Pointer for further analysis (I have not reproduced the issue yet): the relevant Makefile is `libstdc++-v3/src/c++11/Makefile.am` (GCC).
> 
> Things to analyse:
> 
> * is the generated `Makefile` equal in both cases?

Do you mean the two cases of with and without `CXXFLAGS_FOR_TARGET_EXTRA=-v`?

> * is `ENABLE_DUAL_ABI` being set in both cases?

Ditto?

> Also, does this happen with `upstream/master`?

Do you mean when using something like the following?

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master --depth=1
./configure ... --with-gcc-src=`pwd`/gcc-master
```

I can look into these issues soon.

TommyMurphyTM1234, 2024-08-26:
FWIW:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --disable-gdb 

make 2>&1 | tee build.log
cp build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/Makefile Makefile-without-v

make distclean
rm -rf installed-tools
make CXXFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build.log
...
Compiler executable checksum: d5806b54b201f879501338c7bbc69a9b
COMPILER_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
LIBRARY_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-shared-libgcc' '-B' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc' '-nostdinc++' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/.libs' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/libsupc++/.libs' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/sys-include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/../libgcc' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc/libstdc++-v3/libsupc++' '-std=gnu++11' '-fno-implicit-templates' '-Wall' '-Wextra' '-Wwrite-strings' '-Wcast-qual' '-Wabi=2' '-fdiagnostics-show-location=once' '-ffunction-sections' '-fdata-sections' '-frandom-seed=cxx11-ios_failure-lt.s' '-Os' '-v' '-mcmodel=medlow' '-gno-as-loc-support' '-S' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr'
libtool: compile: mv -f "cxx11-ios_failure.o" "tmp-cxx11-ios_failure-lt.o"
mv: cannot stat 'cxx11-ios_failure.o': No such file or directory
make[6]: *** [Makefile:876: cxx11-ios_failure-lt.s] Error 1
make[6]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/c++11'
make[5]: *** [Makefile:792: all-recursive] Error 1
make[5]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src'
make[4]: *** [Makefile:576: all-recursive] Error 1
make[4]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[3]: *** [Makefile:501: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[2]: *** [Makefile:12289: all-target-libstdc++-v3] Error 2
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[1]: *** [Makefile:1035: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make: *** [Makefile:742: stamps/build-gcc-newlib-stage2] Error 2

cp build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/Makefile Makefile-with-v

diff Makefile-with-v Makefile-without-v
228c228
< CXXFLAGS = -Os -v   -mcmodel=medlow
---
> CXXFLAGS = -Os    -mcmodel=medlow
```

TommyMurphyTM1234, 2024-08-26:
I tried with the latest upstream `master` branch GCC sources but got the same failure:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master --depth=1
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-gcc-src=`pwd`/gcc-master
make CXXFLAGS_FOR_TARGET_EXTRA=-v 2>&1 | tee build.log
...
Compiler executable checksum: 5d95cf1bb896048a8e56b563a3db2113
COMPILER_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
LIBRARY_PATH=/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-shared-libgcc' '-B' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc' '-nostdinc++' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/.libs' '-L/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/libsupc++/.libs' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/' '-B' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/lib/' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/include' '-isystem' '/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/sys-include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc-master/libstdc++-v3/../libgcc' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/riscv64-unknown-elf' '-I' '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include' '-I' '/home/user/Downloads/riscv-gnu-toolchain/gcc-master/libstdc++-v3/libsupc++' '-std=gnu++11' '-fno-implicit-templates' '-Wall' '-Wextra' '-Wwrite-strings' '-Wcast-qual' '-Wabi=2' '-fdiagnostics-show-location=once' '-ffunction-sections' '-fdata-sections' '-frandom-seed=cxx11-ios_failure-lt.s' '-Os' '-v' '-mcmodel=medlow' '-gno-as-loc-support' '-S' '-mtune=rocket' '-march=rv64imafdc_zicsr_zaamo_zalrsc' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr_zaamo_zalrsc_zca_zcd'
libtool: compile: mv -f "cxx11-ios_failure.o" "tmp-cxx11-ios_failure-lt.o"
mv: cannot stat 'cxx11-ios_failure.o': No such file or directory
make[6]: *** [Makefile:899: cxx11-ios_failure-lt.s] Error 1
make[6]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src/c++11'
make[5]: *** [Makefile:797: all-recursive] Error 1
make[5]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/src'
make[4]: *** [Makefile:577: all-recursive] Error 1
make[4]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[3]: *** [Makefile:502: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3'
make[2]: *** [Makefile:12244: all-target-libstdc++-v3] Error 2
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[1]: *** [Makefile:1055: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make: *** [Makefile:742: stamps/build-gcc-newlib-stage2] Error 2
```

TommyMurphyTM1234, 2024-08-26:
To summarise:

> Things to analyse:
> 
> * is the generated `Makefile` equal in both cases?

Yes, apart from `CXXFLAGS = -Os` in one case and `CXXFLAGS = -Os -v` in the other.

> * is `ENABLE_DUAL_ABI` being set in both cases?

It's not set in either case.

> Also, does this happen with `upstream/master`?

Yes - same error.