riscv-collab/riscv-gnu-toolchain #1532: RISC-V toolchain with LLVM compiler

RRArinc, 2024-08-14:
Hi, Good Morning.
I build new custom instruction for LLVM/Clang compiler. I am able to build sample program comprising of custom instructions successfully. However when i use llvm-objdump i see all custom instructions as **<unknown>** in objdump file.

   1019a: 03 27 04 fe  	lw	a4, -0x20(s0)
   1019e: ab 86 e6 00  	<unknown>
   101a2: 23 2e d4 fc  	sw	a3, -0x24(s0)
   101a6: 83 26 44 fe  	lw	a3, -0x1c(s0)
   101aa: 03 27 04 fe  	lw	a4, -0x20(s0)
   101ae: ab c6 e6 00  	<unknown>
   101b2: 23 2e d4 fc  	sw	a3, -0x24(s0)
   101b6: 83 26 44 fe  	lw	a3, -0x1c(s0)
   101ba: 03 27 04 fe  	lw	a4, -0x20(s0)
   101be: ab 86 e6 02  	<unknown> 

May i know what is happening here.

Thanks
Ravindra.

TommyMurphyTM1234, 2024-08-14:
Please provide the full details for a reproducible test case.
For example, details of how you modified, configured and built the toolchain, the code that you're compiling, the command lines that you're using to compile and disassemble it etc.

TommyMurphyTM1234, 2024-08-14:
FWIW this post seems to explain how to extend LLVM/Clang with custom instructions and have `llvm-objdump` disassemble them. Although I haven't actually worked through the example myself...

- https://fprox.substack.com/p/adding-a-new-risc-v-instruction-to

TommyMurphyTM1234, 2024-08-16:
Any update @RRArinc?

RRArinc, 2024-08-22:
Hi, Thanks for sharing the link, Yes, one of the instruction was mismatched with same opcode. Hence the instruction was not generating in objdump code.
Now the issue is resolved.

Thanks.

closed 2024-08-22 by RRArinc