riscv-collab/riscv-gnu-toolchain #1535: Custom Sine Instruction

amithmath, 2024-08-19:
This thread is continuation of https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1526  I am trying to  add custom floating point sine instruction.

Actually I was following this video: https://www.youtube.com/watch?v=RT0GqJySnBc to add custom float instructions to risc-v, as per the video one has to modify riscv-opc.c, riscv-opc.h and riscv.md. I added fsin instruction by defining match, mask in the file riscv-opc.h, fsin.s to riscv-opc.c, I did define instruction in riscv-gcc/gcc/config/riscv/riscv.md as shown below

(define_insn "sin<mode>2"
  [(set (match_operand:ANYF            0 "register_operand" "=f")
        (sin:ANYF (match_operand:ANYF 1 "register_operand" " f")))]
  "TARGET_HARD_FLOAT && TARGET_FDIV"
{
    return "fsin.<fmt>\t%0,%1";
}
  [(set_attr "type" "fsin")
   (set_attr "mode" "<UNITMODE>")])

When I build tool chain with make -j 16  CFLAGS_FOR_TARGET_EXTRA='-fno-common',  I am getting error at: set_attr "type" "fsin" unknown value `fsin' for attribute `type' It seems to me I have to add 'fsin' attribute somewhere but I don't know where. Please guide me.

TommyMurphyTM1234, 2024-08-19:
You probably need to familiarise yourself with the GCC `md` Machine Description file specifications and capabilities, in particular the `set_attr type ...` option.

- https://gcc.gnu.org/onlinedocs/gccint/Machine-Desc.html
- https://gcc.gnu.org/onlinedocs/gccint/Attr-Example.html

TommyMurphyTM1234, 2024-08-19:
>  I did define instruction in riscv-gcc/gcc/config/riscv/riscv.md as shown below

The latest `riscv-gnu-toolchain` doesn't use the [now archived `riscv-gcc`](https://github.com/riscvarchive/riscv-gcc) sources so this may suggest that you are also using out of date sources?

TommyMurphyTM1234, 2024-08-19:
FWIW it looks like the `type` attribute values are defined here in `riscv.md`:

- https://github.com/gcc-mirror/gcc/blob/8871489c5162067c72a9b9ab05fe2179560e9986/gcc/config/riscv/riscv.md#L476

TommyMurphyTM1234, 2024-08-19:
Looks like you're calling a function named `sin()` but the linker can't resolve the symbol reference. As such it seems like a simple link error.

It's not really clear what you're trying to do. If the intention of your toolchain changes are to add a custom instruction and have that generated when a call is made to `sin()` then I guess that you haven't done things properly to have this happen. 

As I said before, this isn't a `riscv-gnu-toolchain` issue but a more general question about how to customise the GCC toolchain. As such, it's really outside the scope of this repo and you might be better advised to seek assistant elsewhere - e.g. in an appropriate GCC forum/mailing list, StackOverflow, Reddit etc.

amithmath, 2024-08-22:
Now, I don't  get following error:

sin_example.c:(.text+0x34): undefined reference to `sin'

But when I generate assembly file with following command:

riscv32-unknown-elf-gcc -c -S -march=rv32imaf  sin_example.c

I am seeing 'call sin' in assembly code though I am not using -lm flag.  I am unable to figure out what I may be missing

TommyMurphyTM1234, 2024-08-22:
> I don't get following error:
> 
> sin_example.c:(.text+0x34): undefined reference to `sin'
> 
> When I generate assembly file with following command:
> 
> riscv32-unknown-elf-gcc -c -S -march=rv32imaf sin_example.c
> 
> I am seeing 'call sin' in assembly code though I am not using -lm flag. I am unable to figure out what I may be missing

You're not getting the link error in this case because you're not linking anything! You're just converting the C source file to assembly and/or unlinked object file. I'm not sure how it works when both `-c` and `-S` are used at the same time: 

- https://gcc.gnu.org/onlinedocs/gcc/Overall-Options.html

If your command line produces an object file then the call to `sin()` presumably remains an unresolved external symbol in this case waiting to be resolved at link time. This is pretty basic toolchain usage stuff and nothing really to do with your substantive issue.

amithmath, 2024-08-22:
Yes you are right, I am not linking -lm, if I do for example:

riscv32-unknown-elf-gcc -c   -march=rv32imaf  sin_example.c
riscv32-unknown-elf-gcc -o sin_example sin_example.o    -lm 

undefined reference to sin( ) will be resolved; but since I am adding custom sin ( ) instruction, it seems to me I have to use certain switch/flag while I am building compiler and compiling code, so that compiler should not look for -lm flag. For example, when I use 'fno-common' flag while compiling, the compiler will emit fdiv.s and fsqrt.s instructions otherwise I have to link using -lm flag. Similarly, for sin, cos, exp(x), I don't know which flag to use so that the compiler can emit these instructions....

TommyMurphyTM1234, 2024-08-22:
As I said before, if your expectation is that the changes that you have made to the toolchain mean that your custom instruction will be generated when you call `sin()` but this is not happening, then this is not a `riscv-gnu-toolchain` (or even a RISC-V specific) issue and is way outside the scope of this repository and you might be better served seeking assistance/advice elsewhere such as in the types of places that I already mentioned above.

TommyMurphyTM1234, 2024-08-22:
> For example, when I use 'fno-common' flag while compiling, the compiler will emit fdiv.s and fsqrt.s instructions otherwise I have to link using -lm flag. Similarly, for sin, cos, exp(x)

`-fno-common` has nothing directly to do with math functions/instruction generation. 

- https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html

amithmath, 2024-08-22:
Sorry, I meant -mno-fdiv  https://gcc.gnu.org/onlinedocs/gcc-7.1.0/gcc/RISC-V-Options.html

TommyMurphyTM1234, 2024-08-22:
> Sorry, I meant -mno-fdiv https://gcc.gnu.org/onlinedocs/gcc-7.1.0/gcc/RISC-V-Options.html

So this is a GCC flag that specifically and directly controls code generation in the case of the hardware floating point instructions. If you want something similar for your custom instruction(s?) then presumably you'll have to add it yourself. Again, way outside the scope of this repo.

amithmath, 2024-08-23:
I did try small example for floating div and floating square root where risc-v supports both hardware instructions,

when  I compile the above code with just change in sin () with:  a= angle1/angle1;

riscv32-unknown-elf-gcc -c -S     -march=rv32imaf  sin_example.c

it generates following floating point division instruction in .s file:

fdiv.s  fa5,fa5,fa5

when I use -mno-fdiv switch during compilation, 

it will generate call    __divsf3 soft routine which is Okay

However, if I change the above   a= sqrt(angle1);  

and compile the code without -mno-fpdiv switch it will generate the following warning:

 implicit declaration of function 'sqrt' [-Wimplicit-function-declaration]
    9 |     a= sqrt(angle1); //angle1/angle1; //sqrtf(a); //sin(angle1);
      |        ^~~~
sin_example.c:9:8: warning: incompatible implicit declaration of built-in function 'sqrt'
sin_example.c:3:1: note: include '<math.h>' or provide a declaration of 'sqrt'
    2 | #include <stdio.h>
  +++ |+#include <math.h>

in .s file it will generate, call    sqrt function, which it should not generate this call; instead it should generate fsqrt.s instruction

I am using gcc 10.2.0

As per this link: https://gcc.gnu.org/onlinedocs/gcc-10.2.0/gcc/RISC-V-Options.html#RISC-V-Options one can use -mno-fdiv this switch not use hardware fp div and fp sqrt instructions. 

In my view, it might  be the issue with riscv-gnu-toolchain

TommyMurphyTM1234, 2024-08-23:
> in .s file it will generate, call sqrt function, which it should not generate this call; instead it should generate fsqrt.s instruction
> 
> As per this link: https://gcc.gnu.org/onlinedocs/gcc-10.2.0/gcc/RISC-V-Options.html#RISC-V-Options one can use -mno-fdiv this switch not use hardware fp div and fp sqrt instructions.

Maybe try it with `-fno-math-error`?

- https://github.com/riscvarchive/riscv-newlib/issues/40#issuecomment-436873798

Using `-mno-div` isn't going to magically replace calls to `sqrt()` with a single `fsqrt.s` instruction in all cases since the function call may imply other necessary work that needs to be done (e.g. error handling or other ancillary work). This is a different scenario to directly replacing a high level (e.g. C) code division operation with an `fdiv.s` instruction.

> In my view, it might be the issue with riscv-gnu-toolchain

Even if this is so, what does it have to do with your original problem? It seems like a completely different issue? In any case, if you do think that there's a bug here then it should be logged upstream in the GCC bug tracker since that's where ongoing GCC RISC-V target development and maintenance is done these days, not here. 

- https://gcc.gnu.org/bugs/

> I am using gcc 10.2.0

This is very old. `riscv-gnu-toolchain` uses GCC 13.2 now and will soon move to 14.2.

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

 Using such an old version of GCC given the pace of change to the GCC RISC-V support in recent years isn't a great idea.

TommyMurphyTM1234, 2024-08-23:
FWIW

> riscv32-unknown-elf-gcc -c -S -march=rv32imaf sin_example.c

Note that `-c` is redundant/ignored here since `-S` overrides it - `-c` is for compiling to object file but `-S` is for converting to assembler and the latter takes precedence so that only a `*.s` file is output and not also a `*.o` file:

Using the latest CI release build:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases
  - https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.08.06
    - https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz

```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
tar xzvf riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
cd riscv/bin
```

Create `test.c`:

```
// test.c
#include <math.h>

void f(void)
{
        float a, angle1, angle2;
        a = angle1/angle1;
        a = sqrtf(angle1);
}
```

Now compile with `-O -fno-math-errno` so that the call to `sqrtf()` gets implemented as a simple `fsqrt.s` instruction:

```
./riscv64-unknown-elf-gcc -S -march=rv32imaf -mabi=ilp32f -O -fno-math-errno test.c

cat test.s
        .file   "test.c"
        .option nopic
        .attribute arch, "rv32i2p1_m2p0_a2p1_f2p2_zicsr2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  2
        .globl  f
        .type   f, @function
f:
        addi    sp,sp,-16
        flw     fa5,8(sp)
        flw     fa4,8(sp)
        fdiv.s  fa5,fa5,fa4
        fsw     fa5,12(sp)
        flw     fa5,8(sp)
        fsqrt.s fa5,fa5
        fsw     fa5,12(sp)
        addi    sp,sp,16
        jr      ra
        .size   f, .-f
        .ident  "GCC: () 13.2.0"
```

amithmath, 2024-08-23:
> Even if this is so, what does it have to do with your original problem? It seems like a completely different issue? In any case, >if you do think that there's a bug here then it should be logged [...]

I used sqrt and div as an example since risc-v offers both soft  routines and hardware instructions and learn how it is implemented , using this as a base for custom instruction for sin ( ) implementation. 

I did try with above compilation procedure:

riscv-install/bin/riscv32-unknown-elf-gcc  -S -march=rv32imaf -mabi=ilp32f -O -fno-math-errno   sin_example.c

where I calling a =sqrtf(angle1); 

I am getting following asm code

//sin_example.s

  .file   "sin_example.c"
        .option nopic
        .attribute arch, "rv32i2p0_m2p0_a2p0_f2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  2
        .globl  main
        .type   main, @function
main:
        li      a0,0
        ret
        .size   main, .-main
        .ident  "GCC: (GNU) 10.2.0"

It might be bug in gcc 10.2.0

TommyMurphyTM1234, 2024-08-23:
> It might be bug in gcc 10.2.0

I don't think so.

Using xPack GNU RISC-V Embedded GCC v10.2.0-1.2:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/
    - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2021/11/11/riscv-none-embed-gcc-v10-2-0-1-2-released/
      - https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases/tag/v10.2.0-1.2/
        - https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases/download/v10.2.0-1.2/xpack-riscv-none-embed-gcc-10.2.0-1.2-linux-x64.tar.gz

```
wget https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases/download/v10.2.0-1.2/xpack-riscv-none-embed-gcc-10.2.0-1.2-linux-x64.tar.gz
tar xzvf xpack-riscv-none-embed-gcc-10.2.0-1.2-linux-x64.tar.gz
cd xpack-riscv-none-embed-gcc-10.2.0-1.2/bin/
```

Create `test.c`:

```
// test.c
#include <math.h>

void f(void)
{
        float a, angle1, angle2;
        a = angle1/angle1;
        a = sqrtf(angle1);
}
```

Then process it:

```
./riscv-none-embed-gcc -S -march=rv32imaf -mabi=ilp32f -O -fno-math-errno test.c

cat test.s
        .file   "test.c"
        .option nopic
        .text
        .align  2
        .globl  f
        .type   f, @function
f:
        addi    sp,sp,-16
        flw     fa5,8(sp)
        flw     fa4,8(sp)
        fdiv.s  fa5,fa5,fa4
        fsw     fa5,12(sp)
        flw     fa5,8(sp)
        fsqrt.s fa5,fa5
        fsw     fa5,12(sp)
        addi    sp,sp,16
        jr      ra
        .size   f, .-f
        .ident  "GCC: (xPack GNU RISC-V Embedded GCC x86_64) 10.2.0"

```

Even if you do find a suspected bug in GCC then you need to log it upstream and not here.

amithmath, 2024-08-24:
>There's no mention at all of any square root logic in that output file - did you post the correct output?

Yes, even I built new compiler version 13.2.0

// test.c
#include <math.h>

void f(void)
{
        float a, angle1, angle2;
        a = angle1/angle1;
        a = sqrtf(angle1);
}

and then compiled again with following command:

riscv-install/bin/riscv32-unknown-elf-gcc  -S -march=rv32gc_zfh -mabi=ilp32d -O  -fno-math-errno test.c

I am using -march=rv32gc_zfh -mabi=ilp32d options because I built the compiler with these options, and following is the test.s

.file   "test.c"
        .option nopic
        .attribute arch, "rv32i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0_zifencei2p0_zfh1p0_zfhmin1p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  f
        .type   f, @function
f:
        ret
        .size   f, .-f
        .ident  "GCC: (g) 13.2.0"
                                    
Even I tried with this command:

riscv-install/bin/riscv32-unknown-elf-gcc -S -march=rv32imaf -mabi=ilp32f  -O  -fno-math-errno test.c

following is the output:

.file   "test.c"
        .option nopic
        .attribute arch, "rv32i2p1_m2p0_a2p1_f2p2_zicsr2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  2
        .globl  f
        .type   f, @function
f:
        ret
        .size   f, .-f
        .ident  "GCC: (g) 13.2.0"

I am using CentOS 7.9

TommyMurphyTM1234, 2024-08-24:
If you follow the exact steps that I previously outlined then you will see that it is possible to have the compiler output `fsqrt.s` for a call to `sinf()` using both the latest 13.2.0 and the older 10.2.0 GCC toolchains.

At this stage I'm not really clear what your issue or issues are. It started with problems in modifying the toolchain to add custom instructions (and have them automatically generated when certain intermediate code patterns were recognised?) but now seems to have veered off into issues about hardware floating point instruction generation.

In any case, no issue has been identified with the `riscv-gnu-toolchain` so far and if there are any suspected or actual issues with GCC code generation then they should be discussed/reported upstream.

amithmath, 2024-08-24:
> At this stage I'm not really clear what your issue or issues are. It started with problems in modifying the toolchain to add >custom instructions (and have them automatically generated when certain [...]

Well, the still issue remains same adding custom sine instruction to toolchain;  since RISC-V supports hardware fp div and fp sqrt, I want to see how these instructions are generated  in .s file and once I am successful in generating these instructions, I follow the same steps to adding custom sine instruction.

TommyMurphyTM1234, 2024-08-24:
> Well, the still issue remains same adding custom sine instruction to toolchain; since RISC-V supports hardware fp div and fp sqrt, I want to see how these instructions are generated in .s file and [...]

I've shown you how to get the compiler to generate `fdiv.s` and `fsqrt.s` instructions using the 13.2.0 and 10.2.0 GCC toolchains.

> I guess the issue might be the OS I am using (CentOS 7.9), in your case, fsqrt.s instruction generated in Ubuntu

The host OS should not affect the code generation of the RISC-V cross compiler.

TommyMurphyTM1234, 2024-08-24:
> Yes, even I built new compiler version 13.2.0
>
> // test.c #include <math.h>
>
> void f(void) { float a, angle1, angle2; a = angle1/angle1; a = sqrtf(angle1); }
>
> and then compiled again with following [...]

The problem is that `-O` optimises unnecessary code away.

Making the variable definitions `volatile`:

```
// test.c
#include <math.h>

void f(void)
{
volatile float a, angle1, angle2;
a = angle1/angle1;
a = sqrtf(angle1);
}
```

ensures that the code does not get optimised away and results in the following:

```
./riscv64-unknown-elf-gcc -S -march=rv32gc_zfh -mabi=ilp32d -O -fno-math-errno test.c

cat test.s

        .file   "test.c"
        .option nopic
        .attribute arch, "rv32i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0_zifencei2p0_zfh1p0_zfhmin1p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  f
        .type   f, @function
f:
        addi    sp,sp,-16
        flw     fa5,8(sp)
        flw     fa4,8(sp)
        fdiv.s  fa5,fa5,fa4
        fsw     fa5,12(sp)
        flw     fa5,8(sp)
        fsqrt.s fa5,fa5
        fsw     fa5,12(sp)
        addi    sp,sp,16
        jr      ra
        .size   f, .-f
        .ident  "GCC: () 13.2.0"
```

amithmath, 2024-08-25:
Thanks! Now the issue has been resolved. I would like to keep this thread open for a while, in case any issue arises.