riscv-collab/riscv-gnu-toolchain #1536: Failure to build on recent versions of libc++ / clang

lukawarren, 2024-08-19:
I am attempting to build the newlib toolchain as per the README, with the default toolchain on macOS 14.6.1 (Sonoma):
```
$ gcc --version
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

I have attempted to build the project as so:
```
./configure --prefix=/opt/riscv
make
```

Unfortunately,  various errors occur, all similar to this:
```
In file included from ../.././gcc/gcc/config/riscv/riscv-selftests.cc:23:
In file included from ../.././gcc/gcc/system.h:227:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/map:2529:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/functional:526:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__functional/boyer_moore_searcher.h:27:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:321:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__format/formatter_bool.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__format/formatter_integral.h:32:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/locale:202:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:550:5: error: '__abi_tag__' attribute only applies to structs, variables, functions, and namespaces
    _LIBCPP_INLINE_VISIBILITY
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__config:891:37: note: expanded from macro '_LIBCPP_INLINE_VISIBILITY'
#  define _LIBCPP_INLINE_VISIBILITY _LIBCPP_HIDE_FROM_ABI
                                    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__config:870:26: note: expanded from macro '_LIBCPP_HIDE_FROM_ABI'
          __attribute__((__abi_tag__(_LIBCPP_TOSTRING(_LIBCPP_ODR_SIGNATURE))))
```
[full error output.txt](https://github.com/user-attachments/files/16658961/error.output.txt)

I believe this bug is already well-documented on GCC's bugzilla [here](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111632) and has already been fixed in mainline GCC, though I could be mistaken.

TommyMurphyTM1234, 2024-08-19:
> I believe this bug is already well-documented on GCC's bugzilla here and has already been fixed in mainline GCC, though I could be mistaken.

To check this try building the toolchain using the latest upstream GCC sources from the `master` branch or `releases/gcc-14` branch:

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain 
git clone https://github.com/gcc-mirror/gcc gcc-latest --depth=1
# or...
# git clone https://github.com/gcc-mirror/gcc gcc-latest -b releases/gcc-14 --depth=1
./configure ... --with-gcc-src=`pwd`/gcc-latest
make
```

TommyMurphyTM1234, 2024-08-19:
Actually, according to this thread:

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111632

the fix was backported to various minor versions of GCC 13 so maybe your problem is different and not fixed by this?

lukawarren, 2024-08-19:
I can't speak for GCC 13 but I've just compiled the toolchain with `git clone https://github.com/gcc-mirror/gcc gcc-latest -b releases/gcc-14 --depth=1` (as described above) and it compiled successfully.
```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (gaf97b5eb0) 14.2.1 20240819
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
In other words, that fixed it. I presume then that once #1531 is merged the above will no longer be necessary?
Do let me know if you'd like me to try building from GCC's master or indeed GCC 13 instead.

TommyMurphyTM1234, 2024-08-19:
> I can't speak for GCC 13 but I've just compiled the toolchain with `git clone https://github.com/gcc-mirror/gcc gcc-latest -b releases/gcc-14 --depth=1` (as described above) and it compiled [...]

Thanks - that's interesting and useful feedback.

> I presume then that once #1531 is merged the above will no longer be necessary? 

Correct.

> Do let me know if you'd like me to try building from GCC's master or indeed GCC 13 instead.

So it looks like the fix was backported to GCC 13.3 (and maybe also 13.2.1 but I can'tt seem to find a branch/tag for that version):

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111632#c31

`riscv-gnu-toolchain` currently uses GCC 13.2.0 for its GCC submodule:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1341

If you had the time maybe the following would be worth trying...

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-13.3.0 gcc-13.3.0 --depth=1
./configure ... --with-gcc-src=`pwd`/gcc-13.3.0 
make
```

But if you don't have time don't worry about it given that the eventual imminent (hopefully!) merging of #1531 will address the issue.

lukawarren, 2024-08-19:
> If you had the time maybe the following would be worth trying...
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> git clone https://github.com/gcc-mirror/gcc -b releases/gcc-13.3.0 gcc-13.3.0 --depth=1
> ./configure ... --with-gcc-src=`pwd`/gcc-13.3.0 
> make
> ```

I can confirm that GCC 13.3 works too, after building it in the described manner:
```
$ ./riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (gb71f1de6e) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Thanks for all your help.

TommyMurphyTM1234, 2024-08-19:
> I can confirm that GCC 13.3 works too, after building it in the described manner:
>
> ```
> $ ./riscv64-unknown-elf-gcc --version
> riscv64-unknown-elf-gcc (gb71f1de6e) 13.3.0
> Copyright (C) 2023 Free [...]

That's great - thanks for checking.
So it looks like GCC 13.3 or later includes the fix and the PR to bump `riscv-gnu-toolchain` to GCC 14.2 will integrate the fix here and obviate the need to build using out of tree GCC sources from upstream.
👍 

I'll close the issue since it's now understood.

closed 2024-08-19 by TommyMurphyTM1234