riscv-collab/riscv-gnu-toolchain #1537: The addv standard instruction pattern generates an additional instruction.

leidian977, 2024-08-22:
In riscv.md, the code for the addv&lt;mode&gt;4 instruction pattern when handling TARGET_64BIT and SImode is as follows:
```
{
      rtx t3 = gen_reg_rtx (DImode);
      rtx t4 = gen_reg_rtx (DImode);
      rtx t5 = gen_reg_rtx (DImode);
      rtx t6 = gen_reg_rtx (DImode);

      emit_insn (gen_addsi3 (operands[0], operands[1], operands[2]));
      if (GET_CODE (operands[1]) != CONST_INT)
	emit_insn (gen_extend_insn (t4, operands[1], DImode, SImode, 0));
      else
	t4 = operands[1];
      if (GET_CODE (operands[2]) != CONST_INT)
	emit_insn (gen_extend_insn (t5, operands[2], DImode, SImode, 0));
      else
	t5 = operands[2];
      emit_insn (gen_adddi3 (t3, t4, t5));
      emit_insn (gen_extend_insn (t6, operands[0], DImode, SImode, 0));

      riscv_expand_conditional_branch (operands[3], NE, t6, t3);
    }
```
From the code, it can be observed that emit_insn(gen_addsi3(operands[0], operands[1], operands[2])) generates the addw instruction. This instruction performs an addition operation and sign-extends the lower 32 bits of the result to 64 bits before writing it to the destination register (operands[0]). However, emit_insn(gen_extend_insn(t6, operands[0], DImode, SImode, 0)) performs an additional sign-extension on operands[0], resulting in an extra sext.w extension instruction. The same issue exists with subv&lt;mode&gt;4.

TommyMurphyTM1234, 2024-08-22:
If this is an issue with code generation then it should be reported upstream in the GCC bug tracker since that's where maintenance (including of RISC-V target specific support) is done these days. 

- https://gcc.gnu.org/bugs/

No such maintenance is done in this repository any longer and not for a good while now.

closed 2024-08-23 by leidian977

TommyMurphyTM1234, 2024-08-23:
Seems to have been logged upstream: 

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=116466