riscv-collab/riscv-gnu-toolchain #1538: How to generate dynamic executable file using riscv-unknown-elf-gcc and clang compiler

RRArinc, 2024-08-22:
Hi,
I want to build dynamic executable using riscv32-unknown-elf-gcc and clang compiler.

But when i use the below commands to compile the sample code, generated executable is statically linked executable.

I used the below commands 

clang -o clang_exe -fpie -pie -Wl,--dynamic-linker=/lib64/ld-linux-x86-64.so.2 sample.c -lm
clang -o clang_exe -fpie -pie sample.c -lm
riscv32-unknwon-elf-gcc -o gcc_exe -fpie  sample.c -lm

Thanks

TommyMurphyTM1234, 2024-08-22:
Bare-metal toolchains generally don't support shared libraries because they can't presuppose that the bare-metal target platform provides the necessary infrastructure support for dynamic linking - in particular a dynamic link loader and OS environment in which that operates. You probably need to use the Linux toolchains.

TommyMurphyTM1234, 2024-08-22:
See also...

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1039

RRArinc, 2024-08-23:
Hi Murphy, Thanks for the response.
My goal here is to measure performance of riscv32-unknown-elf-gcc and riscv clang compiler,
So what kind of tools shall i use to carry out this activity. If i were able to create dynamic executable perf tool would have helped me. I also tries qemu-system-riscv32 but i am not able to get the expected results.
perf record -e cycles -a qemu-system-riscv32 -nographic -machine virt -kernel path/to/your_riscv_binary

Please can u sugegst any best tools and its usage to measure compiler performance.

Thanks in Advance

TommyMurphyTM1234, 2024-08-23:
> My goal here is to measure performance of riscv32-unknown-elf-gcc and riscv clang compiler,

What exactly do you mean by this?

1. Compare the runtime performance of the two toolchains?
2. Compare the performance of executables compiled using the two toolchains?
3. Something else?

Doesn't LLVM/Clang require GCC (or parts of this toolchain) in order to compile and link programs so you may not actually be strictly comparing LLVM/Clang with GCC?

> So what kind of tools shall i use to carry out this activity.

Sorry, I don't know. Assuming that you want to check #2 above then I would have expected that Spike or QEMU would provide the necessary capabilities.

> If i were able to create dynamic executable perf tool would have helped me. 

As I said before, if you need dynamic linking then why not use the RISC-V GCC and LLVM/Clang Linux toolchains?

> I also tries qemu-system-riscv32 but i am not able to get the expected results. 

What exactly do you mean by "the expected results"?

RRArinc, 2024-08-23:
Exactly, I am looking forward to evaluate the following:

1. Compare the runtime performance of the two toolchains?
2. Compare the runtime performance of executables compiled using the two toolchains?

**Sorry, I don't know. Assuming that you want to check https://github.com/riscv-collab/riscv-gnu-toolchain/issues/2 above then I would have expected that Spike or QEMU would provide the necessary capabilities.**

When i run command **qemu-system-riscv32 -nographic -machine virt -kernel path/to/your_riscv_binary** it outputs only 
real    0m1.234s
user    0m0.567s
sys     0m0.123s

how about time, cpu-clocks, conetxt switching between functions calls etc. Also i am interested to evaluate the performance of RISC-V GCC and Clang for bare metal RISC-V rv32imc architecture cores only

TommyMurphyTM1234, 2024-08-23:
> Exactly, I am looking forward to evaluate the following:
> 
> 1. Compare the runtime performance of the two toolchains?

For that you presumably just use the profiling features/capabilities of the host Linux platform? 

> how about time, cpu-clocks, conetxt switching between functions calls etc. Also i am interested to evaluate the performance of RISC-V GCC and Clang for bare metal RISC-V rv32imc architecture cores only

I don't know but, as I've mentioned, I presume that others have done this and their work might guide yours. 

Ultimately, this isn't really a `riscv-gnu-toolchain` issue.

TommyMurphyTM1234, 2024-08-23:
Maybe something like this would be useful? 

- https://github.com/mahendraVamshi/riscv-application-profiler

TommyMurphyTM1234, 2024-08-23:
This SIG might also be able to give you some advice. 

- https://lists.riscv.org/g/tech-announce/topic/announcing_the_formation_of_a/90476927

TommyMurphyTM1234, 2024-08-26:
Did any of those links help at all @RRArinc?
Can this issue be closed?

TommyMurphyTM1234, 2024-08-29:
No update so closing the issue - reopen with additional info if necessary.

closed 2024-08-29 by TommyMurphyTM1234