riscv-collab/riscv-gnu-toolchain #1540: clang 'omp.h' file not found

zqb-all, 2024-08-24:
when `omp.h` in 
```
./lib/gcc/riscv64-unknown-linux-gnu/13.2.0/include/omp.h
./include/omp.h
```

`riscv64-unknown-linux-gnu-gcc` can find it but `riscv64-unknown-linux-gnu-clang` will report ` fatal error: 'omp.h' file not found`

should we copy `omp.h` to  `lib/clang/18/include/` ?

TommyMurphyTM1234, 2024-08-24:
Should either or both of `LLVM_ENABLE_PROJECTS` and `LLVM_ENABLE_RUNTIMES` also specify `openmp`?

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/f42ee800388bce42095bdae720539853585aaa76/Makefile.in#L1004-L1005

TommyMurphyTM1234, 2024-08-25:
> Was there some step that I missed for clang to find `omp.h` ?
> 
> When build openmp by clang, `omp.h` will in `./lib/clang/<version>/include/omp.h`

I don't think that you have missed any step.
Perhaps it's the case that this commit is missing an additional step to copy the OpenMP header file(s) for LLVM/Clang - i.e. to `<prefix-dir>/lib/clang/<version>/include/omp.h`?

- https://github.com/riscv-collab/riscv-gnu-toolchain/commit/5aec7d73af4e1672c8327465bc0d562bee0bec95

Maybe @wangpc-pp (the creator of that commit) knows more about this?

FWIW after doing a build I get this:
```
# .../build-llvm-linux/openmp-shared/install_manifest.txt
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/libomp.so
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/include/omp.h
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/include/ompx.h
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/include/omp-tools.h
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/include/ompt.h
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/cmake/openmp/FindOpenMPTarget.cmake
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/libarcher.so
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/libarcher_static.a
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/include/ompt-multiplex.h
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/libompd.so
```
```

wangpc-pp, 2024-08-26:
Yeah, my fault. We probably should copy `omp.h` to `sysroot/usr/include`. Will fix it today.

PR #1541 (2024-08-26): Install OpenMP to sysroot
Or we can't find headers and libraries.

Fixes #1540.
closed 2024-08-26 by wangpc-pp