riscv-collab/riscv-gnu-toolchain #1546: Operands defination

SimSim150, 2024-08-29:
Where are the different operands like s and t being defined?

TommyMurphyTM1234, 2024-08-29:
Can you expand on/clarify your query please?
Maybe with some examples?
If you mean `s` and `t` ABI registers then they are described in the ABI specs:

- https://github.com/riscv-non-isa/riscv-elf-psabi-doc
  - https://github.com/riscv/riscv-isa-manual/releases/tag/20240411

[image]

If you are referring to something else then please clarify what it is.

SimSim150, 2024-08-29:
Hi thank you for the response. No i mean the operands like vd, r, t in the custom instructions in binutils/opcode/riscv-opc.c
This is what i meant by operands, here they are d,s and t. 
[image]

I know this is where some operand constraints are added but i mean the address space itself ie what bit positions will each operand will occupy yet. 
[image]

What im ultimately trying to do is add new operands into the toolchain myself

TommyMurphyTM1234, 2024-08-29:
This should be covered in the relevant GCC internals documentation and isn't RISC-V or `riscv-gnu-toolchain` specific stuff:

- https://gcc.gnu.org/onlinedocs/gccint/

TommyMurphyTM1234, 2024-08-29:
> This should be covered in the relevant GCC internals documentation and isn't RISC-V or `riscv-gnu-toolchain` specific stuff:
> 
> * https://gcc.gnu.org/onlinedocs/gccint/

Sory - such operand string specifiers may be target specific after all and I don't think that they are actually explained in the GCC internals documentation. And I can't find any comments in the RISC-V specific GCC/Binutils opcodes related files that explain them either...

FWIW here's part of what ChatGPT told me about some of the ones that I found - the usual caveats apply to the trustworthiness of AI generated text!

> # Meaning of Specific Operand Strings
> 
> Here’s what these operand strings typically mean in the context of RISC-V (and other architectures):
> 
> "d": Destination register. This operand represents the register where the result of the instruction will be stored.
> 
> "s": Source register. This represents a register whose value is used as an operand in the instruction.
> 
> "t": Second source register. Used in instructions that operate on two source registers.
> 
> "j": Jump target. This represents the target address for jump or branch instructions.
> 
> "o(s)": Offset with base register. This operand represents an address computed by adding an offset (o) to the value in a base register (s). Commonly used in load/store instructions.
> 
> "Ca": A constant value or an immediate operand. The exact nature might depend on the specific instruction, but it generally indicates an immediate value.
> 
> "1": This could be a literal value (e.g., the integer 1), often used in instructions that have a fixed value for an operand (e.g., for setting or clearing a specific bit).
> 
> "Wif(s)": This is a more complex operand string that implies a conditional encoding or specific behavior based on the operand type. It could mean something like "use a specific width (W) for the immediate if the operand is a source register (s)." > This string is used to handle special cases in instruction encoding.

TommyMurphyTM1234, 2024-09-03:
Did that help at all @SimSim150?

SimSim150, 2024-09-03:
> Did that help at all @SimSim150?

It gave me some context. Thanks!

Im working on adding them on my own, if i succeed ill definitely post the steps here

TommyMurphyTM1234, 2024-09-06:
Closing the issue.

closed 2024-09-06 by TommyMurphyTM1234