riscv-collab/riscv-gnu-toolchain #1547: Question: About the `la` instruction

joachimcao, 2024-09-02:
When I run `riscv32-unknown-elf-as -march=rv32i -mno-arch-attr load.s` with the content of load.s as below:

```asm
.section .text
.globl _start
_start:
  la a0, msg2
.section .rodata
  msg:
    .string "Hello World"
  msg2:
    .string "Hello World\n"
```

I expected the `la` would be translated into `auipc` and `addi` to get the right address. But when check by `riscv32-unknown-elf-objdump -D -t -r a.out`, the output isn't.

```asm
Disassembly of section .text:

00000000 <_start>:
   0:	00000517          	auipc	a0,0x0
			0: R_RISCV_PCREL_HI20	msg2
			0: R_RISCV_RELAX	*ABS*
   4:	00050513          	mv	a0,a0
			4: R_RISCV_PCREL_LO12_I	.L0 
			4: R_RISCV_RELAX	*ABS*

Disassembly of section .rodata:

00000000 <msg>:
   0:	6548                	.insn	2, 0x6548
   2:	6c6c                	.insn	2, 0x6c6c
   4:	6f57206f          	j	72ef8 <msg2+0x72eec>
   8:	6c72                	.insn	2, 0x6c72
   a:	0064                	.insn	2, 0x0064

0000000c <msg2>:
   c:	6548                	.insn	2, 0x6548
   e:	6c6c                	.insn	2, 0x6c6c
  10:	6f57206f          	j	72f04 <msg2+0x72ef8>
  14:	6c72                	.insn	2, 0x6c72
  16:	0a64                	.insn	2, 0x0a64
	...
```

Why did LO12_I not get its value from "msg2" as HI20. And also, "a0" get the wrong address, which should be 0xc instead.

The version I used is GNU assembler version 2.42.50.

TommyMurphyTM1234, 2024-09-02:
Isn't this because you are assembling `load.s` into an object file whereas if you were to link this object file into a fully linked executable (ELF) file then the linker would perform relaxation in order to fix up such references?

- https://www.sifive.com/blog/all-aboard-part-3-linker-relaxation-in-riscv-toolchain
- https://maskray.me/blog/2021-03-14-the-dark-side-of-riscv-linker-relaxation

joachimcao, 2024-09-02:
Thanks for your reply and references, that helps me a lot.

closed 2024-09-02 by joachimcao