riscv-collab/riscv-gnu-toolchain #1551: Unable to build on RHEL8 due to missing prerequisites

jcrmargetts, 2024-09-12:
Following the instructions in the README.md
[margej@shaak riscv_toolchain_build]: sudo yum install autoconf automake python3 libmpc-devel mpfr-devel gmp-devel gawk  bison flex texinfo patchutils gcc gcc-c++ zlib-devel expat-devel
[sudo] password for margej: 
Updating Subscription Management repositories.
Last metadata expiration check: 0:10:50 ago on Thu 12 Sep 2024 10:34:34 BST.
Package autoconf-2.69-29.el8.noarch is already installed.
Package automake-1.16.1-8.el8.noarch is already installed.
Package python36-3.6.8-39.module+el8.10.0+20784+edafcd43.x86_64 is already installed.
No match for argument: libmpc-devel
No match for argument: mpfr-devel
No match for argument: gmp-devel
Package gawk-4.2.1-4.el8.x86_64 is already installed.
Package bison-3.0.4-10.el8.x86_64 is already installed.
Package flex-2.6.1-9.el8.x86_64 is already installed.
No match for argument: texinfo
Package patchutils-0.3.4-10.el8.x86_64 is already installed.
Package gcc-8.5.0-22.el8_10.x86_64 is already installed.
Package gcc-c++-8.5.0-22.el8_10.x86_64 is already installed.
Package zlib-devel-1.2.11-25.el8.x86_64 is already installed.
Package expat-devel-2.2.5-13.el8_10.x86_64 is already installed.
Error: Unable to find a match: libmpc-devel mpfr-devel gmp-devel texinfo

Is there a known solution other than building on a different OS?

TommyMurphyTM1234, 2024-09-12:
Please attach your full build log for a clean build:
```
make distclean
rm -rf <prefix-dir>
make 2>&1 | tee build.log
```

closed 2024-09-12 by TommyMurphyTM1234

awstaskuser, 2024-09-12:
It looks like you’re missing some required packages (libmpc-devel, mpfr-devel, gmp-devel, and texinfo) for building your toolchain on RHEL 8. These packages might not be available in the default repositories.

TommyMurphyTM1234, 2024-09-12:
> It looks like you’re missing some required packages (libmpc-devel, mpfr-devel, gmp-devel, and texinfo) for building your toolchain on RHEL 8. These packages might not be available in the default repositories.

I already explained above how to get them via the RHEL CodeReady Linux Builder repo!