riscv-collab/riscv-gnu-toolchain #1556: Show the error about the stamps/build-gcc-newlib-stage1

patricksheu, 2024-09-24:
I compile on the macOS 15 and ubuntu 24，

During the build process, the compilation stops midway, and various related errors appear regarding `stamps/build-gcc-newlib`. I followed the instructions provided in the [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain) of the `riscv-gnu-toolchain` repository to set up the build, but encountered errors such as (but not limited to):

```bash
13 warnings generated.
5 warnings generated.
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2

3 warnings generated.
13 warnings generated.
3 warnings generated.
5 warnings generated.
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2

rm -f /Volumes/1T_SSD/source/tools/riscv/lib/bfd-plugins/libdep.la
rm -f /Volumes/1T_SSD/source/tools/riscv/lib/bfd-plugins/libdep.dll.a
mkdir -p stamps/ && touch stamps/build-binutils-newlib
```

These are the common errors, mostly revolving around these three types.

I am using the macOS 15 platform, and here are my hardware details:

**Hardware Overview**:
- **Model Name**: MacBook Pro
- **Model Identifier**: Mac14,10
- **Model Number**: Z17500016TA/A
- **Chip**: Apple M2 Pro
- **Total Number of Cores**: 12 (8 performance and 4 efficiency)
- **Memory**: 16 GB
- **System Firmware Version**: 11881.1.1
- **OS Loader Version**: 11881.1.1
- **Serial Number (system)**: GF326CFMJG
- **Hardware UUID**: 8CD9B856-4DC0-55AE-ACA8-9F1EA324E769
- **Provisioning UDID**: 00006020-000409D43C87401E
- **Activation Lock Status**: Enabled

and the PC hardware intel 8700 16GRAM

Maybe I am missing some packages, or there is an issue with my environment variables. If that's the case, please provide guidance on how I should debug this.

Please help me. 

Thank.
Patrick

TommyMurphyTM1234, 2024-09-24:
Regarding the macOS issues...

First step in such circumstances is always to capture and upload the build log for a clean non-parallel build and see if that has any clues. E.g.
```
make distclean
rm -rf <prefix-dir>
make 2>&1 | tee build.log
```
Depending on what the build log says it may be useful to search the open and closed issues for info relevant to macOS builds that may be relevant. E.g.:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+macos

A problem with macOS build issues is the none/few of us who are active here have access to a platform for reproducing/investigating such build issues and no macOS users have stepped to to help out on that front.

In my experience builds on Ubuntu work fine so I'm not sure why you're having problems but a build log for this build may also help. 

This (or whatever variant you need on the `configure` step) should work:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
```

patricksheu, 2024-09-24:
I all ready try git submodule update --init --recursive, reinstall backage, reboot, reclone project, used gcc 14 g++ 14, gcc 11 g++ 11,and clang, I have briefly checked similar issues on [riscv-gnu-toolchain GitHub Issues](https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+macos+newlib), but it seems that most solutions are just workarounds, such as bypassing problematic packages, downgrading or upgrading compilers, using pre-built packages, or not providing a method to actually fix the issue. These are not true solutions to the problem.
This is mac log [build.log](https://github.com/user-attachments/files/17121545/build.log)

TommyMurphyTM1234, 2024-09-24:
This worked fine for me - I used a Ubuntu 24.04 virtual machine on WSL2 but that's purely an implementation detail...
```
# Using WSL2 run a clean Ubuntu 24.04 virtual machine

# Update all packages
sudo apt update
sudo apt upgrade

# Install git
sudo apt install git

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Install riscv-gnu-toolchain prerequisites
sudo ./github/setup-apt.sh

# Configure (--disable-gdb just to save build time for this test)
./configure --prefix=`pwd`/installed-tools --disable-gdb

# Build
make 2>&1 | tee build.log
```
- `build.log` (zipped): [build.log.zip](https://github.com/user-attachments/files/17122029/build.log.zip)

TommyMurphyTM1234, 2024-09-24:
> This is mac log [build.log](https://github.com/user-attachments/files/17121545/build.log)

These seem to be the errors:
```
Search "error:" (20 hits in 1 file of 1 searched) [RegEx]
  C:\Users\Tommy Murphy\Downloads\build.log (20 hits)
	Line 55266: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:496:3: error: '__abi_tag__' attribute only applies to structs, variables, functions, and namespaces
	Line 55281: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:496:57: error: expected ';' at end of declaration list
	Line 55284: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:498:68: error: too many arguments provided to function-like macro invocation
	Line 55299: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:504:68: error: too many arguments provided to function-like macro invocation
	Line 55314: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:575:3: error: '__abi_tag__' attribute only applies to structs, variables, functions, and namespaces
	Line 55329: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:575:57: error: expected ';' at end of declaration list
	Line 55332: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:577:68: error: too many arguments provided to function-like macro invocation
	Line 55347: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:583:68: error: too many arguments provided to function-like macro invocation
	Line 55362: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:683:55: error: too many arguments provided to function-like macro invocation
	Line 55377: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:683:35: error: declaration conflicts with target of using declaration already in scope
	Line 55395: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:684:3: error: expected expression
	Line 55398: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:685:2: error: expected ';' at end of declaration
	Line 55401: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:688:55: error: too many arguments provided to function-like macro invocation
	Line 55416: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:688:35: error: declaration conflicts with target of using declaration already in scope
	Line 55434: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:689:3: error: expected expression
	Line 55437: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:690:2: error: expected ';' at end of declaration
	Line 55440: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:693:55: error: too many arguments provided to function-like macro invocation
	Line 55455: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:693:35: error: declaration conflicts with target of using declaration already in scope
	Line 55473: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__locale:694:3: error: expected expression
	Line 55476: fatal error: too many errors emitted, stopping now [-ferror-limit=]
```
My comments:

1. Those errors seem to be occurring in/with SDK files and not with `riscv-gnu-toolchain` files. As such this may be some issue/quirk with the macOS development environment. As I said, unfortunately neither I nor (m)any of the other people generally active here have access to macOS systems to reproduce and investigate such issues. You may get better feedback on a more specialised macOS development forum.
2. I cannot reconcile this build log with the issues mentioned in your original post.
3. As mentioned above a Ubuntu 24.04 build went fine for me so I don't know why you're having problems and have no insight in the absence of your Ubunmtu 24.04 build log.

TommyMurphyTM1234, 2024-09-24:
> These seem to be the errors:
>
> ```
> Search "error:" (20 hits in 1 file of 1 searched) [RegEx]
>   C:\Users\Tommy Murphy\Downloads\build.log (20 hits)
> 	Line 55266: [...]

Maybe this is relevant here?

- https://trac.macports.org/ticket/70754

patricksheu, 2024-09-25:
I encountered some compilation issues on Mac, where parts couldn't be compiled using GCC and G++. However, the process was successful on Linux. I'm exploring other possible solutions. Do you have any recommended directions to investigate further?

TommyMurphyTM1234, 2024-09-25:
> However, the process was successful on Linux.

Great. 

> I encountered some compilation issues on Mac, where parts couldn't be compiled using GCC and G++

Do you mean the errors on the build log that you posted or something else?

> Do you have any recommended directions to investigate further?

I don't really have any specific advice on the macOS build issues other than to maybe check out ask on more specialised macOS development forums. I don't really understand how macOS manages the development environment (XCode?) for stuff like this and, as I said, unfortunately I don't have access to a macOS platform to reproduce and investigate such build issues. 

If you could make do with a binary build of the `riscv-gnu-toolchain` then maybe @ilg-ul's xPack Project binary builds might help? 

- https://xpack.github.io/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
    - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/
      - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v14.2.0-2/
        - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/download/v14.2.0-2/xpack-riscv-none-elf-gcc-14.2.0-2-darwin-arm64.tar.gz

patricksheu, 2024-09-25:
I cannot speculate whether the new errors are related, but --with-sysroot=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk requires using the macOS-provided SDK. I haven't found an alternative solution yet.
This is new log: [build.v2.log](https://github.com/user-attachments/files/17131519/build.v2.log)

ilg-ul, 2024-09-25:
GCC builds on macOS (especially on Apple Silicon) are a bit more tricky and require some patches.

The reference is Iain's fork (https://github.com/iains).

For the xPack GCC builds I generate patches for each GCC version:

- https://github.com/xpack-dev-tools/xbb-helper-xpack/tree/xpack/patches

TommyMurphyTM1234, 2024-09-25:
> I cannot speculate whether the new errors are related, but --with-sysroot=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk requires using the macOS-provided SDK. I haven't found an alternative [...]

Seems to be different errors this time - but still in SDK files and not `riscv-gnu-toolchain` source files?
```
Search "error:" (13 hits in 1 file of 1 searched) [RegEx]
  C:\Users\Tommy Murphy\Downloads\build.v2.log (13 hits)
	Line 17767: ../../../gcc/gcc/graphite.h:24:1: error: 'Include' does not name a type
	Line 17772: ../../../gcc/gcc/graphite.h:24:1: error: 'Include' does not name a type
	Line 17776: ../../../gcc/gcc/sese.h:97:3: error: 'ifsese' does not name a type
	Line 17780: ../../../gcc/gcc/sese.h:97:3: error: 'ifsese' does not name a type
	Line 17787: ../../../gcc/gcc/graphite.h:24:1: error: 'Include' does not name a type
	Line 17791: ../../../gcc/gcc/sese.h:97:3: error: 'ifsese' does not name a type
	Line 17795: ../../../gcc/gcc/graphite-isl-ast-to-gimple.cc:1013:36: error: 'class sese_info_t' has no member named 'if_region'; did you mean 'region'?
	Line 17800: ../../../gcc/gcc/graphite-isl-ast-to-gimple.cc:1492:11: error: 'class sese_info_t' has no member named 'if_region'; did you mean 'region'?
	Line 17804: ../../../gcc/gcc/graphite-isl-ast-to-gimple.cc:1499:11: error: 'class sese_info_t' has no member named 'if_region'; did you mean 'region'?
	Line 17842: ../../../gcc/gcc/graphite.h:24:1: error: 'Include' does not name a type
	Line 17846: ../../../gcc/gcc/sese.h:97:3: error: 'ifsese' does not name a type
	Line 17863: ../../../gcc/gcc/graphite.h:24:1: error: 'Include' does not name a type
	Line 17867: ../../../gcc/gcc/sese.h:97:3: error: 'ifsese' does not name a type
```

patricksheu, 2024-09-25:
> GCC builds on macOS (especially on Apple Silicon) are a bit more tricky and require some patches.
> 
> The reference is Iain's fork (https://github.com/iains).
> 
> For the xPack GCC builds I generate patches for each GCC version:
> 
> * https://github.com/xpack-dev-tools/xbb-helper-xpack/tree/xpack/patches

Your suggestion was correct, it seems like the compilation has succeeded now. Thank you very much!

This is the log: [build.log](https://github.com/user-attachments/files/17134028/build.log)

TommyMurphyTM1234, 2024-09-26:
> Your suggestion was correct, it seems like the compilation has succeeded now. Thank you very much!
> 
> This is the log: [build.log](https://github.com/user-attachments/files/17134028/build.log)

@patricksheu - can this issue be closed so?

patricksheu, 2024-09-26:
You can close the issue now. Thank you very much for everyone's help.

closed 2024-09-26 by TommyMurphyTM1234