riscv-collab/riscv-gnu-toolchain #1557: make linux error: unknown type name 'class'

kuanchiehhsu, 2024-09-25:
I was trying to compile for linux using ```./configure --prefix=/opt/rsicv``` and ```make linux```, the following syntax error keeps shows up. I know this error is related to mistakenly use gcc instead of g++.  I compile in a ```ubuntu:20.04``` docker base image on a Mac M3 chip host machine.

```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/home/binutils/gprofng/libcollector -U_ASM -I.. -I/home/binutils/gprofng/libcollector -I/home/binutils/gprofng/libcollector/../common -I/home/binutils/gprofng/libcollector/../src -I/home/binutils/gprofng/libcollector/../../include -I../../bfd -I/home/binutils/gprofng/libcollector/../.. -Wall -Wno-nonnull-compare -g -O2 -MT libgp_collector_la-mmaptrace.lo -MD -MP -MF .deps/libgp_collector_la-mmaptrace.Tpo -c /home/binutils/gprofng/libcollector/mmaptrace.c  -fPIC -DPIC -o .libs/libgp_collector_la-mmaptrace.o
In file included from /home/binutils/gprofng/libcollector/../src/elf.h:30,
                 from /home/binutils/gprofng/libcollector/mmaptrace.c:32:
/home/binutils/gprofng/libcollector/../src/Data_window.h:33:1: error: unknown type name 'class'
   33 | class Data_window
      | ^~~~~
```

But I'm not sure what could be the root cause of this? Both ```gcc``` and ```g++``` are ```9.4.0```, default versions. I do install prerequisites in my dockerfile.

TommyMurphyTM1234, 2024-09-25:
Must be something with your Ubuntu 20.04 Docker image.
The CI builds are working fine on 20.04:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases

This is the latest full build of the 20.04 Linix/Glibc toolchain:

- https://github.com/riscv-collab/riscv-gnu-toolchain/actions/runs/10938390054/job/30366330258

The recipes for such builds are here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master/.github

TommyMurphyTM1234, 2024-09-26:
I did this and the build completed successfully.
The host OS was Ubuntu 24.04 on WSL2.
```
# Run a Ubuntu 20.04 Docker container interactively
docker run -it ubuntu:20.04

# Update software
apt update
apt upgrade

# Install git
apt install git

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Install prerequisites
.github/setup-apt.sh

# Configure and make 
./configure --prefix=`pwd`/installed-tools
make linux 2>&1 | tee build.log
```

TommyMurphyTM1234, 2024-09-26:
> I was trying to compile for linux using `./configure --prefix=/opt/rsicv` and `make linux`, the following syntax error keeps showing up. I know this error is related to mistakenly use gcc instead of g++. I compile in a `ubuntu:20.04` docker base image on a Mac M3 chip host machine.
> 
> ```
> libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/home/binutils/gprofng/libcollector -U_ASM -I.. -I/home/binutils/gprofng/libcollector -I/home/binutils/gprofng/libcollector/../common -I/home/binutils/gprofng/libcollector/../src -I/home/binutils/gprofng/libcollector/../../include -I../../bfd -I/home/binutils/gprofng/libcollector/../.. -Wall -Wno-nonnull-compare -g -O2 -MT libgp_collector_la-mmaptrace.lo -MD -MP -MF .deps/libgp_collector_la-mmaptrace.Tpo -c /home/binutils/gprofng/libcollector/mmaptrace.c  -fPIC -DPIC -o .libs/libgp_collector_la-mmaptrace.o
> In file included from /home/binutils/gprofng/libcollector/../src/elf.h:30,
>                  from /home/binutils/gprofng/libcollector/mmaptrace.c:32:
> /home/binutils/gprofng/libcollector/../src/Data_window.h:33:1: error: unknown type name 'class'
>    33 | class Data_window
>       | ^~~~~
> ```
FWIW I get this in my log file (and no compilation error/issue obviously):
```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/riscv-gnu-toolchain/binutils/gprofng/libcollector -U_ASM -I.. -I/riscv-gnu-toolchain/binutils/gprofng/libcollector -I/riscv-gnu-toolchain/binutils/gprofng/libcollector/../common -I/riscv-gnu-toolchain/binutils/gprofng/libcollector/../src -I/riscv-gnu-toolchain/binutils/gprofng/libcollector/../../include -I../../bfd -I/riscv-gnu-toolchain/binutils/gprofng/libcollector/../.. -Wall -Wno-nonnull-compare -g -O2 -MT libgp_collector_la-mmaptrace.lo -MD -MP -MF .deps/libgp_collector_la-mmaptrace.Tpo -c /riscv-gnu-toolchain/binutils/gprofng/libcollector/mmaptrace.c -o libgp_collector_la-mmaptrace.o >/dev/null 2>&1
mv -f .deps/libgp_collector_la-mmaptrace.Tpo .deps/libgp_collector_la-mmaptrace.Plo
```

TommyMurphyTM1234, 2024-09-26:
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/17155540/build.log.zip)

TommyMurphyTM1234, 2024-09-29:
Any update @kuanchiehhsu?
Do you still get the same issue if you follow my steps here?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1557#issuecomment-2375615407

kuanchiehhsu, 2024-09-30:
Hi @TommyMurphyTM1234, thank you for your support! By following your steps, it works for me. As for my original docker approach, I haven't figured out the reason, and I'm guessing it might be related to how I create the environment. I cloned the repo without submodule initially on host, bind the whole source directory to container, and do `make linux` inside the container. The on-demand submodule requests under this setup maybe the reason.

TommyMurphyTM1234, 2024-09-30:
> Hi @TommyMurphyTM1234, thank you for your support! By following your steps, it works for me. 

Great. 👍 

> As for my original docker approach, I haven't figured out the reason, and I'm guessing it might be related to how I create the environment. I cloned the repo without submodule initially on host, [...]

If you get a chance maybe you could detail the exact commands that you used to do this? 

If you need to persist the work from the container to the host then you can use docker volumes to create a persistent storage area.

- https://docs.docker.com/engine/storage/volumes

> Volumes are the preferred mechanism for persisting data generated by and used by Docker containers. While bind mounts are dependent on the directory structure and OS of the host machine, volumes are completely managed by Docker. Volumes have several advantages over bind mounts:
> ...

TommyMurphyTM1234, 2024-10-01:
For what it's worth, the toolchain builds fine for me using both a Docker volume and Docker bind mount to persist the results to the host.

# Using Docker volume
```
# On host

# Create shared-dir
sudo rm -rf shared-dir
mkdir shared-dir

# Run Docker container
docker run -it -v `pwd`/shared-dir:/shared-dir ubuntu:20.04

# In container

# Update software
apt update
apt upgrade

# Install git
apt install git

# Clone riscv-gnu-toolchain
cd shared-dir
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Install riscv-gnu-toolchain prerequisites
.github/setup-apt.sh

# Configure and build riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make linux 2>&1 | tee build.log

# Exit container
exit

# Back on host
ls -l shared-dir/riscv-gnu-toolchain/installed-tools/
total 32
drwxr-xr-x 2 root root 4096 Oct  1 07:44 bin
drwxr-xr-x 2 root root 4096 Sep 30 21:57 etc
drwxr-xr-x 4 root root 4096 Oct  1 07:44 include
drwxr-xr-x 5 root root 4096 Oct  1 07:44 lib
drwxr-xr-x 3 root root 4096 Sep 30 22:49 libexec
drwxr-xr-x 5 root root 4096 Oct  1 07:21 riscv64-unknown-linux-gnu
drwxr-xr-x 6 root root 4096 Oct  1 07:44 share
drwxr-xr-x 7 root root 4096 Sep 30 23:03 sysroot
```
# Using Docker bind mount:
```
# On host

# Create shared-dir
sudo rm -rf shared-dir
mkdir shared-dir
cd shared-dir

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
# Change owner:group to root:root to avoid problems inside Docker container
chown -R root:root riscv-gnu-toolchain
cd ..

# Run Docker container
docker run -it --mount type=bind,source=`pwd`/shared-dir,target=/shared-dir ubuntu:20.04

# In container

# Update software
apt update
apt upgrade

# Install riscv-gnu-toolchain prerequisites
cd shared-dir/riscv-gnu-toolchain
.github/setup-apt.sh

# Configure and build riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make linux 2>&1 | tee build.log

# Exit container
exit

# Back on host
ls -l shared-dir/riscv-gnu-toolchain/installed-tools/
total 32
drwxr-xr-x 2 root root 4096 Oct  1 14:11 bin
drwxr-xr-x 2 root root 4096 Oct  1 11:31 etc
drwxr-xr-x 4 root root 4096 Oct  1 14:11 include
drwxr-xr-x 5 root root 4096 Oct  1 14:11 lib
drwxr-xr-x 3 root root 4096 Oct  1 12:23 libexec
drwxr-xr-x 5 root root 4096 Oct  1 13:47 riscv64-unknown-linux-gnu
drwxr-xr-x 6 root root 4096 Oct  1 14:11 share
drwxr-xr-x 7 root root 4096 Oct  1 12:37 sysroot
```

closed 2024-10-01 by TommyMurphyTM1234

bcstrongx, 2025-06-30:
Adding a comment to this old issue in case anyone runs into the same issue I did.  I cloned the repo using WSL2 running Ubuntu 22.04, following the instructions given by @TommyMurphyTM1234 (though not in a docker container), and was still hitting the original error message.  I was working in the Windows filesystem (/mnt/c/...), and noticed this warning:

```
$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9621, done.
remote: Counting objects: 100% (503/503), done.
remote: Compressing objects: 100% (145/145), done.
remote: Total 9621 (delta 422), reused 358 (delta 358), pack-reused 9118 (from 3)
Receiving objects: 100% (9621/9621), 6.47 MiB | 9.13 MiB/s, done.
Resolving deltas: 100% (4801/4801), done.
Updating files: 100% (1009/1009), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'linux-headers/include/linux/netfilter/xt_CONNMARK.h'
  'linux-headers/include/linux/netfilter/xt_connmark.h'
  'linux-headers/include/linux/netfilter/xt_DSCP.h'
  'linux-headers/include/linux/netfilter/xt_dscp.h'
  'linux-headers/include/linux/netfilter/xt_MARK.h'
  'linux-headers/include/linux/netfilter/xt_mark.h'
  'linux-headers/include/linux/netfilter/xt_RATEEST.h'
  'linux-headers/include/linux/netfilter/xt_rateest.h'
  'linux-headers/include/linux/netfilter/xt_TCPMSS.h'
  'linux-headers/include/linux/netfilter/xt_tcpmss.h'
  'linux-headers/include/linux/netfilter_ipv4/ipt_ECN.h'
  'linux-headers/include/linux/netfilter_ipv4/ipt_ecn.h'
  'linux-headers/include/linux/netfilter_ipv4/ipt_TTL.h'
  'linux-headers/include/linux/netfilter_ipv4/ipt_ttl.h'
  'linux-headers/include/linux/netfilter_ipv6/ip6t_HL.h'
  'linux-headers/include/linux/netfilter_ipv6/ip6t_hl.h'
```

When I instead cloned the repo in the Linux filesystem (in my case, /home/\<idsid\>/...), the instructions above worked.