riscv-collab/riscv-gnu-toolchain #1560: Installation for 32 bits baremetal does not work

dgastonochoa, 2024-10-04:
I'm trying to install the RISC-V GNU toolchain in 32 bits baremetal mode, as following:

```
./configure --prefix=$HOME/.local/opt/riscv --with-arch=rv32gc --with-abi=ilp32
make -j $(nproc)
```

However, the installation never finishes.

I have located this error:
```
/usr/bin/install: /usr/bin/install: cannot stat '/home/user/.local/opt/riscv-gnu-toolchain/binutils/gas/po/ro.gmo'cannot stat '/home/user/.local/opt/riscv-gnu-toolchain/binutils/gprof/po/fr.gmo': No such file or directory
: No such file or directory
```

Also, when the installation finishes with error, if I run `make` again (without `-j`), the installation gets stuck doing:

```
if test -f /home/user/.local/opt/riscv-gnu-toolchain/gcc/contrib/download_prerequisites && test "true" = "true"; then cd /home/user/.local/opt/riscv-gnu-toolchain/gcc && ./contrib/download_prerequisites; fi
```

I have tried to run such script manually, but I also get an error:
```
$ ./contrib/download_prerequisites
failed: Connection timed out.
2024-10-04 23:48:48 URL:http://gcc.gnu.org/pub/gcc/infrastructure/gettext-0.22.tar.gz [26105696/26105696] -> "gettext-0.22.tar.gz" [1]
failed: Connection timed out.
2024-10-04 23:51:04 URL:http://gcc.gnu.org/pub/gcc/infrastructure/gmp-6.2.1.tar.bz2 [2493916/2493916] -> "gmp-6.2.1.tar.bz2" [1
```

TommyMurphyTM1234, 2024-10-04:
This is almost certainly something to do with your local setup/environment.
I have done several builds today with no problems. 
I am doing one with your configuration above but expect it to complete successfully as normal.
You need to at least capture and attachd your (non parallel) build log for analysis - e.g.:
```
# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
# Install prerequisites 
cd riscv-gnu-toolchain
sudo .github/setup-apt.sh
# If you have previously done a build then clean up first
make distclean && rm -rf <prefixdir>
# Configure
./configure --prefix=... --with-arch=rv32gc --with-abi=ilp32
# Non parallel build because parallel builds logs are too difficult to analyse
make 2>&1 | tee build.log
# Zip and upload build.log
```
There should be no need to manually run `download_prerequisites`.

dgastonochoa, 2024-10-04:
It turns out that `./contrib/download_prerequisites` did download the required dependencies despite of throwing a timeout error.

I have re-run `make -j $(nproc)` after the script finished and now it did finished compiling correctly.

closed 2024-10-04 by TommyMurphyTM1234

TommyMurphyTM1234, 2024-10-05:
> I am doing one with your configuration above but expect it to complete successfully as normal. 

It did...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
sudo .github/setup-apt.sh
# Just disable GDB to speed build up
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32 --disable-gdb
make 2>&1 | tee build.log
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/17264755/build.log.zip)