riscv-collab/riscv-gnu-toolchain #1561: Can't include stdint.h and compile with baremetal 32 bits RISC-V toolchain

dgastonochoa, 2024-10-04:
I have built the RISC-V toolchain with the following instructions:
```
./configure --prefix=$HOME/.local/opt/riscv --with-arch=rv32gc --with-abi=ilp32
make -j $(nproc)
```
It builds correctly but, when I try to compile the following:
```
#include <stdint.h>

void main(void)
{
    int32_t n = 0;
    n++;
}
```
with this command:
```
$ riscv32-unknown-elf-gcc -o main -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g -O0 main.c
```
I get this error:
```
/home/user/.local/opt/riscv/lib/gcc/riscv32-unknown-elf/14.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
```

TommyMurphyTM1234, 2024-10-05:
I built the toolchain as mentioned here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1560#issuecomment-2394852695

Then...
```
cd installed-tools/bin
vi main.c
# Insert code as per the previous post
...
 ./riscv32-unknown-elf-gcc -o main -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g -O0 main.c
/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 00010094
file main
main: ELF 32-bit LSB executable, UCB RISC-V, soft-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
```
So, again, your problem is most likely with your local setup.
You need to use the `-v` option to see what's going on.
E.g.:
```
./riscv32-unknown-elf-gcc -v -o main -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g -O0 main.c
Using built-in specs.
COLLECT_GCC=./riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/home/user/riscv-gnu-toolchain/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/riscv-gnu-toolchain/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 ()
COLLECT_GCC_OPTIONS='-v' '-o' 'main' '-march=rv32g' '-mabi=ilp32' '-static' '-mcmodel=medany' '-fvisibility=hidden' '-nostdlib' '-nostartfiles' '-g' '-O0' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imafd_zicsr_zifencei'
 /home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/cc1 -quiet -v main.c -quiet -dumpbase main.c -dumpbase-ext .c -march=rv32g -mabi=ilp32 -mcmodel=medany -mtune=rocket -misa-spec=20191213 -march=rv32imafd_zicsr_zifencei -g -O0 -version -fvisibility=hidden -o /tmp/ccNBCehg.s
GNU C17 () version 14.2.0 (riscv32-unknown-elf)
        compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/home/user/riscv-gnu-toolchain/installed-tools/riscv32-unknown-elf/usr/local/include"
ignoring duplicate directory "/home/user/riscv-gnu-toolchain/installed-tools/riscv32-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/include
 /home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/include-fixed
 /home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/include
End of search list.
Compiler executable checksum: 7450daff86b1fa1d77d568bcc5906559
COLLECT_GCC_OPTIONS='-v' '-o' 'main' '-march=rv32g' '-mabi=ilp32' '-static' '-mcmodel=medany' '-fvisibility=hidden' '-nostdlib' '-nostartfiles' '-g' '-O0' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imafd_zicsr_zifencei'
 /home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/as -v --gdwarf-5 --traditional-format -march=rv32g -march=rv32imafd_zicsr_zifencei -mabi=ilp32 -misa-spec=20191213 -o /tmp/ccFkBjMb.o /tmp/ccNBCehg.s
GNU assembler version 2.43.1 (riscv32-unknown-elf) using BFD version (GNU Binutils) 2.43.1
COMPILER_PATH=/home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/:/home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/:/home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/:/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/:/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/:/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/
LIBRARY_PATH=/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/:/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/lib/:/home/user/riscv-gnu-toolchain/installed-tools/riscv32-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'main' '-march=rv32g' '-mabi=ilp32' '-static' '-mcmodel=medany' '-fvisibility=hidden' '-nostdlib' '-nostartfiles' '-g' '-O0' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imafd_zicsr_zifencei' '-dumpdir' 'main.'
 /home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/collect2 -plugin /home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/liblto_plugin.so -plugin-opt=/home/user/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv32-unknown-elf/14.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cckExXmj.res --sysroot=/home/user/riscv-gnu-tool [... cut ...] olchain/installed-tools/riscv32-unknown-elf/lib /tmp/ccFkBjMb.o
/home/user/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 00010094
COLLECT_GCC_OPTIONS='-v' '-o' 'main' '-march=rv32g' '-mabi=ilp32' '-static' '-mcmodel=medany' '-fvisibility=hidden' '-nostdlib' '-nostartfiles' '-g' '-O0' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imafd_zicsr_zifencei' '-dumpdir' 'main.'
```

dgastonochoa, 2024-10-05:
I think it was a missing package. Thanks.

closed 2024-10-05 by dgastonochoa

TommyMurphyTM1234, 2024-10-05:
> I think it was a missing package. Thanks.

What exactly did you to address the issue?
Run `.github/setup-apt.sh` and rebuild?
Or something else?

dgastonochoa, 2024-10-08:
> Collaborator

No, it was a missing package. The build shown an error which I missed; I was building with -j $(nproc) and it was more difficult to see it. But I saw it and installed with `apt install`. After that, the compilation worked.

Thanks.

TommyMurphyTM1234, 2024-10-08:
> > Collaborator

?

> No, it was a missing package. The build shown an error which I missed; I was building with -j $(nproc) and it was more difficult to see it. But I saw it and installed with `apt install`. After that, the compilation worked.

Thanks for clarifying. Yes, when there are build problems it makes sense to try the build again without any `-j` parallelism in order to get a log that's easier to analyse. The interleaved logging of a parallel build doesn't help with analysis.