riscv-collab/riscv-gnu-toolchain #1565: Should Fortran be enabled for all Linux toolchain variants?

TommyMurphyTM1234, 2024-10-07:
For the Linux/glibc and Linux native toolchains the enabled languages are C, C++ and Fortran:

- Linux/Glibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/Makefile.in#L521
- Linux native: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/Makefile.in#L579

See here for some background on why this is so:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1554#issuecomment-2366728731
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1554#issuecomment-2367085033

At the moment only C and C++ are enabled for other Linux toochain variants:

- Linux/musl: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/Makefile.in#L924
- Linux/uClibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/Makefile.in#L1056

Should Fortran also be enabled for these too?

cmuellner, 2024-10-09:
It is best to keep the defaults as we have them now (nobody ever requested more).
And with #1559, this becomes a user-configurable setting anyway.

TommyMurphyTM1234, 2024-10-09:
> It is best to keep the defaults as we have them now (nobody ever requested more). And with #1559, this becomes a user-configurable setting anyway.

OK - thanks @cmuellner.
I'll close this issue so.

closed 2024-10-09 by TommyMurphyTM1234