riscv-collab/riscv-gnu-toolchain #1566: Should Linux/uClibc toolchain be added to CI builds?

TommyMurphyTM1234, 2024-10-07:
See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/.github/workflows/build.yaml#L12-L24
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/795267df36fcf1269f2bd596ac42e0e91ad9cfd3/.github/workflows/nightly-release.yaml#L45-L59

Should the Linux/uClibc toolchain also be represented there?

cmuellner, 2024-10-09:
I'd like it to be included, so we detect if uClibc-builds break.

TommyMurphyTM1234, 2024-10-09:
> I'd like it to be included, so we detect if uClibc-builds break.

Ok - thanks for the feedback @cmuellner.
I will try to look at doing a PR for this soon.

TommyMurphyTM1234, 2024-10-11:
Addressed by this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1578#event-14613065426

closed 2024-10-11 by TommyMurphyTM1234