riscv-collab/riscv-gnu-toolchain #1569: Why is latest CI build still GCC 13.2 and not 14.2?

TommyMurphyTM1234, 2024-10-09:
The PR to bump the toolchain to GCC 14.2 was merged a few weeks ago:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

but the latest CI build of the toolchain is still GCC 13.2:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases

Why?

cmuellner, 2024-10-09:
Thanks for bringing this up! I noticed the request from @szedani, but illness kept me from moving this thing forward.

In essence, the reason why the CI builds fail is that Ubuntu 20.04 LTS is too old to build a recent version of QEMU.

TommyMurphyTM1234, 2024-10-09:
Thanks @cmuellner - hope you're feeling better now/soon!

cmuellner, 2024-10-09:
The proposed fix is to drop Ubuntu 20.04  (see https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1570).

TommyMurphyTM1234, 2024-10-09:
> The proposed fix is to drop Ubuntu 20.04 (see #1570).

Thanks @cmuellner - I had actually seen that but failed to join the dots... :-)

TommyMurphyTM1234, 2024-10-10:
When does a "release" job get triggered?
It doesn't seem to have happened even though the PR in question was merged yesterday:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1570

cmuellner, 2024-10-10:
The PR was merged today, not yesterday.

The nightly release is defined here: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/workflows/nightly-release.yaml
The relevant part is:
```
on:
  schedule:
    - cron:  '0 0 * * ?'
```

Cron columns are minute, hour, day-of-month, month-of-year, day-of-week.
So, the action is triggered at midnight (likely 00:00 UTC) on any day of the month, any month of the year, and any day of the week.

TommyMurphyTM1234, 2024-10-10:
Thanks @cmuellner. 👍 
Timezone issues confused me - I'm in Ireland and according to GitHub the PR was merged yesterday for me. :-)

cmuellner, 2024-10-10:
Here in Austria the PR page states "kito-cheng merged commit cbd3989 into riscv-collab:master 5 hours ago". And when I hover over the "5 hours ago" the hover text says "Oct 10, 2024, 11:42 AM GMT+2".

TommyMurphyTM1234, 2024-10-10:
> Here in Austria the PR page states "kito-cheng merged commit [cbd3989](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/cbd3989791e9d77426c064562e306e7906768ea6) into riscv-collab:master 5 [...]

Sorry - my mistake. I glanced at the PR and misread this as "committed" and assumed that it was the merging of the PR.

[image]

TommyMurphyTM1234, 2024-10-10:
Should be fixed by this:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1570

TommyMurphyTM1234, 2024-10-25:
See also...

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1591

TommyMurphyTM1234, 2024-11-02:
Closing this since it's effectively already covered here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1591

And also by various work in progress PRs.

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1608
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1607
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1606
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1605
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1601

closed 2024-11-02 by TommyMurphyTM1234