riscv-collab/riscv-gnu-toolchain #1572: Make `--enable-multilib` work for Linux with musl/uClibc toolchain builds

TommyMurphyTM1234, 2024-10-09:
For some reason `Makefile.in` explicitly uses `--disable-multilib` for the Linux/musl and Linux/uClibc toolchain builds:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/071f07c9364856cdcf49dfee0c645f564305df52/Makefile.in#L910-L941
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/071f07c9364856cdcf49dfee0c645f564305df52/Makefile.in#L934

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/071f07c9364856cdcf49dfee0c645f564305df52/Makefile.in#L1046-L1073
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/071f07c9364856cdcf49dfee0c645f564305df52/Makefile.in#L1066

I tried changing this for the Linux/musl build:

```
		$(GCC_CHECKING_FLAGS) \
		$(GCC_MULTILIB_FLAGS) \
		$(WITH_ABI) \
```
but the build failed:
```
# @multilib_dir@ is not really necessary, but sometimes it has
# more uses than just a directory name.
/bin/bash ../../../../.././gcc/libgcc/../mkinstalldirs lib32/ilp32
mkdir -p -- lib32/ilp32
/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2/./gcc/xgcc -B/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2/./gcc/ -B/home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-musl/bin/ -B/home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-musl/lib/ -isystem /home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-musl/include -isystem /home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-musl/sys-include    -O2  -O2    -mcmodel=medlow -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include  -fPIC -DGTHREAD_USE_WEAK=0 -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector  -shared -nodefaultlibs -Wl,--soname=libgcc_s.so.1 -Wl,--version-script=libgcc.map -march=rv32imac -mabi=ilp32 -o lib32/ilp32/libgcc_s.so.1.tmp -O2 -mcmodel=medlow -march=rv32imac -mabi=ilp32 -B./ _negdi2_s.o _lshrdi3_s.o _ashldi3_s.o _ashrdi3_s.o _cmpdi2_s.o _ucmpdi2_s.o _clear_cache_s.o _trampoline_s.o __main_s.o _absvsi2_s.o _absvdi2_s.o _addvsi3_s.o _addvdi3_s.o _subvsi3_s.o _subvdi3_s.o _mulvsi3_s.o _mulvdi3_s.o _negvsi2_s.o _negvdi2_s.o _ctors_s.o _ffssi2_s.o _ffsdi2_s.o _clz_s.o _clzsi2_s.o _clzdi2_s.o _ctzsi2_s.o _ctzdi2_s.o _popcount_tab_s.o _popcountsi2_s.o _popcountdi2_s.o _paritysi2_s.o _paritydi2_s.o _powisf2_s.o _powidf2_s.o _powixf2_s.o _powitf2_s.o _mulhc3_s.o _mulsc3_s.o _muldc3_s.o _mulxc3_s.o _multc3_s.o _divhc3_s.o _divsc3_s.o _divdc3_s.o _divxc3_s.o _divtc3_s.o _bswapsi2_s.o _bswapdi2_s.o _clrsbsi2_s.o _clrsbdi2_s.o _mulbitint3_s.o _fixunsxfsi_s.o _fixxfdi_s.o _fixunsxfdi_s.o _floatdixf_s.o _floatundixf_s.o _divdi3_s.o _moddi3_s.o _divmoddi4_s.o _udivdi3_s.o _umoddi3_s.o _udivmoddi4_s.o _udiv_w_sdiv_s.o _divmodbitint4_s.o adddf3_s.o divdf3_s.o eqdf2_s.o gedf2_s.o ledf2_s.o muldf3_s.o negdf2_s.o subdf3_s.o unorddf2_s.o fixdfsi_s.o fixunsdfsi_s.o floatsidf_s.o floatunsidf_s.o fixdfdi_s.o fixunsdfdi_s.o floatdidf_s.o floatundidf_s.o addtf3_s.o divtf3_s.o eqtf2_s.o getf2_s.o letf2_s.o multf3_s.o negtf2_s.o subtf3_s.o unordtf2_s.o fixtfsi_s.o fixunstfsi_s.o floatsitf_s.o floatunsitf_s.o fixtfdi_s.o fixunstfdi_s.o floatditf_s.o floatunditf_s.o addsf3_s.o divsf3_s.o eqsf2_s.o gesf2_s.o lesf2_s.o mulsf3_s.o negsf2_s.o subsf3_s.o unordsf2_s.o fixsfsi_s.o fixunssfsi_s.o floatsisf_s.o floatunsisf_s.o fixsfdi_s.o fixunssfdi_s.o floatdisf_s.o floatundisf_s.o fixsfbitint_s.o floatbitintsf_s.o fixdfbitint_s.o floatbitintdf_s.o extendsfdf2_s.o extendsftf2_s.o extenddftf2_s.o extendhfsf2_s.o extendhfdf2_s.o extendhftf2_s.o truncdfsf2_s.o trunctfsf2_s.o trunctfdf2_s.o trunctfhf2_s.o truncdfhf2_s.o truncsfhf2_s.o fixhfsi_s.o fixhfdi_s.o fixunshfsi_s.o fixunshfdi_s.o floatsihf_s.o floatdihf_s.o floatunsihf_s.o floatundihf_s.o save-restore_s.o muldi3_s.o multi3_s.o div_s.o atomic_s.o enable-execute-stack_s.o hardcfr_s.o strub_s.o unwind-dw2_s.o unwind-dw2-fde-dip_s.o unwind-sjlj_s.o unwind-c_s.o emutls_s.o libgcc.a -lc && rm -f lib32/ilp32/libgcc_s.so && if [ -f lib32/ilp32/libgcc_s.so.1 ]; then mv -f lib32/ilp32/libgcc_s.so.1 lib32/ilp32/libgcc_s.so.1.backup; else true; fi && mv lib32/ilp32/libgcc_s.so.1.tmp lib32/ilp32/libgcc_s.so.1 && (echo "/* GNU ld script"; echo "   Use the shared library, but some functions are only in"; echo "   the static library.  */"; echo "GROUP ( libgcc_s.so.1 -lgcc )" ) > lib32/ilp32/libgcc_s.so
/home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-musl/bin/ld: /home/user/musl/riscv-gnu-toolchain/installed-tools/sysroot/lib/libc.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:1005: libgcc_s.so] Error 1
make[5]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2/riscv64-unknown-linux-musl/lib32/ilp32/libgcc'
make[4]: *** [Makefile:1221: multi-do] Error 1
make[4]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2/riscv64-unknown-linux-musl/libgcc'
make[3]: *** [Makefile:127: all-multi] Error 2
make[3]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2/riscv64-unknown-linux-musl/libgcc'
make[2]: *** [Makefile:13228: all-target-libgcc] Error 2
make[2]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2'
make[1]: *** [Makefile:1059: all] Error 2
make[1]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-musl-stage2'
make: *** [Makefile:911: stamps/build-gcc-musl-stage2] Error 2
```

- `build-multilib.log.zip`: [build-multilib.log.zip](https://github.com/user-attachments/files/17308686/build-multilib.log.zip)

I haven't checked the Linux/uClibc build yet but I presume that something similar happens there.

I guess that some work needs to be done to make these toolchain variant builds work with `--enable-multilib`?

TommyMurphyTM1234, 2024-10-09:
> I haven't checked the Linux/uClibc build yet but I presume that something similar happens there.

This also fails but in a different way it seems...
```
Running configure in multilib subdirs lib32/ilp32 lib32/ilp32d lib64/lp64 lib64/lp64d
pwd: /home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2/riscv64-unknown-linux-uclibc/libgomp
Running configure in multilib subdir lib32/ilp32
pwd: /home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2/riscv64-unknown-linux-uclibc
configure: creating cache ./config.cache
checking for --enable-version-specific-runtime-libs... no
checking for --enable-generated-files-in-srcdir... no
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-uclibc
checking target system type... riscv64-unknown-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for riscv64-unknown-linux-uclibc-strip... /home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-uclibc/bin/strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... /usr/bin/gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for riscv64-unknown-linux-uclibc-gcc... /home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2/./gcc/xgcc -B/home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2/./gcc/ -B/home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-uclibc/bin/ -B/home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-uclibc/lib/ -isystem /home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-uclibc/include -isystem /home/user/musl/riscv-gnu-toolchain/installed-tools/riscv64-unknown-linux-uclibc/sys-include  -march=rv32imac -mabi=ilp32
checking whether the C compiler works... no
configure: error: in `/home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2/riscv64-unknown-linux-uclibc/lib32/ilp32/libgomp':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[2]: *** [Makefile:14891: configure-target-libgomp] Error 1
make[2]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2'
make[1]: *** [Makefile:1059: all] Error 2
make[1]: Leaving directory '/home/user/musl/riscv-gnu-toolchain/build-gcc-uclibc-stage2'
make: *** [Makefile:1044: stamps/build-gcc-uclibc-stage2] Error 2
```
- [build-uclibc-multilib.log.zip](https://github.com/user-attachments/files/17317235/build-uclibc-multilib.log.zip)

heyuanjie87, 2024-10-18:
也许可以用其他方法实现musl-multilib:
first,, use newlib to build multilib gcc,then repalce libc of newlib with musl libc

heyuanjie87, 2024-10-19:
[image]
[image]

I do  "make musl" in WSL2, but no gcc is generated。
Luckily, “make newlib” can work well。
so,i building the bare-metal/Newlib (multilib) toolchain to get a multilib supported gcc.
libc of newlib is not what I wanted,just delete it,and use musl to replace them

mickflemm, 2024-10-31:
Multilib is not supported by musl: https://wiki.musl-libc.org/guidelines-for-distributions
As for uclibc-ng it's a bit messy to make it work, it needs config changes etc.

TommyMurphyTM1234, 2024-11-01:
> Multilib is not supported by musl: https://wiki.musl-libc.org/guidelines-for-distributions As for uclibc-ng it's a bit messy to make it work, it needs config changes etc.

Thanks @mickflemm. Perhaps this info renders my issue moot and I should close it since it's not something that can reasonably be dealt with in this repo?

mickflemm, 2024-11-01:
We could have multilib uClibc I guess but I haven't played with that setup, if you feel that's something useful leave this issue open, if not let's close it and maybe review it in the future.

TommyMurphyTM1234, 2024-11-01:
Thanks @mickflemm. Nobody has been requesting this so far and I only logged it in case it was an easy gap to plug. So I'll close the issue for now and if it crops up again it can be reexamined/reinvestigated at that stage.

closed 2024-11-01 by TommyMurphyTM1234