riscv-collab/riscv-gnu-toolchain #1573: Update DejaGnu?

TommyMurphyTM1234, 2024-10-09:
As far as I can see, `riscv-gnu-toolchain` uses a very old version of DejaGnu:
```
git submodule status dejagnu
 ca371cf9c48186716d26c8e10d442affaf002c80 dejagnu (dejagnu_1_4_3-911-gca371cf)
```
And again. as far as I can see, DejaGnu v1.4.3 dates back to 2002!

- https://ftp.gnu.org/gnu/dejagnu/

Is this correct? 
If so then surely it's long overdue a bump to a more recent version - maybe the latest version v1.6.5?

However I'm a bit confused by some of the DejaGnu related PRs...

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1468
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1247
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1234
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1133

Am I misunderstanding something here?

If it does merit a bump then how can this be tested?
I have tried running the test suite on multilib enabled toolchains on my hardware but it takes far too long to be practical.
Is there some other and more efficient way to do this - e.g. on some remote/cloud server perhaps?

cmuellner, 2024-10-09:
`dejagnu_1_4_3-911-gca371cf` means tag `dejagnu_1_4_3` plus 911 commits on top, which is commit `ca371cf` from `Sat Apr 13 19:36:47 2024`.

We cannot roll-back much, because upstream DejaGnu had an issue, which caused fails in RISC-V multilib builds. I think this was fixed last year and once this was done, we switched to upstream DejaGnu. Afaik, there is no release tag upstream that we can use.

TommyMurphyTM1234, 2024-10-09:
> `dejagnu_1_4_3-911-gca371cf` means tag `dejagnu_1_4_3` plus 911 commits on top, which is commit `ca371cf` from `Sat Apr 13 19:36:47 2024`.

Ah - thanks for the explanation @cmuellner. I didn't realise that but maybe it's a common convention?

> We cannot roll-back much, because upstream DejaGnu had an issue, which caused fails in RISC-V multilib builds. I think this was fixed last year and once this was done, we switched to upstream DejaGnu. Afaik, there is no release tag upstream that we can use.

OK - thanks. I'll close this issue so.

closed 2024-10-09 by TommyMurphyTM1234