riscv-collab/riscv-gnu-toolchain #1575: Update gcc version in .gitmodules

sendaoYan, 2024-10-10:
Hi all,
https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531 update gcc version to gcc14.2, but the gcc version in `.gitmodules` sitll is gcc13

sendaoYan, 2024-10-10:
https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1472/files

TommyMurphyTM1234, 2024-10-10:
Not sure why this was closed as it's a valid issue. 

The PR to bump to GCC 14.2 failed to update `.gitmodules`.

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

This PR to bump to GCC 14.1 was rendered moot by the one above: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1472

I'm not sure that this omission has any practical/negative impact but it makes sense to keep the file up to date even just to avoid confusion. 

I'll prepare a PR to update `.gitmodules` this later today.

pz9115, 2024-10-10:
I think I forgot change the git submodule part, thank you point it out.

TommyMurphyTM1234, 2024-10-10:
Resolved by this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1576

closed 2024-10-10 by TommyMurphyTM1234