riscv-collab/riscv-gnu-toolchain #1577: Cannot build on Mac (aarch64)

MarcoBonino, 2024-10-10:
Hello!
I'm trying to build on MacOS aarch64 (M3 chip).
However some tools like riscv64-unknown-elf-gcc and riscv64-unknown-elf-g++ are missing.

I'm from a case sensitive filesystem. And I installed the required tools:
```
brew install python3 gawk gnu-sed gmp mpfr libmpc isl zlib expat texinfo flock libslirp
```
This is my configure:
```
# get gmp, mpfr and libmpc available during ./configure script execution 
export CPPFLAGS="-I/opt/homebrew/include"
export LDFLAGS="-L/opt/homebrew/lib"
./configure --prefix=/Volumes/linux/bin-riscv-gnu-toolchain
```
And I get:
```
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /opt/homebrew/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
```

Then I simply run `make`.

Could you provide support on this ? 
If possible would be also nice to have the MacOS toolchain already built as it was done for the Ubuntu builds.

Thanks!

TommyMurphyTM1234, 2024-10-10:
Do you actually need to build the toolchain yourself or could you use an existing binary distribution like the xPack version perhaps?

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/
    - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v14.2.0-2/
      - https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/download/v14.2.0-2/xpack-riscv-none-elf-gcc-14.2.0-2-darwin-arm64.tar.gz

If you do really need to build your own toolchain and are having problems then the first step is to provide the complete steps that you are taking (from a clean `git clone` through `configure` and `make`) and also capture and upload a build log for analysis (e.g. `make 2>&1 | tee build.log`).

However, bear in mind that most active participants on this repo do not have access to the necessary hardware in order to reproduce/investigate macOS build problems and  unfortunately no macOS users have stepped up to help here in this regard to date.

TommyMurphyTM1234, 2024-10-10:
And, another option is to use the xPack build mechanism:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/getting-started/#maintainer--developer-information

MarcoBonino, 2024-10-11:
@TommyMurphyTM1234  thanks for sharing useful information.
Right now it is ok to use a prebuilt toolchain, in case the build is really need I will come back in case I cannot build.

I'm using the toolchain you liked, but I'm getting a linker error: https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/37

Do you think is something about the toolchain itself or a gcc problem or something else ?

TommyMurphyTM1234, 2024-10-12:
- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1332#issuecomment-2408574447

wossoneri, 2024-10-14:
try this https://github.com/riscv-software-src/homebrew-riscv

TommyMurphyTM1234, 2024-10-17:
> try this https://github.com/riscv-software-src/homebrew-riscv

Did this help at all @MarcoBonino?

TommyMurphyTM1234, 2024-10-23:
Closing due to lack of activity.
If needed reopen and update with relevant info.

closed 2024-10-23 by TommyMurphyTM1234

MarcoBonino, 2024-10-29:
Hello! I just want to update the ticket with more info.
riscv-none-elf-gcc-xpack works, but it has a practical limitation when you want to do printf(): I already linked the discussion (https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/37)

https://github.com/riscv-software-src/homebrew-riscv was ok to use, but is not simply providing a zip with the toolchain.
In addition both tools are only providing the elf version of the toolchain and not the linux one.

Since I would like to build applications running on linux (and using things like threads) I would like to get a linux toolchain to run on macos. If there is a prebuilt binary I can use it, otherwise would be good to understand how to compile it on macos.

TommyMurphyTM1234, 2024-10-29:
> Hello! I just want to update the ticket with more info. riscv-none-elf-gcc-xpack works, but it has a practical limitation when you want to do printf(): I already linked the discussion ([xpack-dev-tools/riscv-none-elf-gcc-xpack#37](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/37))

That's not a limitation of the toolchain per se but of whatever board/target support package being used. The bare-metal toolchain can't a priori assume the existence of any specific resources/devices so stuff like this has to be supported on a custom basis using an appropriate board suport package and/or libraries.

> https://github.com/riscv-software-src/homebrew-riscv was ok to use, but is not simply providing a zip with the toolchain. 

I don't understand what this means.

> In addition both tools are only providing the elf version of the toolchain and not the linux one.
>
> Since I would like to build applications running on linux (and using things like threads) I would [...]

You might need to ask about this on a macOS specific forum - as I mentioned before, unfortunately non of the regular contributors here have access to macOS hardware in order to reproduce, diagnose and address macOS build issues and no macOS user has so far stepped up to help here in that respect.

TommyMurphyTM1234, 2024-10-29:
It would be ideal if somebody with macOS hardware access (all relevant types - e.g. x86 and Arm based?) and experience were able to volunteer instructions on how to use the different (?) macOS development environments/tools to build the RISC-V toolchains. It would certainly eliminate a lot of the repeated questions about build issues on macOS which usually cannot be answered clearly/authoritatively due to lack of such hardware and experience. But unfortunately nobody has volunteered to do this so these questions will presumably persist... :-|

ilg-ul, 2024-10-29:
Apple switched from gcc to clang some years ago, and since then the macOS support in upstream gcc is more and more problematic, to the point where the official sources no longer pass the builds for native gcc on recent macOS. :-(

Iain Sandoe maintains a gcc fork that fixes this (for example https://github.com/iains/gcc-14-branch), and he recommends using it even for cross-builds on macOS.

If someone volunteers to make this gcc fork build on macOS, probably must check Iain's fork.

TommyMurphyTM1234, 2024-10-29:
Thanks @ilg-ul, that's useful info. 

I also wonder if crosstool-NG might be another practical option? 

- https://crosstool-ng.github.io/docs/os-setup/

I've had good experiences with it for building cross toolchains to run on Linux and also on Windows but I have no experience with it in relation to macOS.

MarcoBonino, 2024-10-30:
@TommyMurphyTM1234 I have both aarch64 and x86 mac. If someone from this repo wants, we can setup a video call with screen sharing in about 2 weeks... 
Are you aware of any clang toolchains for riscv ? At least for now I just need to compile applications for riscv bare metal and running linux (and would be better if I can do it from my mac).

TommyMurphyTM1234, 2024-10-30:
> @TommyMurphyTM1234 I have both aarch64 and x86 mac. If someone from this repo wants, we can setup a video call with screen sharing in about 2 weeks... 

Thanks, but I wouldn't be able to do that myself. I don't know if anyone else might be able to do so?

> Are you aware of any clang toolchains for riscv ? 

This repo is capable of building an LLVM/Clang based toolchain:

- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#llvm--clang

> At least for now I just need to compile applications for riscv bare metal and running linux (and would be better if I can do it from my mac).

Did you look at the previous posts from @ilg-ul and me above in case they help at all?

MarcoBonino, 2024-10-30:
I checked them... right now the limitation is that under macos I cannot find a suitable linux toolchain.
Regarding building llvm/clang: the instructions are not working recently (at least after 2024.08.28):
we get` Makefile:51: *** commands commence before first target.  Stop.`
Anyway on macos on 2024.08.28 I get the same errors as in the original message of this issue.

TommyMurphyTM1234, 2024-10-30:
> I checked them... right now the limitation is that under macos I cannot find a suitable linux toolchain.

Unfortunately, so far I cannot find either instructions on how to build the `riscv-gnu-toolchain` Linux/glibc toolchain on macOS or a pre-built binary distribution that can be used as-is.

> we get` Makefile:51: *** commands commence before first target. Stop.` 

See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1598

> Anyway on macos on 2024.08.28 I get the same errors as in the original message of this issue.

Yes - the problem seems to be that the instructions for building stuff like this on macOS are simply not clear and documented.