riscv-collab/riscv-gnu-toolchain #1583: CMO not support

F-eng666666, 2024-10-17:
An error occurred when version 13.2.0 was used to compile files with cbo instructions
[image]

TommyMurphyTM1234, 2024-10-17:
The error seems correct - your operands are incorrect.

- https://github.com/riscv/riscv-CMOs/issues/47#issuecomment-112139224

This compiles/assembles OK for example:
```
/* test.c */
void f(void)
{
    asm("cbo.zero 0(x5)";
}
```

TommyMurphyTM1234, 2024-10-17:
FWIW...

- https://github.com/riscv/riscv-CMOs/blob/master/specifications/cmobase-v1.0.1.pdf

[image]

closed 2024-10-17 by TommyMurphyTM1234