riscv-collab/riscv-gnu-toolchain #1584: extension doesn't work as it was supported in GCC13

RRArinc, 2024-10-18:
Hi, I implemented custom instruction as i did in GCC 13, however toolchain builds successfully but fails to compile the custom instructions with below error.
**Error:
riscv32-unknown-elf-gcc: error: '-march=rv32imc_xwfe': extension 'xwfe' starts with 'x' but is unsupported non-standard extension**

May i know the GCC14 is ready to use and build custom instructions

Thanks

TommyMurphyTM1234, 2024-10-18:
How to add custom instructions to the toolchain is outside the scope of this repo. You may find some useful information in the existing open and closed issues and PRs here. But you need to familiarise yourself with the relevant GCC, Binutils, etc. documentation in order to understand how to customise the relevant toolchain components and, if necessary, all questions in the relevant upstream forums, mailing lists etc.

TommyMurphyTM1234, 2024-10-18:
Also, I think that non-standard/non-ratified custom extensions are supposed to be specified using an upper-case rather than a lower-case 'X' in the architecture string passed to tools?

TommyMurphyTM1234, 2024-10-18:
If there's a difference in behaviour between GCC 13 and 14 then you should probably look at the GCC sources and mailing list patches for changes to how the architecture string is processed and custom non-standard extensions are detected.

TommyMurphyTM1234, 2024-10-18:
What does `riscv64-unknown-elf-gcc -march=help` show for your toolchain?

- https://gcc.gnu.org/gcc-14/changes.html#:~:text=%2Dmarch%3Dhelp%20option%20has%20been%20introduced%20to%20dump%20all%20supported%20extensions.

If it doesn't list your custom extension then you probably need to tweak something in GCC 14 to enable it.

RRArinc, 2024-10-18:
Hi,
Yes i don't see the custom extension, please can you help/share me link how to make it happen.
You said something to tweak in GCC14, please can you help me with more information or links.

Thanks in Advance

TommyMurphyTM1234, 2024-10-18:
You should probably ask about this on one of the GCC mailing lists or some similarly appropriate forum:

- https://gcc.gnu.org/lists.html

Specifically:

1. Why you are seeing different behaviour between GCC 13 and 14
2. What patch(es) changed this behaviour (I haven't been able to identify this so far)
3. How to modify your customisations to the toolchain to allow your custom extension to be used with GCC 14

Liaoshihua, 2024-10-21:
I'm guessing you forgot to add something when you implement your extensions.
When you implement a new riscv extension in  gcc, you should add their   **[RISCV_EXT_FLAG_ENTRY](https://github.com/gcc-mirror/gcc/blob/0b6d94ce72b2f35dbee7c42774d6972671c86f97/gcc/common/config/riscv/riscv-common.cc#L1587)**  and **[riscv_ext_version](https://github.com/gcc-mirror/gcc/blob/0b6d94ce72b2f35dbee7c42774d6972671c86f97/gcc/common/config/riscv/riscv-common.cc#L239)**  in riscv-common.cc . Otherwise, an error will be reported as you shown

RRArinc, 2024-10-21:
Hi Liaoshihua,
Is this something new in GCC14? reason same extension code (binutils and GDB) working in GCC 13 and the same code is ported for GCC14.
Please can u confirm?

Thanks

TommyMurphyTM1234, 2024-10-21:
> Is this something new in GCC14? 

You should be able to answer this question by studying the GCC GitHub/Git history and other metadata.

RRArinc, 2024-10-22:
Hi Liaoshihua,
Ok. then this is something new as part of GGC-14 which was not required in GCC-13.
In GCC-13, i implemented the custom instruction only in binutils, GDB and bfd (bfd is one responsible for custom flags)

P.S: I have not implemnted your suggestions yet, definetly will give a try.

Hi Murphy,
Yes i see the difference b/w GCC13 and GCC14 source in riscv-common.cc where new cheks are added for extension 's'. 'x' etc.

Thanks.

TommyMurphyTM1234, 2024-10-27:
Any update @RRArinc?

TommyMurphyTM1234, 2024-10-30:
Closing due to lack of activity/update.

closed 2024-10-30 by TommyMurphyTM1234

RRArinc, 2024-11-21:
Hi,
Sorry for being late due to sick.

Yeah i gave a try to confirm the changes in GCC-14.
I see there is a check for extension 'x' flag, but i don't see the same in GCC-13.3.0 and below version.

Thanks

RRArinc, 2025-04-09:
Hi, Thanks for the support

Issue resolved.

Regards