riscv-collab/riscv-gnu-toolchain #1585: Use compiler toolchain in web with webAssembly

Jotaceee, 2024-10-18:
Hey, I just wanna adapt this toolchain to use the elf-as and elf-ld in web and i want to compile it with WebAssembly. 

Is it possible? 

Or there is another way to use it in a custom web?

It is as personal project to integrate this kind of tools in web.

Regards,

TommyMurphyTM1234, 2024-10-18:
> Hey, I just wanna adapt this toolchain to use the elf-as and elf-ld

What exactly do you mean by `elf-as` and `elf-ld`?

The readme explains how to build the toolchains in this repo and what the prerequisites are. 

- https://github.com/riscv-collab/riscv-gnu-toolchain

WebAssembly isn't mentioned there. 

I don't really understand what your aim is here exactly. 

There may be other efforts/projects relating to the intersection of RISC-V and WebAssembly that may be a better place to start and maybe ask. E.g.

- https://github.com/d0iasm/rvemu
- https://www.phoronix.com/news/Wasmer-3.2
- https://github.com/lupyuen/tcc-riscv32-wasm
- Etc...

Jotaceee, 2024-10-18:
Yes i understand, but i want to know if there is the possible to use the toolchain on a web. Not just local. And referred to elf-as and elf-ld, I refer to use only the binaries riscv64-unknown-elf-as and riscv64-unknown-elf-ld.

TommyMurphyTM1234, 2024-10-18:
> i want to know if there is the possible to use the toolchain on a web.

What exactly do you mean by this? 

The toolchain is just a set of binaries. What you do with it and how/where you run it is up to you. 

For what it's worth, there is already a web based form of the `riscv-gnu-toolchain` toolchains in the form of Compiler Explorer:

- https://godbolt.org/

TommyMurphyTM1234, 2024-10-18:
> / What exactly do you mean by this?
> 
> I mean that use the toolchain on web like the links you send me. Because i've designed a application to run elf files in a simulator of riscv.

So use Compiler Explorer?
Or replicate the way that they do it?

closed 2024-10-18 by TommyMurphyTM1234

ppenzin, 2024-11-28:
@Jotaceee there was some effort to run clang in the browser for wasm target, see here: https://www.youtube.com/watch?v=5N4b-rU-OAA

Doing this for risc-v would probably be a bit more involved.