riscv-collab/riscv-gnu-toolchain #1587: regarding PIC option differences

yf13, 2024-10-20:
Hello there,

I am porting a Cortex-m4 project to rv32, the project uses position independent code for downloadable app modules. The project uses GNU arm toolchain with `-fpic -fno-plt -mno-pic-data-is-text-relative -msingle-pic-base` options for compilation.

I noticed that [issue#1521] and alike alredy explains that RiscV elf toolchain doesn't support PIC code generation and RiscV linux toolchain should be used instead.

Then I tried both `riscv32-unknown-linux-gnu-gcc` and `riscv64-unknown-linux-musl-gcc` but found that they both:

- accept the `-fpic -fno-plt` options
- reject the `-mno-pic-data-is-text-relative -msingle-pic-base` options.

I guess the first two are general GCC options and the last two are ARM specific options.
I am unsure if RiscV linux toolchain:

- has same function for `-fpic -fno-plt` options like ARM toolchain?
- has any equivalent options for `-mno-pic-data-is-text-relative -msingle-pic-base` ARM toolchain? if no such options, then what is RiscV toolchain's behavior regarding these two options?

TommyMurphyTM1234, 2024-10-20:
> I guess the first two are general GCC options 

Correct: 

- https://gcc.gnu.org/onlinedocs/gcc-14.2.0/gcc/Code-Gen-Options.html

> and the last two are ARM specific options.

Correct:

- https://gcc.gnu.org/onlinedocs/gcc-14.2.0/gcc/ARM-Options.html

> I am unsure if RiscV linux toolchain: ...
> has same function for `-fpic` `-fno-plt` options like ARM toolchain?

It does. 

> has any equivalent options for `-mno-pic-data-is-text-relative` `-msingle-pic-base` ARM toolchain?

It doesn't seem to:

- https://gcc.gnu.org/onlinedocs/gcc-14.2.0/gcc/RISC-V-Options.html

> what is RiscV toolchain's behavior regarding these two options?

It doesn't accept the Arm specific options. And it doesn't seem to have any equivalent capability. Not all GCC target architectures support the same compiler options. 

Why not simply try compiling without those Arm specific options?

yf13, 2024-10-20:
Thank you @TommyMurphyTM1234 for the prompt reply.

So I will take that RiscV linux toolchain support `-fpic -fno-plt` but not `-mno-pic-data-is-text-relative -msingle-pic-base`, meaning the RiscV PIC feature has the assumption of `data segments are relative to text segment`.

TommyMurphyTM1234, 2024-10-20:
> So I will take that RiscV linux toolchain support `-fpic -fno-plt` but not `-mno-pic-data-is-text-relative -msingle-pic-base`, 

Yes, that's what the GCC documentation says. 

> meaning the RiscV PIC feature has the assumption of `data segments are relative to text segment`.

I don't know. You might want to ask about this on the ` gcc-help` mailing list. 

- https://gcc.gnu.org/lists.html

TommyMurphyTM1234, 2024-10-20:
In case these are of any use...

- https://github.com/riscvarchive/riscv-gcc/issues/158
- https://github.com/riscv-non-isa/riscv-elf-psabi-doc/issues/128

aswaterman, 2024-10-20:
The existing RISC-V PIC scheme requires the GOT and local data be at a fixed offset from text. 
 IIRC, people have made proposals for an ABI that supports your use case, but they haven't yet been accepted, let alone implemented.

TommyMurphyTM1234, 2024-10-23:
Closing this since it is something that needs to be dealt with in the relevant RISC-V ABI specification forum - here I think?

- https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154769/RISC-V+Technical+Specifications
  - https://lists.riscv.org/g/apps-tools-software
  - https://github.com/riscv-non-isa/riscv-elf-psabi-doc

And then any concomitant changes implied by the ABI changes/enhancements implemented in the upstream toolchain projects.

closed 2024-10-23 by TommyMurphyTM1234

aswaterman, 2024-10-23:
Sorry for the delay in responding.  Yes, the RISC-V ELF psABI spec is where these issues are discussed.

TommyMurphyTM1234, 2024-10-23:
> Sorry for the delay in responding. Yes, the RISC-V ELF psABI spec is where these issues are discussed.

Thanks a lot @aswaterman. 👍

ZhekaS, 2025-02-12:
It appears that currently accesses to `.bss` variables are not PC-relative but GOT-based, so I guess the actual behavior is consistent with `-mno-pic-data-is-text-relative`. While it would be nice to have it configurable, as this behavior is messing things up when we want to preserve the GOT entries for the actual absolute symbols...