riscv-collab/riscv-gnu-toolchain #1590: Building Linux with make binded-pkg fails after 6.12.0-rc4

drmpeg, 2024-10-22:
Reporting this issue here because it will probably come up eventually. The way that ```make bindeb-pkg``` works has been changed in Linux 6.12.0-rc1 with this patch.

https://lore.kernel.org/linux-kernel/20240727074526.1771247-5-masahiroy@kernel.org/

With the risc-gnu-toolchain, it fails to build because ```openssl/opensslv.h``` is not available in the cross compiler include directories. I reported the problem here.

https://lore.kernel.org/linux-kernel/b3d4f49e-7ddb-29ba-0967-689232329b53@w6rz.net/

A fix has been posted today.

https://lore.kernel.org/linux-kernel/20241022181703.1710116-2-masahiroy@kernel.org/

The recommendation is to use ```DEB_BUILD_PROFILES=pkg.linux-upstream.nokernelheaders``` or to install the libssl-dev package for the target architecture.

I don't think installing the libssl-dev RISC-V package will work for risc-gnu-toolchain. Any ideas about the correct solution here?

TommyMurphyTM1234, 2024-10-22:
Could it be that the Linux headers need to be updated to address this?

- https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master/linux-headers/include

They haven't been updated since Linux kernel 6.6:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1373

drmpeg, 2024-10-22:
Thanks for the pointer. Not sure if a refresh is required, but we would have to add ```linux-headers/include/openssl/*.h``` and ```linux-headers/include/riscv64-linux-gnu/openssl/*.h```

TommyMurphyTM1234, 2024-10-23:
OK - just generating the headers for Linux kernel v6.11 doesn't help...
```
git clone https://github.com/torvalds/linux
cd linux
git checkout v6.11
make headers_install ARCH=riscv INSTALL_HDR_PATH=`pwd`/linux-headers
cd linux-headers/include
find . -name "*openssl*"
# No files found
```

TommyMurphyTM1234, 2024-10-23:
> I'm not sure how the OpenSSL headers should be installed for a cross compilation scenario such as this?

Maybe they need to be added to the `sysroot` that is created when building the RISC-V Linux toolchain?

cmuellner, 2024-10-23:
It looks like the Linux kernel has a build dependency on OpenSSL headers.

Adding OpenSSL into the riscv-gnu-toolchain project (i.e. building it and installing it into the sysroot) would be an option, but it would open the door for a lot of similar low-level libraries (jemalloc, libelf, ncurses, ...). This would massively shift the focus of this repo (from a project similar to crosstool-ng into the area of Buildroot and Yocto) and, therefore, is not a realistic option.

However, nobody can stop you from cross-compiling OpenSSL on your own and installing it into the sysroot. This OpenSSL Configure options might work (untested): `--cross-compile-prefix=mytoolchain/bin/ --prefix=mytoolchain/`.

TommyMurphyTM1234, 2024-10-23:
> However, nobody can stop you from cross-compiling OpenSSL on your own and installing it into the sysroot. This OpenSSL Configure options might work (untested): `--cross-compile-prefix=mytoolchain/bin/ --prefix=mytoolchain/`.

Thanks @cmuellner - I was slowly coming around to that conclusion myself... :-)

This is what I tried...
```
cd ~

# Build RISC-V Linux toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1590
cd issue-1590
./configure --prefix=`pwd`/installed-tools
make linux

# Cross compile OpenSSL for RISC-V Linux
# https://openssl-library.org/source/
wget https://github.com/openssl/openssl/releases/download/openssl-3.4.0/openssl-3.4.0.tar.gz
tar xzvf openssl-3.4.0.tar.gz
cd openssl-3.4.0
./Configure linux-generic64 --cross-compile-prefix=riscv64-unknown-linux-gnu- --prefix=$HOME/issue-1590/installed-tools/sysroot
PATH=$HOME/issue-1590/installed-tools/bin:$PATH make
PATH=$HOME/issue-1590/installed-tools/bin:$PATH make install

ls $HOME/issue-1590/installed-tools/sysroot/include/
openssl

 ls $HOME/issue-1590/installed-tools/sysroot/lib/libssl*
/home/user/issue-1590/installed-tools/sysroot/lib/libssl.a  /home/user/issue-1590/installed-tools/sysroot/lib/libssl.so  /home/user/issue-1590/installed-tools/sysroot/lib/libssl.so.3
```
This seems to successfully cross compile OpenSSL and install it in the RISC-V Linux toolchain's `sysroot` from where I presume that it can be used to cross compile stuff that needs OpenSSL - such as the Linux kernel.

drmpeg, 2024-10-23:
Works good. Tested with a build of Linux 6.12.0-rc4. A few notes:

1) The openssl install prefix should be ```--prefix=mytoolchain/sysroot/usr```

2) It's a good size project. make -j$(nproc) suggested.

3) If installing to the suggested ```/opt/riscv```, use ```sudo --preserve-env=PATH make install```

Should this be documented somewhere (aside from just leaving this issue open)?

TommyMurphyTM1234, 2024-10-23:
> Should this be documented somewhere 

I don't think that the readme can or should deal with such specifics. I think that if additional supplementary packages/libraries are needed when cross compiling stuff then the developer/user should know that they need to do this themselves.

> (aside from just leaving this issue open)?

I also don't think that this issue should be kept open. It doesn't actually identify any issue per se with the `riscv-gnu-toolchain` repo. And people should be searching all issues, open and closed, for info.

closed 2024-10-23 by drmpeg

drmpeg, 2024-11-10:
Just a small addendum. With Linux 6.12.0-rc6 and after, a couple more tricks are required.

1) make bindeb-pkg looks for libssl-dev:riscv64 to be present on your system. You can use ```DPKG_FLAGS=-d``` to suppress that.

2) make bindeb-pkg tries to invoke ```riscv64-linux-gnu-gcc``` to build the headers. You can work around that by symbolically linking to ```riscv64-unknown-linux-gnu-gcc``` somewhere in your path. For example, ```ln -s /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc riscv64-linux-gnu-gcc``` in ~/bin.