riscv-collab/riscv-gnu-toolchain #1597: risv-gnu-toolchain spike module "Bad or missing arguments for command str"

cryptolovi, 2024-10-27:
I'm running into problems with Spike reporting "Bad or missing arguments for command str", which I don't know how to resolve or why this is occurring in a simple hello world program.

The riscv-gnu-toolchain is cleanly build. The RISCV and PATH variable is set in the USER environment only (so using sudo -E). The environment variable are set RISCV=/opt/riscv and PATH includes $RISCV/bin. Compiling is done on  Ubuntu 24.04.1 LTS
Getting & Building riscv-gnu-toolchain is done by;
```
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain/
./configure --prefix=/opt/riscv --enable-multilib
sudo -E make
```
Which results in the riscv64-unknown-elf compiiler/tools in /opt/riscv.

I build pk from the riscv-gnu-toolchain directory with;
`sudo -E SIM=spike make build-sim`

Which results with the spike binaries in /opt/riscv/bin (spike, spike-dasm, spike-log-parser etc.) and pk32, pk64 in /opt/riscv/riscv64-unknown-elf/bin

The sample program;
```
#include <stdio.h>
int main() {
  printf("Hello World!\n");
}
```

The compile commando for the example program (with intermediate files enabled);
riscv64-unknown-elf-gcc --save-temps -o example example.c

The generated assembly;
```
.file	"example.c"
	.option nopic
	.attribute arch, "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0"
	.attribute unaligned_access, 0
	.attribute stack_align, 16
	.text
	.section	.rodata
	.align	3
.LC0:
	.string	"Hello World!"
	.text
	.align	1
	.globl	main
	.type	main, @function
main:
	addi	sp,sp,-16
	sd	ra,8(sp)
	sd	s0,0(sp)
	addi	s0,sp,16
	lui	a5,%hi(.LC0)
	addi	a0,a5,%lo(.LC0)
	call	puts
	li	a5,0
	mv	a0,a5
	ld	ra,8(sp)
	ld	s0,0(sp)
	addi	sp,sp,16
	jr	ra
	.size	main, .-main
	.ident	"GCC: () 14.2.0"
	.section	.note.GNU-stack,"",@progbits
```

The disassembly of the executable with riscv64-unknown-elf-objdump --disassemble=main example;
```
example:     file format elf64-littleriscv

Disassembly of section .text:

00000000000101d4 <main>:
   101d4:	1141                	addi	sp,sp,-16
   101d6:	e406                	sd	ra,8(sp)
   101d8:	e022                	sd	s0,0(sp)
   101da:	0800                	addi	s0,sp,16
   101dc:	67c9                	lui	a5,0x12
   101de:	61878513          	addi	a0,a5,1560 # 12618 <__errno+0x8>
   101e2:	384000ef          	jal	10566 <puts>
   101e6:	4781                	li	a5,0
   101e8:	853e                	mv	a0,a5
   101ea:	60a2                	ld	ra,8(sp)
   101ec:	6402                	ld	s0,0(sp)
   101ee:	0141                	addi	sp,sp,16
   101f0:	8082                	ret
```

The following command starts the example in spike;
spike -d pk64 example

In spike I run to the address of the jal instruction, display the contents of the a0 register (which should point to the string "hello world" and try to display it.  The following is the result;
```
(spike) until pc 0 101e2
bbl loader
(spike) reg 0 a0
0x0000000000012618
(spike) str 0 12618
Bad or missing arguments for command str
```
Omitting the core parameter for str commando has the same result. Trying the mem commando to dump the content hexidecimal gives; "0xBad or missing arguments for command mem"

What am I missing/doing wrong or not understanding?

cryptolovi, 2024-10-27:
From what I know, is that the hex address values for these commands str/mem/until etc don't have to be pre formatted with 0x.
I tried your suggestion however though;
```
(spike) until pc 0 0x101e2
bbl loader
(spike) reg 0 a0
0x0000000000012618
(spike) str 0 0x12618
Bad or missing arguments for command str
(spike) 
```
Which gives the same result.

Indeed a good idea to be asking it on the spike repo. I was assuming that since certain submodules are specifically tagged from their respective repo's that there would be more assurance or testing of the tooling within riscv-gnu-toolchain as a whole.

TommyMurphyTM1234, 2024-10-27:
> From what I know, is that the hex address values for these commands str/mem/until etc don't have to be pre formatted with 0x. 

Ah, sorry - you are correct - `strtol()` is passed a `base` of `16` so treats all strings as hex:

- https://github.com/riscv-software-src/riscv-isa-sim/blob/88fc84ded155a9e01987c4dfb7a77800e69b232b/riscv/interactive.cc#L735

```
  reg_t addr = strtol(addr_str.c_str(),NULL,16);
```

- https://cplusplus.com/reference/cstdlib/strtol/

> 
> I was assuming that since certain submodules are specifically tagged from their respective repo's that there would be more assurance or testing of the tooling within riscv-gnu-toolchain as a whole.

No - most submodule projects such as Spike/pk are simply taken from upstream as-is. The only difference would be the fact that the DejaGnu based GCC test suite is sometimes run on the actual toolchains. In general, specific questions about such submodule projects are better asked upstream. Questions about how `riscv-gnu-toolchain` builds and integrates them and issues that arise with this are suitable material for this repo.

cryptolovi, 2024-10-27:
> No - most submodule projects such as Spike/pk are simply taken from upstream as-is. The only difference would be the fact that the DejaGnu based GCC test suite is sometimes run on the actual [...]

Thank you for pointing this out. I did not know this. I have opened up the issue on the correct repo [issue on spike repo](https://github.com/riscv-software-src/riscv-isa-sim/issues/1845)

TommyMurphyTM1234, 2024-10-27:
Does it make any difference if you do `str 12618` instead of `str 0 12618`?

cryptolovi, 2024-10-27:
> Does it make any difference if you do `str 12618` instead of `str 0 12618`?

No, in the first message I wrote that omitting the core parameter doesn’t make a difference, so in this case leaving out the ‘0’.

TommyMurphyTM1234, 2024-10-27:
What if you use Spike's `mem` command to read memory from that address? Does that work?

cryptolovi, 2024-10-27:
> What if you use Spike's `mem` command to read memory from that address? Does that work?

I also tried that, see the initial message.

TommyMurphyTM1234, 2024-10-27:
If both `str` and `mem` are giving the same error then I wonder if it's failing when trying to read the memory, throwing an exception, and resulting in the slightly misleading error message. If it was me I'd probably add some instrumentation code/logging to the Spike code to try to home in on what's happening/going wrong.

cryptolovi, 2024-10-30:
I'm digging a bit more into the problem. If the example program is ran without the debug it works and it's able to address the string pointer. However in debug (-d with spike) I now get a "Received trap: trap_load_page_fault" with the newly bumped spike.
I'm unable to dump memory (str or mem) outside the .text segment with spike in interactive debugging (the string is in the .rodata section).

TommyMurphyTM1234, 2024-10-30:
OK - probably better to pursue these issues here (and maybe in additional issues) given that they seem to touch on the minutiae of how Spike operates:

- https://github.com/riscv-software-src/riscv-isa-sim/issues/1845

closed 2024-10-30 by TommyMurphyTM1234