riscv-collab/riscv-gnu-toolchain #1598: unable to build in mac

quantrpeter, 2024-10-28:
hi
   unable to build in mac
   
   ```
/Users/peter/workspace/riscv-gnu-toolchain>./configure --prefix=/opt/riscv
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... no
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
/Users/peter/workspace/riscv-gnu-toolchain>make
Makefile:51: *** commands commence before first target.  Stop.
```

thanks

TommyMurphyTM1234, 2024-10-28:
What does the generated `Makefile` contain? 

You might want to review existing open and closed issues in this repo that relate to macOS build issues in case they contain any useful information. E.g.:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+mac

Unfortunately none of the active contributors here have access to macOS systems in order to reproduce and investigate macOS build issues and no macOS user has stepped up to offer assistance so such issues can be difficult to diagnose efficiently and effectively.

quantrpeter, 2024-10-28:
Dear @TommyMurphyTM1234  here you go

```
srcdir := /Users/peter/workspace/riscv-gnu-toolchain
builddir := /Users/peter/workspace/riscv-gnu-toolchain
INSTALL_DIR := /opt/riscv

GCC_SRCDIR := $(srcdir)/gcc
BINUTILS_SRCDIR := $(srcdir)/binutils
NEWLIB_SRCDIR := $(srcdir)/newlib
GLIBC_SRCDIR := $(srcdir)/glibc
MUSL_SRCDIR := $(srcdir)/musl
UCLIBC_SRCDIR := $(srcdir)/uclibc-ng
LINUX_HEADERS_SRCDIR := $(srcdir)/linux-headers/include
GDB_SRCDIR := $(srcdir)/gdb
QEMU_SRCDIR := $(srcdir)/qemu
SPIKE_SRCDIR := $(srcdir)/spike
PK_SRCDIR := $(srcdir)/pk
LLVM_SRCDIR := $(srcdir)/llvm
DEJAGNU_SRCDIR := $(srcdir)/dejagnu
DEBUG_INFO := 
ENABLE_DEFAULT_PIE := --disable-default-pie
DEJAGNU_SRCDIR := $(srcdir)/dejagnu

SIM ?= qemu

# Shared lib suffix
IS_DARWIN := $(shell uname -s | grep Darwin)
SHARED_LIB_SUFFIX := $(if $(IS_DARWIN),dylib,so)

ifeq ($(srcdir)/gcc,$(GCC_SRCDIR))
# We need a relative source dir for the gcc configure, to make msys2 mingw64
# builds work.  Mayberelsrcdir is relative if a relative path was used to run
# configure, otherwise absolute, so we have to check.
mayberelsrcdir := .
gccsrcdir := $(shell case $(mayberelsrcdir) in \
		  ([\\/]* | ?:[\\/]*)  echo $(mayberelsrcdir)/gcc ;; \
		  (*)  echo ../$(mayberelsrcdir)/gcc ;; \
		esac)
else
gccsrcdir := $(abspath $(GCC_SRCDIR))
endif

WITH_ARCH ?= --with-arch=rv64imafdc
WITH_ABI ?= --with-abi=lp64d
WITH_TUNE ?= --with-tune=rocket
WITH_ISA_SPEC ?= --with-isa-spec=20191213
SYSROOT := $(INSTALL_DIR)/sysroot
ENABLE_LIBSANITIZER ?= --disable-libsanitizer
QEMU_TARGETS ?= riscv64-linux-user,riscv32-linux-user

ENABLED_LANGUAGES ?= 
ifeq ($(ENABLED_LANGUAGES),)
	undefine ENABLED_LANGUAGES
endif

SHELL := /bin/sh
AWK := /usr/local/bin/gawk
SED := /usr/local/bin/gsed
PATH := $(INSTALL_DIR)/bin:$(PATH)

# Check to see if we need wrapper scripts for awk/sed (which point to
# gawk/gsed on platforms where these aren't the default), otherwise
# don't override these as the wrappers don't always work.
ifneq (/usr/local/bin/gsed, $(shell realpath $(shell which sed)))
	PATH := $(builddir)/scripts/wrapper/sed:$(PATH)
endif
ifneq (/usr/local/bin/gawk, $(shell realpath $(shell which awk)))
	PATH := $(builddir)/scripts/wrapper/awk:$(PATH)
endif
[... 1302 lines cut ...]
	stat $(patsubst %,$(srcdir)/test/gdb-newlib/%.log,$(NEWLIB_MULTILIB_NAMES)) || exit 1
# Fail if there are blank lines in the log file used as input for grep below.
	if grep '^$$' $(patsubst %,$(srcdir)/test/gdb-newlib/%.log,$(NEWLIB_MULTILIB_NAMES)); then exit 1; fi
	if find build-gdb-newlib -iname '*.sum' | xargs grep ^FAIL | sort | grep -F -v $(patsubst %,--file=$(srcdir)/test/gdb-newlib/%.log,$(NEWLIB_MULTILIB_NAMES)); then false; else true; fi

.PHONY: report-gdb-linux
report-gdb-linux: stamps/check-gdb-linux
	stat $(patsubst %,$(srcdir)/test/gdb-linux/%.log,$(GLIBC_MULTILIB_NAMES)) || exit 1
# Fail if there are blank lines in the log file used as input for grep below.
	if grep '^$$' $(patsubst %,$(srcdir)/test/gdb-linux/%.log,$(GLIBC_MULTILIB_NAMES)); then exit 1; fi
	if find build-gdb-linux -iname '*.sum' | xargs grep ^FAIL | sort | grep -F -v $(patsubst %,--file=$(srcdir)/test/gdb-linux/%.log,$(GLIBC_MULTILIB_NAMES)); then false; else true; fi

distclean: clean
	rm -rf src

# All of the packages install themselves, so our install target does nothing.
install:

# Rebuilding Makefile.
Makefile: $(srcdir)/Makefile.in config.status
	CONFIG_FILES=$@ CONFIG_HEADERS= $(SHELL) ./config.status

config.status: $(srcdir)/configure
	CONFIG_SHELL="$(SHELL)" $(SHELL) ./config.status --recheck
```

TommyMurphyTM1234, 2024-10-28:
It would be much better to attach rather than embed such large files. That way the contents are preserved and it doesn't clutter up the thread. 

The error `commands commence before first target` usually (always?) occurs when there is a rogue tab in the `Makefile`. E.g.:

- https://stackoverflow.com/questions/4713663/gnu-make-yields-commands-commence-before-first-target-error

Are you sure that you're cloning the very latest contents of this repo, you're not editing `Makefile.in` or `Makefile` or any other files, and nothing else is modifying their contents automatically? 

If your generated `Makefile` is ending up with a rogue tab character that causes this problem then, unfortunately, I have no idea why this might be happening. And, as I said, unfortunately no way to reproduce/diagnose this directly.

Maybe try starting from scratch to see if it happens again. E.g.
```
cd ~
mkdir test
cd test
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv
make
```

TommyMurphyTM1234, 2024-10-28:
This seems to be the case of the error:

[image]

I don't understand why because I have the same tab in my generated `Makefile` on Ubuntu 24.04 and the build proceeds fine. 

Maybe there is some difference between the Linux `make` and the one used on macOS?

TommyMurphyTM1234, 2024-10-28:
Bit of a stab in the dark, but...
Does it make any difference if you remove the space at the end of line 49?

[image]

TommyMurphyTM1234, 2024-10-28:
Actually, it doesn't seem to be a trailing space but a CR/LF Windows/DOS style line terminator:

[image]

Did you perhaps clone the sources on a Windows/DOS system or NTFS/FAT[32] filesystem and then try to use them on macOS?

TommyMurphyTM1234, 2024-10-28:
Please start afresh as I suggested earlier and then upload/attach (not embed) the `Makefile.in` that is checked out and then the `Makefile` generated by the `configure` step.

quantrpeter, 2024-10-28:
Hi, i redo again in mac, here are the files.

[Archive.zip](https://github.com/user-attachments/files/17542147/Archive.zip)

Here is the video:  https://www.youtube.com/watch?v=kMN1WMY6MfM

TommyMurphyTM1234, 2024-10-28:
Thanks. Unfortunately I can't identify what's going wrong  at your end to cause this problem. You might need to ask about this on a more macOS specific forum where you might get some useful feedback. As I said, unfortunately there hasn't been much in the way of support from macOS users here on this repo and most others who are active don't have access to macOS systems in order to reproduce and investigate such issues.

BTW - your use of a case insensitive filesystem may cause other problems with your build and certainly with a build of the Linux toolchain which requires a case senstitive system. I seem to recall that there is some way to tell macOS to enable case sensitivity but I don't know the details offhand.

TommyMurphyTM1234, 2024-10-28:
Just a thought - perhaps this PR adversely impacted the macOS build as it introduced the lines in `Makefile.in` that seem to be causing problems here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1559

Can you try again with the following:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/usr/opt --with-languages="c.c++"
make
```

Arielfoever, 2024-10-28:
OK let me see what happened? But I don't have a Mac.

quantrpeter, 2024-10-28:
1. only this work : ./configure --prefix=/opt/riscv --with-languages="c,c++"
1. manually remove space after ?= not work
1. manually paste your code to Makefile.in not work
thanks

TommyMurphyTM1234, 2024-10-28:
> 1. only this work : ./configure --prefix=/opt/riscv --with-languages="c,c++"
> 2. manually remove space after ?= not work
> 3. manually paste your code to Makefile.in not work
>    thanks

Thanks @quantrpeter - looks like that PR may have caused the problem for some reason.

Arielfoever, 2024-10-28:
Sounds like make on mine is different from your Mac. how about revert it and try again?

quantrpeter, 2024-10-28:
which version i should revert to? thanks

TommyMurphyTM1234, 2024-10-28:
Whave version of `make` are you using at the moment?

quantrpeter, 2024-10-28:
```
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
```

TommyMurphyTM1234, 2024-10-28:
For now you should probably just use `configure ... --with-languages="c,c++"` for the bare-metal toolchain or `configure ... --with-languages="c,c++,fortran"` for the Linux toolchain as a workaround. (For at least the Linux toolchain you will also need to address the filesystem case insensitivity issue mentioned earlier).

quantrpeter, 2024-10-28:
bingo, after i use homebrew make, which is 4.4.1, then problem gone, thank you Sir

closed 2024-10-28 by quantrpeter

TommyMurphyTM1234, 2024-10-28:
> bingo, after i use homebrew make, which is 4.4.1, then problem gone, thank you Sir

Ok - that's interesting. Thanks for clarifying. Maybe the closed issue will help others who might encounter the same issue. 👍