riscv-collab/riscv-gnu-toolchain #1604: `uclibc-ng` submodule needs a `sync`

TShapinsky, 2024-11-01:
Current failure in submodule update looks like this.
```
remote: Enumerating objects: 4924, done.
remote: Counting objects: 100% (4924/4924), done.
remote: Compressing objects: 100% (4020/4020), done.
remote: Total 4924 (delta 2040), reused 1695 (delta 762), pack-reused 0
Receiving objects: 100% (4924/4924), 3.87 MiB | 4.37 MiB/s, done.
Resolving deltas: 100% (2040/2040), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: Server does not allow request for unadvertised object ecbc412d0d93c8b7b77bec6539b290c9da5401cf
fatal: Fetched in submodule path 'uclibc-ng', but it did not contain ecbc412d0d93c8b7b77bec6539b290c9da5401cf. Direct fetching of that commit failed.
fatal:
```

Running `git submodule sync` generates the change:
```
diff --git a/uclibc-ng b/uclibc-ng
index ecbc412..f8028af 160000
--- a/uclibc-ng
+++ b/uclibc-ng
@@ -1 +1 @@
-Subproject commit ecbc412d0d93c8b7b77bec6539b290c9da5401cf
+Subproject commit f8028afeaffd727bc335681491b25a6f8fe51f4d
```

After which it works. I didn't make a PR for this as it's possible there's an underlying problem or other decision which should be made.

TommyMurphyTM1234, 2024-11-01:
I don't understand, `ecbc412d0d93c8b7b77bec6539b290c9da5401cf` is the correct commit hash for uClibc-ng release v1.0.50:

- https://cgit.uclibc-ng.org/cgi/cgit/uclibc-ng.git/tag/?h=v1.0.50

Which is why I used it for this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1564

TShapinsky, 2024-11-01:
> I don't understand. `ecbc412d0d93c8b7b77bec6539b290c9da5401cf` is the correct commit hash for uClibc-ng release v1.0.50:
> 
>     * https://cgit.uclibc-ng.org/cgi/cgit/uclibc-ng.git/tag/?h=v1.0.50
> 
> 
> Which is why I used it for this PR:
> 
>     * [Bump uClibc from v1.0.48-7-gad21b95c9 to v1.0.50 #1564](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1564)

Yeah, sorry. I closed the issue because I realized I was wrong. I'm still trying to figure out why [this ](https://github.com/TShapinsky/riscv-gnu-toolchain/actions/runs/11633132560/job/32408436893) is failing to clone the submodule when `--depth 1` is present.

TommyMurphyTM1234, 2024-11-01:
> Yeah, sorry. I closed the issue because I realized I was wrong. I'm still trying to figure out why [this ](https://github.com/TShapinsky/riscv-gnu-toolchain/actions/runs/11633132560/job/32408436893) is failing to clone the submodule when `--depth 1` is present.

Ah, ok, thanks. I wasn't sure why the issue was closed either. So the problem seems to be that when the clone depth is 1 it seems to trigger some issue with uClibc-ng? If I can do anything to investigate let me know. 👍