riscv-collab/riscv-gnu-toolchain #1610: How to add non-standard custom-instruction to the toolchain

claudio-frongia, 2024-11-06:
Hello everyone!
I need to update the riscv-gnu-toolchain to recognize new custom instructions with two destination registers (starting with the R4 format where rs3 is used as the write register/second destination register). 
My goal is to obtain hex/bin files for use in a 32-bit custom RISC-V processor that supports those custom instructions. 
So, the hex/bin files generated must contain the correct encodings of my instructions in order to take advantage of the dedicated hardware in the processor. 

**I am following this strategy:**
-  git clone --recursive https://github.com/pulp-platform/riscv-gnu-toolchain.git (probably not the most up-to-date version)
- configuration with rv32imf (for R4 format support) and --enable-multilib and build the entire toolchain in the /home/name/project1/riscv-gnu-toolchain folder
- _binutils modifications_: adding mask, match, and declare for my instructions in riscv-opc.h and defining my instructions and registers used in riscv-opc.c (in this case I use a d,s,t,d syntax to indicate the two target registers “d”)
- at this point, given also the solution to other similar issues, the changes should be finished and then I recompile the toolchain (it always takes a while).

At this point I get the following error from the assembler:

**Assembler messages: Error: internal: bad RISC-V opcode (bits 0xf80000 undefined or invalid): cust1 d,s,t,d Fatal error: internal: broken assembler. 
No assembly attempted Internal error (Segmentation fault). 
Please report this bug.**

It would appear that the bits associated with the rs3 register (31..27) are not recognized correctly. 
I don't get this error if I use d,s,t,r but I don't think this is the best solution for use two destination registers.
I am wondering at this point if it is possible to achieve my goal by only modifying the binutilis (riscv-opc.h and riscv-opc.c), 
without modifying other parts of the binutils or gcc to define the correct format (riscv.md, riscv.h, riscv-dis.c and maybe others).
Also, does modifying only binutils involve using inline assembly in the C code to specify input and output registers and full encoding with “.word”, or are there similar strategies that allow the compiler to recognize the custom instruction without using inline and without modifying the gcc backend?
I would be ready to also modify riscv.md or other files in gcc to achieve my purpose, however at the moment I do not know if it is necessary and I am open to advice and alternative solutions.
Thank you

TommyMurphyTM1234, 2024-11-06:
The details of how to customise the toolchain to add custom instructions are outside the scope of this repo and, fundamentally, this is not a RISC-V or `riscv-gnu-toolchain` specific question. Having said that, there are several (mostly closed) issues in this repo that may contain some useful information. E.g.:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+custom+instruction

 However, ultimately, for help on this you really need to post in a more appropriate forum such as the GCC mailing lists.

- https://gcc.gnu.org/lists.html

ideally after first familiarising yourself with the relevant GCC internals documentation:

- https://gcc.gnu.org/onlinedocs/gccint/

claudio-frongia, 2024-11-06:
Thank you very much for your reply, I am still studying the GCC documentation and have ideas on how to go about modifying gcc. 
But it was my intention to know if anyone has already faced and solved my problem. 
I will also post my question on the GCC mailing list.
Thanks again @TommyMurphyTM1234

claudio-frongia, 2024-11-06:
Thank you again @TommyMurphyTM1234 
I'm still studying an appropriate strategy, if it's inappropriate to keep my issue open in this channel then I'll close it.

TommyMurphyTM1234, 2024-11-06:
Your issue here is almost certainly down to some mistake in how you customised the toolchain to support your custom instructions and not down to any bug. As such, it makes more sense to close it and pursue a solution in other channels as suggested previously. Unless you can show evidence of a suspected bug in the toolchain ideally with a minimal reproducible test case and clear instructions on how to reproduce the issue.

claudio-frongia, 2024-11-06:
Perfect, I'm closing this issue and looking for support in other channels, thank you!

closed 2024-11-06 by claudio-frongia

NAbuShaibeh, 2024-11-18:
Hello @claudio-frongia 
I am experiencing the same problem you had when I tried to add a new custom instruction, and I haven't found a solution yet for this issue.
```
Assembler messages:
Error: internal: bad RISC-V opcode (bits 0xffffffff00000000 undefined or invalid): lws d,o(s)
Fatal error: internal: broken assembler.  No assembly attempted
Internal error (Segmentation fault).
Please report this bug.
```
If you have solved it, please post the solution. 
Thank you in advance!

TommyMurphyTM1234, 2024-11-18:
> Hello @claudio-frongia I am experiencing the same problem you had when I tried to add a new custom instruction, and I haven't found a solution yet for this issue.
>
> ```
> Assembler messages:
> Error: [...]

As the error message clearly states, there is some problem with the changes that you made to the toolchain to add support for your custom instruction.

You should search for similar issues in case there's any useful information there, e.g.:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+bad+RISC-V+opcode

and not then you should ask about this upstream - e.g. on the GCC mailing lists or a similar forum.