riscv-collab/riscv-gnu-toolchain #1611: .gitmodules does not reflect changes in LLVM submodule

JohRust, 2024-11-07:
The LLVM submodule was updated to 18.1.8 in commit https://github.com/riscv-collab/riscv-gnu-toolchain/commit/97fce6f183992f2823caa19fe2c03c2eb7ec63c6 and now points to https://github.com/llvm/llvm-project/tree/3b5b5c1ec4a3095ab096dd780e84d7ab81f3d7ff, the latest commit of branch "release/18.x". However .gitmodules still points to branch "release/17.x":
```
[submodule "llvm"]
	path = llvm
	url = https://github.com/llvm/llvm-project.git
	branch = release/17.x
```

TommyMurphyTM1234, 2024-11-07:
Thanks @JohRust - and I see that you've also posted a PR. 👍 

Can you clarify what, if anything, was the symptom of this mismatch between the `.gitmodules` file and the submodule commit?
Or was it simply something that you noticed "manually"?

Thanks.

JohRust, 2024-11-07:
Hi @TommyMurphyTM1234!
I noticed it "manually". I was using the toolchain with LLVM and simply wanted to check what LLVM version I was using. The mismatch seemed like a minor oversight similar to #1575. Keep in mind I have still limited knowledge about this project, I only checked if "git submodule update --remote llvm" still refers to the same LLVM commit after my change. So let me know if you have any concerns or need more tests. :)

TommyMurphyTM1234, 2024-11-07:
Thanks @JohRust - I was just curious because I recall similar mismatches for other submodules and not noticing any substantive issue. But it's a good spot and worth fixing to avoid confusion. I had a glance at your PR and it looks fine but I'll check it properly in a little while. Thanks for spotting and addressing this. 👍

PR #1612 (2024-11-07): Update .gitmodules to reflect LLVM bump to 18.1
Fixes #1611 
The LLVM submodule was updated to 18.1 in commit 97fce6f and now points to https://github.com/llvm/llvm-project/tree/3b5b5c1ec4a3095ab096dd780e84d7ab81f3d7ff. Make .gitmodules point to that branch as well.
closed 2024-11-07 by cmuellner