riscv-collab/riscv-gnu-toolchain #1613: Repository does not clone

mateoconlechuga, 2024-11-08:
Due to changes in the EDK2 submodule (https://github.com/tianocore/edk2/issues/6398) this repository will no longer clone / checkout submodules correctly. It looks like a  backport to the stable version may be provided at some time. Not sure how this should be handled

TommyMurphyTM1234, 2024-11-08:
I presume you mean this?
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9405, done.
remote: Counting objects: 100% (399/399), done.
remote: Compressing objects: 100% (108/108), done.
remote: Total 9405 (delta 319), reused 331 (delta 291), pack-reused 9006 (from 1)
Receiving objects: 100% (9405/9405), 6.43 MiB | 2.70 MiB/s, done.
Resolving deltas: 100% (4679/4679), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Submodule 'uclibc-ng' (https://git.uclibc-ng.org/git/uclibc-ng.git) registered for path 'uclibc-ng'

...

Cloning into '/home/user/riscv-gnu-toolchain/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook'...
Username for 'https://github.com':
```

mateoconlechuga, 2024-11-08:
Yes that's right

TommyMurphyTM1234, 2024-11-08:
FWIW QEMU builds even with the clone failures/errors above.
A possible (temporary) workaround might be to explicitly exclude the affected sub(sub?)modules when cloning `riscv-gnu-toolchain`?

TommyMurphyTM1234, 2024-11-08:
No problem occurs if you do this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-sim=qemu
make build-sim

...

Installing /home/user/riscv-gnu-toolchain/qemu/pc-bios/keymaps/th to /home/user/riscv-gnu-toolchain/installed-tools/share/qemu/keymaps
Installing /home/user/riscv-gnu-toolchain/qemu/pc-bios/keymaps/tr to /home/user/riscv-gnu-toolchain/installed-tools/share/qemu/keymaps
make[1]: Leaving directory '/home/user/riscv-gnu-toolchain/build-qemu'
mkdir -p stamps/
date > stamps/build-qemu
```
It only seems to be a problem if you do this?
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
...
```

TommyMurphyTM1234, 2024-11-08:
I see that you have logged the issue in upstream QEMU which is the correct place for it:

- https://gitlab.com/qemu-project/qemu/-/issues/2660

Since the issue doesn't impact `riscv-gnu-tooolchain` unless one does a recursive clone from the top, I'm not sure that the issue needs to remain open here?

mateoconlechuga, 2024-11-08:
Sounds good to me

closed 2024-11-08 by mateoconlechuga