riscv-collab/riscv-gnu-toolchain #1618: objdump has bug, it can't dump out all sections

quantrpeter, 2024-11-17:
Hi All, objdump has bug, it can't dump out all sections, thanks

[image]

[image]

[rv64ui-v-addi.zip](https://github.com/user-attachments/files/17791208/rv64ui-v-addi.zip)

TommyMurphyTM1234, 2024-11-17:
If you want `objdump` to dump all sections then you need to use `-D` or `--disassemble-all`:

- https://sourceware.org/binutils/docs/binutils/objdump.html#:~:text=used%20when%20disassembling.-,%2DD,%2D%2Ddisassemble%2Dall,-Like%20%2Dd%2C%20but

If you really think that there's a bug on `objdump` and you can provide firm evidence of this - including a reproducible test case and instructions on how to exercise it - then you should log a bug in the upstream Binutils project where ongoing maintenance is done these days:

- https://www.gnu.org/software/binutils/
  - https://sourceware.org/bugzilla/

quantrpeter, 2024-11-17:
-D still the same, thank you, let me submit to binutils