riscv-collab/riscv-gnu-toolchain #1620: make change to binutils, don't build

quantrpeter, 2024-11-19:
hi
   make a change to binutils, changed the dwarf.c, then i type "make", it don't build. It said everything is updated.

thanks
Peter

TommyMurphyTM1234, 2024-11-19:
The top level `riscv-gnu-toolchain` `Makefile` isn't structured to detect such changes to submodule component sources and do an incremental build.

You'll have to force an appropriate rebuild.

Try this to just rebuild Binutils:
```
rm stamps/build-binutils*
make 
```
If that doesn't help then you may need to force a complete clean build: 
```
make distclean
rm -rf <prefix-dir>
make 
```

TommyMurphyTM1234, 2024-11-22:
Any update @quantrpeter?
Did what I suggested help?

quantrpeter, 2024-11-24:
Dear Tommy, yes it works. But rm stamps/build-binutils* will rebuild the whole binutils, rather than rebuild my changed file. So it takes me long time to complete. thanks

TommyMurphyTM1234, 2024-11-24:
That's just the way that the top level `riscv-gnu-toolchain` `Makefile` works. It's not set up to do incremental builds of submodule components. Pull requests are always welcome if you come up with a way to enable this from the top level `Makefile`. Maybe you can do an incremental build by changing to the Binutils directory and invoking `make` with the appropriate arguments manually?

TommyMurphyTM1234, 2024-11-25:
This seems to work:
```
# Initial build of toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1620
cd issue-1620
./configure --prefix=`pwd`/installed-tools
make

# Update Binutils
touch binutils/binutils/objdump.c
cd build-binutils-newlib
make install
...
make[4]: Leaving directory '/home/user/issue-1620/build-binutils-newlib/binutils/po'
make[4]: Entering directory '/home/user/issue-1620/build-binutils-newlib/binutils'
  CC       objdump.o
  CCLD     objdump
...
```

closed 2024-11-26 by TommyMurphyTM1234