riscv-collab/riscv-gnu-toolchain #1621: Toolchain doesn't support "Format specifier like %g" in printf

RRArinc, 2024-11-21:
Hi,
I build the toolchain GCC13 version and able to compile the sample c code as below and observed the following.

// program to printf fixed and float point values
#include <stdio.h>
int main() {
     printf( "hello -%g- world -%f-\n", 1.0, 1.0f);
    return 0;
}

Any GCC compiler provides output as 
hello -1- world -1.000000-

However riscv32-unknown-elf-gcc compiles and outputs as 
hello -g- world -1.000000-

Here format specifier %g representing "fixed point" value is not printed correctly, but only prints "floating point" values.

Note: Toolchain is build for bare metal and not Linux flavour.

Any inputs/suggestions are welcome.

Thanks

TommyMurphyTM1234, 2024-11-24:
Any update @RRArinc?

RRArinc, 2024-11-24:
Hi,
Sorry i build the toolchain with source GCC 13.3.0
I will give a try with version GCC 13.2. By the way please can you confirm why GCC 13.3.0 doesn't support fixed point format (%g) specifier. Do i need to apply any patch?

Thanks in Advance.

TommyMurphyTM1234, 2024-11-24:
> By the way please can you confirm why GCC 13.3.0 doesn't support fixed point format (%g) specifier. Do i need to apply any patch?

Sorry, I don't know.
But I checked with the latest CI build which is now based on GCC 14.2.0 and that also works:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases
  - https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.11.22
    - https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.11.22/riscv32-elf-ubuntu-24.04-gcc-nightly-2024.11.22-nightly.tar.xz
```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.11.22/riscv32-elf-ubuntu-24.04-gcc-nightly-2024.11.22-nightly.tar.xz
tar xvf riscv32-elf-ubuntu-24.04-gcc-nightly-2024.11.22-nightly.tar.xz
cd riscv/bin

./riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/riscv/bin/../libexec/gcc/riscv32-unknown-elf/14.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/mnt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/mnt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32d --with-arch=rv32gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 14.2.0 ()

vi test.c
# Add the code above
./riscv32-unknown-elf-gcc test.c -o test

# Latest CI builds no longer bundle QEMU so I reused the version from the earlier 13.2.0 based CI build
~/temp/riscv/bin/qemu-riscv32 test
hello -1- world -1.000000-
```

RRArinc, 2024-11-25:
Hi, 
Please can you help me how to build and install qemu-riscv32.
I see in your recent comment using the qemu-riscv32 to debug the risc-v code.

Thanks

TommyMurphyTM1234, 2024-11-27:
Any update @RRArinc?
As already mentioned, I can't reproduce the original problem with GCC 13.2.0 or GCC 14.2.0.

TommyMurphyTM1234, 2024-11-28:
Closing the issue since it cannot be reproduced.

closed 2024-11-28 by TommyMurphyTM1234

RRArinc, 2024-12-04:
Hi, Thanks for the support, my apologies for being late here. i tried with qemu-riscv32 and it produced same results as expected.

Thank you very much.

TommyMurphyTM1234, 2024-12-04:
> Hi, Thanks for the support, my apologies for being late here. i tried with qemu-riscv32 and it produced same results as expected.

I presume by "same results" you mean the correct results and not the erroneous results? I.e. 

> ```
> ~/temp/riscv/bin/qemu-riscv32 test
> hello -1- world -1.000000-
> ```

RRArinc, 2024-12-05:
Yes, Your are right, output was 
hello -1- world -1.000000-