riscv-collab/riscv-gnu-toolchain #1622: Missing Documentation for --enable-qemu-system in README

karaketir16, 2024-11-22:
Hello,

I noticed that the --enable-qemu-system configuration option is not mentioned in the README file. It would be helpful to include information about this option, such as what it does and when it should be used.

Thank you!

TommyMurphyTM1234, 2024-11-22:
Isn't it covered in the QEMU documentation?

- https://www.qemu.org/documentation/

If not then the issue should probably be logged upstream? 

- https://www.qemu.org/contribute/

I don't think that the `riscv-gnu-toolchain` readme should duplicate instructions/documentation that are (or should be) covered adequately elsewhere.

karaketir16, 2024-11-22:
it's not about qemu docs, it's a configuration of this toolchain.

https://github.com/riscv-collab/riscv-gnu-toolchain/blob/20f615317e2ce888dfc11b29ccde4a649494b654/configure#L753

https://github.com/riscv-collab/riscv-gnu-toolchain/blob/20f615317e2ce888dfc11b29ccde4a649494b654/configure#L4639

if it's enabled
we set 
```qemu_targets=riscv64-linux-user,riscv32-linux-user,riscv64-softmmu,riscv32-softmmu```
if not enabled
```qemu_targets=riscv64-linux-user,riscv32-linux-user```

TommyMurphyTM1234, 2024-11-25:
> At least, `readme` can say like "use this flag, if you want to enable `qemu-system` targets"
> 
> I need to check `configure` file to find out this flag exist.

Please log a PR for this so.

closed 2025-01-19 by kito-cheng