riscv-collab/riscv-gnu-toolchain #1623: Unable to Build riscv-gnu-toolchain: libunwind Dependency Error

404allen404, 2024-11-25:
Hello,
I encountered an error while building the `riscv-gnu-toochain` using the following configuration:
`
./configure --prefix=$RISCV           \
              --enable-llvm             \
              --enable-linux            \
              --with-arch=rv64ima_zicsr \
              --with-abi=lp64
make
`

The error message is as follows:
`
CMake Error at /home/yiteng/riscv-gnu-toolchain/llvm/libunwind/src/CMakeLists.txt:102 (message):
  Compiler doesn't support generation of unwind tables if exception support
  is disabled.  Building libunwind DSO with runtime dependency on C++ ABI
  library is not supported.
`

It seems to be related to the dependency on the C++ ABI library for building `libunwind`. I am unsure how to rsolve this issue.
Could you provide guidance on how to fix this error or suggest the correct configuration flags to avoid it?
Thank you in advance for your assistance!

TommyMurphyTM1234, 2024-11-25:
Could this be relevant?

- https://gitlab.kitware.com/cmake/cmake/-/issues/24561

404allen404, 2024-11-25:
@TommyMurphyTM1234 
This is my build.log
[build.log](https://github.com/user-attachments/files/17901599/build.log)

TommyMurphyTM1234, 2024-11-25:
Both of these look quite old:

> and tried changing the GCC and G++ versions to 11,
> as well as using CMake 3.24.3.

My LLVM build is still in progress (it's very slow on the hardware that I have) but I don't think that it's suffering from the same problem. In the build log I see this:
```
-- Performing Test HAVE_CXX_FLAG_FNO_EXCEPTIONS
-- Performing Test HAVE_CXX_FLAG_FNO_EXCEPTIONS - Success
```
so I presume that I have got past the step where yours fails?

For what it's worth I'm using the following:
```
cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.1 LTS"
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

gcc --version
gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cmake --version
cmake version 3.28.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

Maybe your build platform and/or tools are simply too out of date?

Note that the CI builds take place on Ubuntu 22,04 and 24.04

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/20f615317e2ce888dfc11b29ccde4a649494b654/.github/workflows/nightly-release.yaml#L51

404allen404, 2024-11-25:
I will try it again, my os is also Ubuntu24.04, and gcc version is 13.2
[image]

TommyMurphyTM1234, 2024-11-25:
I don't understand:

> my OS is also Ubuntu 24.04, and gcc version is 13.2

The build log that you previously uploaded has this:

> ```
> -- The C compiler identification is GNU 11.4.0
> -- The CXX compiler identification is GNU 11.4.0
> ```

404allen404, 2024-11-25:
@TommyMurphyTM1234 
I used the same version as you, but it still didn't work.
[image]

TommyMurphyTM1234, 2024-11-25:
> Sorry, this is new log using GCC 13.2.0 [build.log](https://github.com/user-attachments/files/17905274/build.log)

That doesn't seem to be a full build log?
You need to capture a full build log for a clean build:
```
make distclean 
rm -rf $RISCV
make 2>&1 | tee build.log
```

404allen404, 2024-11-25:
@TommyMurphyTM1234 
This is full log
[build.log](https://github.com/user-attachments/files/17905840/build.log)

TommyMurphyTM1234, 2024-11-25:
My build finally completed and did so successfully.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1623
cd issue-1623
./configure --prefix=`pwd`/installed-tools --enable-llvm --with-arch=rv64ima_zicsr --with-abi=lp64
make 2>&1 | tee build.log

ls installed-tools/bin
amdgpu-arch            clang-offload-packager  lld-link      llvm-pdbutil                   riscv64-unknown-elf-as          riscv64-unknown-elf-gcc-ranlib     riscv64-unknown-elf-objdump
analyze-build          clang-refactor          llvm-ar       llvm-profdata                  riscv64-unknown-elf-c++         riscv64-unknown-elf-gcov           riscv64-unknown-elf-ranlib
clang                  clang-rename            llvm-cov      llvm-ranlib                    riscv64-unknown-elf-c++filt     riscv64-unknown-elf-gcov-dump      riscv64-unknown-elf-readelf
clang++                clang-repl              llvm-cxxfilt  llvm-rc                        riscv64-unknown-elf-clang       riscv64-unknown-elf-gcov-tool      riscv64-unknown-elf-run
clang-18               clang-scan-deps         llvm-dlltool  llvm-readobj                   riscv64-unknown-elf-clang++     riscv64-unknown-elf-gdb            riscv64-unknown-elf-size
clang-check            diagtool                llvm-dwp      llvm-size                      riscv64-unknown-elf-cpp         riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-strings
clang-cl               git-clang-format        llvm-lib      llvm-strings                   riscv64-unknown-elf-elfedit     riscv64-unknown-elf-gprof          riscv64-unknown-elf-strip
clang-cpp              hmaptool                llvm-mca      llvm-strip                     riscv64-unknown-elf-g++         riscv64-unknown-elf-ld             scan-build
clang-extdef-mapping   intercept-build         llvm-ml       llvm-symbolizer                riscv64-unknown-elf-gcc         riscv64-unknown-elf-ld.bfd         scan-build-py
clang-format           ld.lld                  llvm-nm       nvptx-arch                     riscv64-unknown-elf-gcc-14.2.0  riscv64-unknown-elf-lto-dump       scan-view
clang-linker-wrapper   ld64.lld                llvm-objcopy  riscv64-unknown-elf-addr2line  riscv64-unknown-elf-gcc-ar      riscv64-unknown-elf-nm             wasm-ld
clang-offload-bundler  lld                     llvm-objdump  riscv64-unknown-elf-ar         riscv64-unknown-elf-gcc-nm      riscv64-unknown-elf-objcopy

installed-tools/bin/clang --version
clang version 18.1.8 (https://github.com/llvm/llvm-project.git 3b5b5c1ec4a3095ab096dd780e84d7ab81f3d7ff)
Target: riscv64-unknown-unknown-elf
Thread model: posix
InstalledDir: /home/user/issue-1623/installed-tools/bin
```
My build log:

- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/17908453/build.log.zip)

As before, same host details as posted here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1623#issuecomment-2498140975

404allen404, 2024-11-27:
@TommyMurphyTM1234 
I noticed that your configuration is missing `--enable-linux`.

404allen404, 2024-11-27:
OK, thank you so much

TommyMurphyTM1234, 2024-11-27:
OK - I was able to reproduce the issue:
```
-- Performing Test CXX_SUPPORTS_WERROR_EQ_RETURN_TYPE_FLAG
-- Performing Test CXX_SUPPORTS_WERROR_EQ_RETURN_TYPE_FLAG - Failed
-- Performing Test CXX_SUPPORTS_FSTRICT_ALIASING_FLAG
-- Performing Test CXX_SUPPORTS_FSTRICT_ALIASING_FLAG - Failed
-- Performing Test CXX_SUPPORTS_EHSC_FLAG
-- Performing Test CXX_SUPPORTS_EHSC_FLAG - Failed
-- Performing Test CXX_SUPPORTS_FUNWIND_TABLES_FLAG
-- Performing Test CXX_SUPPORTS_FUNWIND_TABLES_FLAG - Success
-- Performing Test CXX_SUPPORTS_FNO_EXCEPTIONS_FLAG
-- Performing Test CXX_SUPPORTS_FNO_EXCEPTIONS_FLAG - Failed
-- Performing Test CXX_SUPPORTS_FNO_RTTI_FLAG
-- Performing Test CXX_SUPPORTS_FNO_RTTI_FLAG - Failed
CMake Error at /home/user/issue-1623/llvm/libunwind/src/CMakeLists.txt:102 (message):
  Compiler doesn't support generation of unwind tables if exception support
  is disabled.  Building libunwind DSO with runtime dependency on C++ ABI
  library is not supported.

-- Configuring incomplete, errors occurred!
make[3]: *** [runtimes/CMakeFiles/runtimes-riscv64-unknown-linux-gnu.dir/build.make:101: runtimes/runtimes-riscv64-unknown-linux-gnu-stamps/runtimes-riscv64-unknown-linux-gnu-configure] Error 1
make[3]: Leaving directory '/home/user/issue-1623/build-llvm-linux'
make[2]: *** [CMakeFiles/Makefile2:82268: runtimes/CMakeFiles/runtimes-riscv64-unknown-linux-gnu.dir/all] Error 2
make[2]: Leaving directory '/home/user/issue-1623/build-llvm-linux'
make[1]: *** [Makefile:156: all] Error 2
make[1]: Leaving directory '/home/user/issue-1623/build-llvm-linux'
make: *** [Makefile:1151: stamps/build-llvm-linux] Error 2
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/17939118/build.log.zip)

TommyMurphyTM1234, 2024-11-27:
Curiously this builds OK:
```
./configure --prefix=... --enable-llvm --enable-linux
make 2>&1 | tee build.log
```
but this fails:
```
./configure --prefix=... --enable-llvm --enable-linux --with-arch=rv64ima_zicsr --with-abi=lp64
make 2>&1 | tee build.log
```
So, for some reason, the non-default `arch/abi` seems to be triggering the problem.
At the moment I have no idea why.
Anybody else? E.g. @cmuellner or @kito-cheng perhaps?

cmuellner, 2024-11-27:
That's too much in the LLVM details. Maybe Kito has more experience with this.

TommyMurphyTM1234, 2024-11-29:
Below seems to be the first anomalous difference between the working and failing builds - maybe it is significant?
I.e. why does detection of the target ABI fail?

Working build:

- `build-rv64gc-lp64d.log.zip`: [build-rv64gc-lp64d.log.zip](https://github.com/user-attachments/files/17958931/build-rv64gc-lp64d.log.zip)

```
[100%] Performing configure step for 'runtimes-riscv64-unknown-linux-gnu'
CMake Deprecation Warning at /home/user/issue-1623/llvm/cmake/Modules/CMakePolicy.cmake:6 (cmake_policy):
  The OLD behavior for policy CMP0114 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

Not searching for unused variables given on the command line.

CMake Deprecation Warning at /home/user/issue-1623/llvm/cmake/Modules/CMakePolicy.cmake:11 (cmake_policy):
  The OLD behavior for policy CMP0116 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

-- The C compiler identification is Clang 18.1.8
-- The CXX compiler identification is Clang 18.1.8
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /home/user/issue-1623/build-llvm-linux/./bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
```

Failing build:

- `build-rv64ima_zicsr-lp64.log.zip`: [build-rv64ima_zicsr-lp64.log.zip](https://github.com/user-attachments/files/17958937/build-rv64ima_zicsr-lp64.log.zip)

```
[100%] Performing configure step for 'runtimes-riscv64-unknown-linux-gnu'
CMake Deprecation Warning at /home/user/issue-1623/llvm/cmake/Modules/CMakePolicy.cmake:6 (cmake_policy):
  The OLD behavior for policy CMP0114 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

Not searching for unused variables given on the command line.

CMake Deprecation Warning at /home/user/issue-1623/llvm/cmake/Modules/CMakePolicy.cmake:11 (cmake_policy):
  The OLD behavior for policy CMP0116 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

-- The C compiler identification is Clang 18.1.8
-- The CXX compiler identification is Clang 18.1.8
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /home/user/issue-1623/build-llvm-linux/./bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
```

yuv418, 2025-03-02:
We're facing the same error with `./configure --with-arch=rv64gc --with-abi=lp64 --enable-linux --enable-llvm --enable-libsanitizer`. We're on `gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0` here.

TommyMurphyTM1234, 2025-03-02:
> We're facing the same error

As I said above...

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1623#issuecomment-2551298379

> The only other suggestion that I have at the moment is to see if the same sort of issue arises if LLVM/Clang is built for RISC-V using upstream sources and build scripts rather than the `riscv-gnu-toolchain` top level `Makefile`. If it does then the issue should probably be logged/discussed upstream. If it doesn't then there may be some issue with this repo's `Makefile`?

yuv418, 2025-03-02:
I haven't tried that yet because I don't have time to set that up (I'm not super experienced with toolchains and I'll need to spend a bit reading clang documentation on how to build it how we want) at the moment. If I get a chance, I'll try it and report back. 

I just thought I'd share my ./configure options as another data point in the meantime. I also tried to build without libunwind by editing `Makefile.in` which didn't work.