riscv-collab/riscv-gnu-toolchain #1624: Mismatch Between MASK_VLE32V in binutils And riscv-opcodes

wLui, 2024-11-25:
I noticed that the `MASK_VLE32V` in binutils doesn't match the one that's generated in [riscv-opcodes](https://github.com/riscv/riscv-opcodes).

binutils definition `binutils/include/opcode/riscv-opc.h`:
```
...
#define MATCH_VLE32V  0x00006007
#define MASK_VLE32V  0xfdf0707f
...
```

riscv-opcodes definition `encoding.out.h`
```
...
#define MATCH_VLE32_V 0x6007
#define MASK_VLE32_V 0x1df0707f
...
```

For reference the binary instruction for `vle32` is defined in `rv_v`:
```
vle32.v        nf 28=0 27..26=0 vm 24..20=0 rs1 14..12=0x6  vd 6..0=0x07
```

I was wondering which of these masks are correct. In the case that the binary instruction in the `riscv-opcodes` version is wrong, what should the correct binary representation be?

cmuellner, 2024-11-25:
The `nf` field (opcode[31:29]`) is defined in the instruction encoding of all vector load and store instructions.
However, it only encodes relevant information for segment loads/stores and whole vector register loads/stores.
`vle` and `vse` are vector unit-stride loads/stores, and don't the `nf` field does not have a meaning.
So, while it is possible to parse the nf field for `vle` and `vse`, there is not point in doing so.

That said, I would claim both referenced sources (Binutils and riscv-opcodes) are correct w.r.t. the RVV specification.

wLui, 2024-11-25:
Thanks for letting me know. I guess those particular bits can be flexible then. However, if I do substitute the riscv-opcodes' `MASK_VLE32_V`  into binutils' `riscv-opc.h`, the build ends up failing.

Excerpt from `build-newlib/riscv64-unknown-elf/newlib/config.log`, configured with ` $ ./configure --prefix=/opt/riscv-modified --with-arch=rv64gcv --with-abi=lp64d`

```
...
Error: internal: bad RISC-V opcode (bits 0xe0000000 undefined or invalid): vle32.v Vd,0(s)Vm
...
```

Where this error message led me to `binutils/gas/config/tc-riscv.c` in the function `validate_riscv_insn`. I'd think if those bits are flexible this error shouldn't occur but I could be missing something.

cmuellner, 2024-11-25:
Binutils puts a `0` into the `nf` field when assembling. It also expects a `0` when parsing.
This is a reasonable assumption.

If you insist on encoding a different value than 0, you can use the `.insn` directive.

Regarding the build error that you are observing:
Binutils does not allow you to have undefined bits in an instruction word.
Either have the `nf` field fixed (using the mask) or parse it properly like it is done for the segmented loads/stores.

aswaterman, 2024-11-26:
riscv-opcodes uses a single entry that encompasses both the segment loads (e.g. vlseg2e8.v) and regular loads (e.g. vle8.v).  binutils expands that entry out into all 8 instructions, which is more explicit.  Neither is wrong; they are just different representations of the same information.

TommyMurphyTM1234, 2024-11-27:
From the replies above I take it that this issue doesn't identify any actual bug with the `riscv-gnu-toolchain` tools?

wLui, 2024-11-27:
Alright thanks for the explanations, everyone. I wasn't aware that the instruction was being represented differently and handled a certain way in binutils, so I guess it isn't a bug.

closed 2024-11-27 by TommyMurphyTM1234