riscv-collab/riscv-gnu-toolchain #1628: fetching submodules requires authentication for qemu

chamalis, 2024-11-27:
Reproduce:

```bash
git clone https://github.com/riscv-collab/riscv-gnu-toolchain rgt
cd rgt
git submodule update --init --recursive
```

Output:
```
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'                                                                                                                  
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'                                                                                                                   

...
                                                                                                           
Cloning into '/home/user/Downloads/rgt/qemu/roms/edk2/UnitTestFrameworkPkg/Library/GoogleTestLib/googletest'...                                                                                                     
Cloning into '/home/user/Downloads/rgt/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook'...                                                                                                           

(gnome-ssh-askpass:3386053): Gtk-WARNING **: 16:32:28.259: cannot open display:                                                                                                                                    
error: unable to read askpass response from '/usr/libexec/openssh/gnome-ssh-askpass'                                                                                                                               
Username for 'https://github.com':
```

I have opened all the repos in `.gitmodules` file, at the browser, they open without any auth, only one typo (fixed in [latest PR](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1627)).

TommyMurphyTM1234, 2024-11-27:
Probably the same issue as raised here in this closed issue?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-2495661939

Unfortunately my attempt to address it got stuck:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-2498981276

TommyMurphyTM1234, 2024-11-27:
The workaround for the moment is to let the top level `riscv-gnu-toolchain` `Makefile` update/clone the submodules rather than manually doing a recursive update which is actually not necessary in order to build the `riscv-gnu-toolchain` tools.

chamalis, 2024-11-27:
Thanks Tommy.
Indeed, the github search in issue pages doesn't find that. 
The issue is when letting makefile handle it, this doesn't build all the extensions:

`./configure --prefix=/opt/riscv --enable-multilib`

mateoconlechuga, 2024-11-29:
No, this is just a duplicate of #1613

TommyMurphyTM1234, 2024-12-05:
Link to posts that explain how to bump QEMU which I will do when 9.2 releases imminently:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-2498981276
- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-2510343259
- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480#issuecomment-2520313412

TommyMurphyTM1234, 2024-12-12:
Fixed by this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1639

closed 2024-12-12 by TommyMurphyTM1234