riscv-collab/riscv-gnu-toolchain #1631: Question: Does auto-vectorization work with xtheadvector with gcc 14.2.0?

piratepanda47, 2024-11-30:
Basically I want to use auto vectorrization with t-head c906 processor and I am using Sipeed M1s Dock hardware. I have compiled the toolchain with standard vector ISA enabled `--with-arch=rv64gcv` and auto vectorization is working with it.

**But the problem is that toolchain compiled with `xtheadvector` extension does not auto-vectorize the code. Does it work with `xtheadvector` ISA in gcc 14?**

### Details of toolchain where auto-vectorization is not working

**Toolchain configure command**
`./configure --prefix=/home/username/apps/bl808_toolchain --enable-multilib --enable-gdb --with-arch=rv64imafdc_zicsr_zifencei_xtheadba_xtheadbb_xtheadbs_xtheadcmo_xtheadcondmov_xtheadfmemidx_xtheadfmv_xtheadint_xtheadmac_xtheadmemidx_xtheadmempair_xtheadsync_xtheadvector --with-abi=lp64d`

**Sample Program**
```
#include <stdio.h>

int main()
{
  int a[256], b[256], c[256];
  int i;

  for (i=0; i<256; i++){
    b[i] = i;
    c[i] = i;
    a[i] = b[i] + c[i];
  }

  return 0;
}
```
**Command to compile the program**
`./riscv64-unknown-elf-gcc -o test.o -s test.c -march=rv64gc_xtheadvector -mcpu=thead-c906 -O3` (Also tried with O2 but didn't worked)

**Command to check object dump**
`./riscv64-unknown-elf-objdump -d test.o`

**Disassembly of object dump with no auto vectorization [no_autovectorized.txt](https://github.com/user-attachments/files/17965014/no_autovectorized.txt)**

**Disassembly of object dump with auto vectorization [autovectorized.txt](https://github.com/user-attachments/files/17965026/autovectorized.txt)**

cmuellner, 2024-11-30:
No, auto-vectorization is not supported for XTheadVector.
The primary use case for XTheadVector is to write vectorized code using [intrinsics](https://github.com/XUANTIE-RV/thead-extension-spec/blob/master/xtheadvector/intrinsics.adoc).

piratepanda47, 2024-11-30:
Thanks @cmuellner for the support. It really saved me a lot of trouble.

closed 2024-11-30 by piratepanda47