riscv-collab/riscv-gnu-toolchain #1632: clang ignores newlib nano

AgamemnonasKyriazis, 2024-11-30:
I built the toolchain for the rv32i_zicsr_zifencei  architecture and ilp32 abi and llvm enabled. When I tried to compile using 	
-specs=nano.specs and -specs=nosys.specs  the compiler ignored these flags and created a huge executable that could not fit into my memory. 

`./configure --prefix=/opt/riscv --with-arch= rv32i_zicsr_zifencei --with-abi=ilp32 --enable-llvm`

Is newlib-nano supported by clang? Did I do something wrong with the configuration ?

I use the following configuration for building my elf file

`CC = riscv32-unknown-elf-gcc 

CFLAGS = \
	-march=rv32i_zicsr_zifencei \
	-mabi=ilp32 \
	-O0 \
	-I ./src

LDFLAGS = \
	-march=rv32i_zicsr_zifencei \
	-mabi=ilp32 \
	-specs=nano.specs \
	-specs=nosys.specs \
	-fno-common \
	-fverbose-asm \
	-nostartfiles \
	-mstrict-align \
	-mcmodel=medany \
	-fno-builtin \
	-fsigned-char \
	-fdata-sections \
	-ffunction-sections \
	-Wno-unused \
	-T ./link.ld \
	-Wl,--gc-sections`

TommyMurphyTM1234, 2024-12-01:
> With clang it looks like it ignores the newlib nano directive

Isn't that literally what the build log is telling you? 

> ```
> riscv32-unknown-elf-clang: warning: argument unused during compilation: '-specs=nano.specs' [-Wunused-command-line-argument]
> riscv32-unknown-elf-clang: warning: argument unused during compilation: '-specs=nosys.specs' [-Wunused-command-line-argument]
> ```

You might need to ask about this in an appropriate upstream LLVM/Clang forum.

Also...

- https://www.reddit.com/r/RISCV/comments/1h4061l/does_riscv_toolchain_provided_clang_not_work_with/

> clang apparently doesn't have an equivalent of GCC's specs files. I suggest either use gcc, or read the clang manual (sorry, I don't use clang, I can't help you there). Once you get into writing linker scripts and spec files, you're getting far away from what the normal compiler user does, and I suspect clang and gcc start to diverge in how they do these things.

TommyMurphyTM1234, 2024-12-01:
You'll probably enable verbose logging to see what exactly is going on.

If this is an issue with LLVM/Clang then you should log/discuss it upstream.
Only issues that specifically impact the `riscv-gnu-toolchain` building and integration of the various tools are dealt with here.
Otherwise issues/bugs with specific components should be logged and discussed in the relevant upstream project.

closed 2024-12-01 by AgamemnonasKyriazis