riscv-collab/riscv-gnu-toolchain #1633: bool type variable assign to uint32_t variable problem

chenzhihong007, 2024-12-05:
# Problem
Bool type variable `test_b` convert to uint32_t, then `& 0x01` assign to uint32_t variabe `test_u32`, assembly code don't have `& 0x1` code.

```c
bool test_b;
uint32_t test_u32;

test_u32 = (uint32_t)test_b & 0x01;
```

asm code:
[image]

But LLVM is ok as follows.
[image]

TommyMurphyTM1234, 2024-12-05:
The source code in the two screenshots is not identical so not directly comparable.
Please provide a self contained, minimal, reproducible, test case and details of how you're compiling and disassembling it in the two cases (GCC and LLVM/Clang) including all command line options etc.
Please try to avoid screenshots as they are difficult to read and impossible to search/process. 
Use logs instead. And if they are large please attach rather than copy and paste them.

cmuellner, 2024-12-05:
The compiler assigns only values of `0` and `1` to variables of type `bool`.
So, an and-mask with `0x01` can be eliminated.
What is more relevant in the context of booleans is a test for zero or non-zero.
In RISC-V, we have specific instructions for such tests (e.g. `snez`).

Regarding your example:
You are trying to convince a compiler to believe that your boolean `test_b` is actually a uint32_t (by casting it), but your experiment is too simple. Therefore, the compiler can optimize away casting and masking.

The LLVM code below is likely a different case because `ctrl_pin_value` seems to have a different type than `bool`. But I'm not going to make any guesses here. Please provide a proper isolated function that shows a problem.

Here are some examples to get a better understanding:

```c
bool bar();

bool foo1(bool b)
{
    uint32_t tmp = (uint32_t)b;
    return b & 0x01;
}

bool foo2()
{
    return 2;
}

bool foo3(uint32_t v)
{
    bool b = (bool) v;
    return b & 0x01;
}
```

This compiles to (GCC and Clang produce the same code):
```
foo1(bool):
        ret
foo2():
        li      a0,1
        ret
foo3(unsigned int):
        snez    a0,a0
        ret
```

Observe the following:
* `foo1` shows the example that you provided. The compiler can eliminate all code.
* `foo2` returns `1` and not `2`, because the constant is evaluated at compile time to be non-zero.
* `foo3` is the experiment you may be interested in the most. Here, `snez` is the test for non-zero values.

closed 2024-12-05 by cmuellner

chenzhihong007, 2024-12-06:
# Test code
The Test code as follows:
```c
#include <stdio.h>

int main(void)
{
    _Bool test_b;
    unsigned char *ptr= (unsigned char *)&test_b;
    unsigned int test_u32;

    *ptr = 0x5A;
    test_u32 = (unsigned int)test_b & 0x01;

    printf("test_b: %x, test_u32: %x\n", test_b, test_u32);

    while(1)
    {
        ;
    }

    return 0;
}
```
# Result
## Gcc
- Gcc result is:
```
test_b: 5a, test_u32: 5a
```
- The asm code is:
[image]

## LLVM
- Gcc result is:
```
test_b: 0, test_u32: 0
```
- The asm code is:
[image]

TommyMurphyTM1234, 2024-12-06:
How are you compiling this code in both cases?
The options - especially optimization options - used may influence the code generation. 
Also, the casting in this code looks very questionable.

chenzhihong007, 2024-12-06:
I compiling the code in linux，use the cmd `gcc test.c -o test`，don't have any optimization options.

I also compiling the code in `SEGGER Embedded Studio v8.20`，the compiler can choose `gcc` or `SEGGER`(LLVM),  and I don't set any optimization options.
[image]

Why I assign a value 0x5A to the bool type variable `test_b`? Because I simulated that the bool variable `test_b` was not initialized, and it's memory value is 0x5A, not 0 or 1. Then I used the the bool variable `test_b`, casting it to uint32_t and `& 0x01`，normally this will only affect bit0, but this is not as expected, it affect other bits.

When bool type casting to int32_t, it works ok. So bool type casting to uint32_t, then `& 0x1`, the result has problem.

## Bool casting to int32_t Demo
Delete `test_u32` variable `unsigned` as follows:
``` c
#include <stdio.h>

int main(void)
{
    _Bool test_b;
    unsigned char *ptr= (unsigned char *)&test_b;
    int test_u32;

    *ptr = 0x5A;
    test_u32 = (int)test_b & 0x01;

    printf("test_b: %x, test_u32: %x\n", test_b, test_u32);

    while(1)
    {
        ;
    }

    return 0;
}
```

The result as follows. As expected.
```
test_b: 5a, test_u32: 0
```

cmuellner, 2024-12-06:
Let me simplify the code:
```c
unsigned int badcast1(bool *bool_ptr)
{
    /* Cast an address to a boolean to an address to uint.  */
    unsigned int *uint_ptr = (unsigned int *)bool_ptr;
    /* Write via uint pointer.  */
    *uint_ptr = 0x5A;
    /* Read back via bool pointer.  */
    unsigned int test_u32 = (unsigned int)(*bool_ptr);
    /* GCC returns 0x5A ; Clang returns (0x5A & 0x01)*/
    return test_u32;
}
```

GCC complies to:
```
badcast1(bool*):
        li      a5,90
        sw      a5,0(a0)
        mv      a0,a5
        ret
```

Clang compiles to:
```
badcast1(bool*):
        li      a1, 90
        sw      a1, 0(a0)
        li      a0, 0
        ret
```

So you are arguing that violating the type system by forcing an address to a boolean into an address to unsigned int leads to different behavior in different compilers.

I believe that both generated sequences are correct. However, I'm not a language lawyer.
If you can prove one of the sequences wrong, then file a bug in the upstream compiler issue tracker, where you reference the rules that make the sequence incorrect.

Anyway, this cast is always a bad idea.
So, what the compiler does here is irrelevant because nobody would use such code for a real project.

TommyMurphyTM1234, 2024-12-06:
> I believe that both generated sequences are correct. However, I'm not a language lawyer. If you can prove one of the sequences wrong, then file a bug in the upstream compiler issue tracker, where you reference the rules that make the sequence incorrect.

And/or against the Segger tools: 

> I also compiling the code in SEGGER Embedded Studio v8.20，the compiler can choose gcc or SEGGER(LLVM), and I don't set any optimization options.