riscv-collab/riscv-gnu-toolchain #1634: `smrnmi` extension is not supported?

ZhekaS, 2024-12-06:
When trying to build multilib with `smrnmi` extension, the following error occurs:
```
xgcc: error: '-march=rv32imac_zicsr_zifencei_zba_zbb_smrnmi': extension 'smrnmi' starts with 's' but is unsupported standard supervisor extension
xgcc: error: '-march=rv32imc_zicsr_zifencei_zba_zbb_smrnmi': extension 'smrnmi' starts with 's' but is unsupported standard supervisor extension
```

Without this extension, the assmebler does not recognize the Smrnmi CSRs:

```
../../arch/riscv/startup.S: Assembler messages:
../../arch/riscv/startup.S:128: Error: unknown CSR `mncause'
../../arch/riscv/startup.S:129: Error: unknown CSR `mnepc'
../../arch/riscv/startup.S:130: Error: unknown CSR `mnstatus'

```

TommyMurphyTM1234, 2024-12-06:
I can't find any mention of terms like "smrnmi", "mncause" etc. on the GCC mailing lists so it seems to me that it may be that no work has yet been done to add support upstream, in which case the issue should be asked about and maybe logged there.

- https://gcc.gnu.org/lists.html
- https://gcc.gnu.org/bugs/

TommyMurphyTM1234, 2024-12-06:
Oh - maybe this is a Binutils rather than a GCC issue?

- https://sourceware.org/pipermail/binutils/2022-July/121690.html
-

ZhekaS, 2024-12-06:
> Oh - maybe this is a Binutils rather than a GCC issue?
> 
> * https://sourceware.org/pipermail/binutils/2022-July/121690.html

I think it is... If I understand it correctly, it is about AS recognizing these CSRs

TommyMurphyTM1234, 2024-12-06:
Binutils does seem to know about the smrnmi extension CSRs:

- https://github.com/bminor/binutils-gdb/blob/9d09e3919293185225638a7f29e4028b30f9e317/include/opcode/riscv-opc.h#L4104-L4108

TommyMurphyTM1234, 2024-12-06:
Maybe `smrnmi` is not the correct extension specifier for the GCC architecture string when using `-march=...`?

TommyMurphyTM1234, 2024-12-06:
OK - maybe the Binutils support hasn't made it into a public Binutils release yet?

- https://github.com/bminor/binutils-gdb/commit/f0bdf8c1691c9509eaebab323754e0530d4ff5d7

Have you tried building the toolchain with upstream master Binutils sources?

- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

ZhekaS, 2024-12-06:
> OK - maybe the Binutils support hasn't made it into a public Binutils release yet?
> 
> * [bminor/binutils-gdb@f0bdf8c](https://github.com/bminor/binutils-gdb/commit/f0bdf8c1691c9509eaebab323754e0530d4ff5d7)
> 
> Have you tried building the toolchain with upstream master Binutils sources?
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

I could give it a try..

TommyMurphyTM1234, 2024-12-06:
> OK - maybe the Binutils support hasn't made it into a public Binutils release yet?
> 
> * [bminor/binutils-gdb@f0bdf8c](https://github.com/bminor/binutils-gdb/commit/f0bdf8c1691c9509eaebab323754e0530d4ff5d7)

Yes - that commit was September but the most recent Binutils release (2.43) was August:

- https://sourceware.org/pub/binutils/releases/

ZhekaS, 2024-12-06:
> OK - maybe the Binutils support hasn't made it into a public Binutils release yet?
>
> * [...]

Apparently, a small gcc patch is also needed. With this one I got it to build and work:
```diff
diff --git a/gcc/common/config/riscv/riscv-common.cc b/gcc/common/config/riscv/riscv-common.cc
index 1c71ea43f..bb5d63a2c 100644
--- a/gcc/common/config/riscv/riscv-common.cc
+++ b/gcc/common/config/riscv/riscv-common.cc
@@ -187,6 +187,7 @@ static const riscv_implied_info_t riscv_implied_info[] =
   {"smaia", "ssaia"},
   {"smstateen", "ssstateen"},
   {"smepmp", "zicsr"},
+  {"smrnmi", "zicsr"},
   {"ssaia", "zicsr"},
   {"sscofpmf", "zicsr"},
   {"ssstateen", "zicsr"},
@@ -351,6 +352,7 @@ static const struct riscv_ext_version riscv_ext_version_table[] =

   {"smaia",     ISA_SPEC_CLASS_NONE, 1, 0},
   {"smepmp",    ISA_SPEC_CLASS_NONE, 1, 0},
+  {"smrnmi",    ISA_SPEC_CLASS_NONE, 1, 0},
   {"smstateen", ISA_SPEC_CLASS_NONE, 1, 0},

   {"ssaia",     ISA_SPEC_CLASS_NONE, 1, 0},
```

TommyMurphyTM1234, 2024-12-06:
> Apparently, a small gcc patch is also needed. With this one I got it to build and work:

That's interesting @ZhekaS. Do you have a link for the source of that patch? 

And do you mean that `-march=rvXX..._smrnmi` is accepted and you can use (inline assembly?) instructions and CSRs from the `smrnmi` extension now?

ZhekaS, 2024-12-06:
> That's interesting @ZhekaS. Do you have a link for the source of that patch?
It is something I came up with after digging into the error message.

> And do you mean that `-march=rvXX..._smrnmi` is accepted and you can use (inline assembly?) instructions and CSRs from the `smrnmi` extension now?

Yep... here is the interleaved disassembly:
```
void test_smrnmi(void)
{
    __asm__("csrr    a0, mncause \t\n"
  817dde:	74202573          	csrr	a0,mncause
  817de2:	741025f3          	csrr	a1,mnepc
  817de6:	74402673          	csrr	a2,mnstatus
            "csrr    a1, mnepc  \t\n"
            "csrr    a2, mnstatus");
}
  817dea:	8082                	ret
```

TommyMurphyTM1234, 2024-12-06:
> > That's interesting @ZhekaS. Do you have a link for the source of that patch?
> 
> It is something I came up with after digging into the error message. Not anywhere online.

Ah, ok, thanks. Sounds like something that should maybe be submitted as a patch to upstream GCC? Unless this extension isn't expected to need `-march=rvXX..._smrnmi` at all but online assembly will just work for the extension's custom instructions and CSRs via the changes to the upstream Binutils? 

And eventually Binutils will be released with these changes integrated and `riscv-gnu-toolchain` will be updated so that its Binutils submodule will use the new release?

ZhekaS, 2024-12-06:
> Ah, ok, thanks. Sounds like something that should maybe be submitted as a patch to upstream GCC? Unless this extension isn't expected to need `-march=rvXX..._smrnmi` at all but online assembly will just work for the extension's custom instructions and CSRs via the changes to the upstream Binutils?

In fact, it's a good point. I did not try to build the upstream binutils without that string. I guess will do next.

TommyMurphyTM1234, 2024-12-08:
> void test_smrnmi(void)
> {
>     __asm__("csrr    a0, mncause \t\n"
>   817dde:	74202573          	csrr	a0,mncause
>   817de2:	741025f3          	csrr	a1,mnepc
>   817de6:	74402673          	csrr	a2,mnstatus
>             "csrr    a1, mnepc  \t\n"
>             "csrr    a2, mnstatus");
> }

I did this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1634
cd issue-1634
git clone https://github.com/bminor/binutils-gdb binutils-master --depth=1
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-binutils-src=`pwd`/binutils-master
make
cd installed-tools/bin
```
Create `test.c`:
```
// test.c
void test_smrnmi(void)
{
    __asm__("csrr    a0, mncause \t\n"
            "csrr    a1, mnepc  \t\n"
            "csrr    a2, mnstatus");
}
```
Then compile it...
```
./riscv64-unknown-elf-gcc -c test.c
./riscv64-unknown-elf-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <test_smrnmi>:
   0:   1141                    addi    sp,sp,-16
   2:   e406                    sd      ra,8(sp)
   4:   e022                    sd      s0,0(sp)
   6:   0800                    addi    s0,sp,16
   8:   74202573                csrr    a0,mncause
   c:   741025f3                csrr    a1,mnepc
  10:   74402673                csrr    a2,mnstatus
  14:   0001                    nop
  16:   60a2                    ld      ra,8(sp)
  18:   6402                    ld      s0,0(sp)
  1a:   0141                    addi    sp,sp,16
  1c:   8082                    ret
```
So I don't think that support for `-march=rvXX..._smrnmi` is required or expected. After all I don't think that the compiler itself would ever generate the instructions added by this extension but these would be expected to be used explicitly (e.g. in inline assembly or a raw assembly file) if needed?

TommyMurphyTM1234, 2024-12-08:
The only thing that I can't ascertain at the moment is the likely timeline for Binutils 2.44 to be released.

ZhekaS, 2024-12-09:
Based on the discussion and tests, it looks like no further action is needed, as the support will be out of the box once the next(?) version of Binutils (v2.44 ?) is released. Alternatively the toolchain can be built with the current upstream Binutils master, using the `--with-binutils-src=...` option.

closed 2024-12-09 by ZhekaS