riscv-collab/riscv-gnu-toolchain #1635: rv32imcb extensions

Adam11072000, 2024-12-08:
Hello all,

i am trying to install risc-v compiler with rv32imcb flags, for 32bit systems, and for some reason the make is failing:
i am using this command as the configure command:
./configure --prefix=/opt/riscv --with-arch=rv32imcb --with-abi=ilp32
and then:
sudo make

the make is failing with the following message:
xgcc: error: '-march=rv32imcb': extension 'b' is unsupported standard single letter extension

TommyMurphyTM1234, 2024-12-08:
Some additional background...

- https://github.com/riscv/riscv-b/issues/5

cmuellner, 2024-12-08:
The error states that your compiler does not support the single letter extension `b`.

The B extension (Zba + Zbb + Zbs) was ratified in April 2024.
See also: https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154732/Ratified+Extensions

GCC support was added in July 2024 and will be in the next GCC release (GCC 15).
See also: https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=2a90c41a131080e5fdd2b5554fcdba5c654cb93f

closed 2024-12-08 by cmuellner

TommyMurphyTM1234, 2024-12-08:
> GCC support was added in July 2024 and will be in the next GCC release (GCC 15). See also: https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=2a90c41a131080e5fdd2b5554fcdba5c654cb93f

Hi @cmuellner - just to clarify ... I presume you mean that GCC 15 will allow, for example, `rv32imcb` implying `rv32imc_zba_zbb_zbs`? The current `riscv-gnu-toolchain` repo using GCC 14.2 accepts the latter (`rv32imc_zba_zbb_zbs`) OK but not `rv32imcb` as originally reported.

cmuellner, 2024-12-08:
Yes. Before the ratification of `B` earlier this year, RISC-V did not have a (single-letter) `B` extension. Instead, we had four individual B extensions (`Zba`, `Zbb`, `Zbc`, `Zbs`). With `B` being a ratified single-letter extension now, it can be used the same way as all other single-letter extensions in ISA strings (e.g. rv64gcb) if tool support is available.

TommyMurphyTM1234, 2024-12-08:
> Yes. Before the ratification of `B` earlier this year, RISC-V did not have a (single-letter) `B` extension. Instead, we had four individual B extensions (`Zba`, `Zbb`, `Zbc`, `Zbs`). With `B` being [...]

Thanks @cmuellner. Just one point - I believe that using `b` in the architecture string will mean `zba_zbb_zbs` and `zbc` remains a separate extension to these.

cmuellner, 2024-12-08:
Yes, the `B` extension implies `Zba` + `Zbb` + `Zbs`. So, `Zbc` is not part of `B`.

TommyMurphyTM1234, 2024-12-08:
Thanks a lot. 👍